merbanan/rtl_433 #2054: RTL_TCP Passthrough?

Dinosaur-Ceiling, 2022-04-24:
I am attempting to use a RTL_SDR RTL2832U with an OTG cable on a phone which has Kali Nethunter installed. In order to use the RTL_SDR I have to use the RTL-SDR driver application on Android which is a port of rtl_tcp to the platform which allows other applications to start rtl_tcp and use the raw I/Q samples that it reads from USB DVB-T dongles based on the RTL2832U chip. My end goal is it use the RTL with an OTG cable to survey TPMS data and visualize it with the Kismet GUI.

I have been able to get RTL_SDR data to populate in Kismet on a normal install on a Kali Linux computer. But I cannot get data to populate in Kali Nethunter. The Kali Nethunter Kismet can see the RTL_SDR USB driver, but from what I can tell, since the RTL_SDR is only accessable to Android via the ported RTL_TCP app, Kismet it is not seeing any of the data being generated as it only recognizes the USB. My guess is that it would required a TCP passthrough. Is something like the following possible? `rtl_433 -f 315000000 -d rtl_tcp:127.0.0.1:14423 -F rtl_tcp:127.0.0.1:2501`

zuckschwerdt, 2022-04-24:
Yes, that is the intended use-case ;) For now you need a recent version from git master and compile yourself for the rtl_tcp output. The next release will include it.

Dinosaur-Ceiling, 2022-04-24:
@zuckschwerdt  Thanks for the follow up! Very cool to see that being added. Your other post is what got me thinking that this option may be the solution I have been looking for to get the data visualized in Kismet. 

One more question, and I apoligize because this might be a dumb one. I have never used git master before. How would I get that version and compile it? Any good references I can look at on how to do that?

zuckschwerdt, 2022-04-24:
That refers to our up-to-date and default Git branch. If you have all prerequisite installed it's these steps:
https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#cmake

Dinosaur-Ceiling, 2022-04-24:
Thank you so much for the answer and information! Will be giving this a try today to see if this is the answer I have been looking for.

Dinosaur-Ceiling, 2022-04-24:
Attempted to follow the directions you provided, but running into the following error:

Cmake Error at CMakeLists: txt: 223
RTL-SDR development files not found.

zuckschwerdt, 2022-04-24:
You will need some deps: https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#linux--mac-os-x

Dinosaur-Ceiling, 2022-04-24:
Thanks, was able to get the correct dependencies and finish the install. Just one more dumb question (sorry, first time doing this with code). How do you exectue the compiled version of rtl_433?

zuckschwerdt, 2022-04-25:
Execute it just like before, `rtl_433`. If however the path was given as `/usr/bin/rtl_433` then change that to `/usr/local/bin/rtl_433`

Marhein963, 2022-05-07:
Sup dude, I'm having a super annoying issue with termux when I try to run NetHunter and I was hoping U could help me. The problem seems unsolvable, some how my I can run a VNC connection for any linger than 5 min, it's getting very annoying. 

When I run NetHunter on Termux and run the command kex I always get a VNC connection but after some minutes the server always falls down. Can U help me with that?

Ps: I'm currently running NetHunter Rootless.

[image]

After several minutes this message always appears: [Process completed (signal 9) - press Enter]. 

I can run Termux on background for hours but as soon as I star running a VNC connection by Kex on NetHunter it's just some minutes before Termux kill all process.

zuckschwerdt, 2022-05-07:
Closing as there is nothing to do in rtl_433. Feel free to continue the discussion anyway.

closed 2022-05-07 by zuckschwerdt