merbanan/rtl_433 #2057: option usage of trl_433

AutomGuy, 2022-04-26:
Hi Christian,
ist that usage posible? rtl_433 -F json:127.0.0.1:1514
to transmit everything in json format to a udp server?
Best Regards

zuckschwerdt, 2022-04-26:
Looks interesting. But for now we only have the `-F syslog:127.0.0.1:1514` output which prefixes a short header to the JSON.
For the format see https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_custom.py#L19-L25
What exactly is the UDP target? Is it custom software or some other project?

AutomGuy, 2022-04-26:
First of all i use a debian machine, at the moment i use the "-F syslog:127.0.0.1:1514" option. I have taken a look in that file "rtl_433_statsd_relay.py".
And have that rewritten in php and modify that to a udp server. The server starts as system service. That is really easily to do in linux. this server handles all my 433/868 devices. a config.json contains all my device with "id". that ensures that i handle only my devices. The server write a json file for each device into a RAM Disk and also for a unknown devices (from Neighbors). RAM Disk is neccessary because i have other devices in my HA system-> like  (DECT Sockets), MODBUS devices, Ethernet devices with web server on board. My main HA Server send all the json file over a running websocket server to my HA webpage. all unkown devices will be send over a signal client to my mobil. When i go into areas without LTE or WLAN (only GSM) important message will be send by SMS. :)

AutomGuy, 2022-04-26:
The thing is, syslog writes also to the system or only to ip:port?

zuckschwerdt, 2022-04-26:
Sounds like a good setup. Since you can already accept Syslog format the question is then about "what else does Syslog do"? Simple: nothing ;) It's just a UDP output with a Syslog header, it does not report to any system facility.

zuckschwerdt, 2022-04-26:
And rtl_433 has a built-in REST-API and WebSocket server. Start with `-F http` (open the UI at http://127.0.0.1:8433/) and see https://github.com/merbanan/rtl_433/blob/master/src/http_server.c#L15-L23 for options, if you like.

AutomGuy, 2022-04-26:
Under these circumstances, my question is solved. i extract the payload like "parse_syslog" its do.
i will take a look into the http server file.

closed 2022-04-26 by zuckschwerdt

nrc4cyber, 2022-05-22:
> And rtl_433 has a built-in REST-API and WebSocket server. Start with `-F http` (open the UI at http://127.0.0.1:8433/) and see https://github.com/merbanan/rtl_433/blob/master/src/http_server.c#L15-L23 for options, if you like.

Is there any way to enable `-F http` in the config file? Everytime I try `output http` it just returns `segmentation fault` and quits.

zuckschwerdt, 2022-05-23:
> try `output http` it just returns `segmentation fault`

That was just recently fixed by #2068