merbanan/rtl_433 #2060: Not reading pressure (barometer) from Fineoffset-WH24

handly-aero, 2022-04-27:
Hi,
I am using rtl_433 to read data from a Fineoffset-WH24 PWS. 
rtl_433 correctly identifies the signal as "[78]  Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor", but it does not read pressure data (baromin & absbaro from the native station software).
Same thing with a WS2032 station, in the same frequency...
Can someone help me?
Thanks!

handly-aero, 2022-04-27:
> Hi, I am using rtl_433 to read data from a Fineoffset-WH24 PWS. rtl_433 correctly identifies the signal as "[78] Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 [...]

Attaching samples...

zuckschwerdt, 2022-04-27:
Ambient pressure is an indoor sensor and not in the RF data.