merbanan/rtl_433 #2063: Adding support for Carchet TPMS

kesuhm, 2022-05-04:
Hi, there!  I would like to contribute to your project.  

I have (mostly) characterized [this](http://carchet.easyofficial.com/carchet-rv-trailer-car-solar-tpms-tire-pressure-monitoring-system-6-sensor-lcd-display-p6.html) aftermarket Carchet TPMS sensor.  I lost one of my sensors and, instead of buying a replacement external sensor, I would like to install internal sensors.  I currently have to remove the hubcaps from my outer back tires to remove the sensors from my inner back tires to add air when the tires are low.  It's a headache! Carchet doesn't sell internal TPMS sensors, so I am using a different model.  My goal is to intercept the signal from the new internal sensors and relay it to the Carchet monitor in a language that the receiver understands.  Here is a short summary of what I've done so far:

I began with Universal Radio Hacker (URH) software. Data collection parameters were as follows: 
	Sensor frequency: 433.92 MHz
	Sample rate: 2.0 MSps
	Bandwidth: 2.0 Hz
	Gain: 125

The modulation is ASK (OOK) and data packets had the following format:
	aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa [Pause: 897679 samples]
	aaaaaaaa5956a5a5a6555aaa65959999a5aaaaaa [Pause: 6030 samples]
	aaaaaaaa5956a5a5a6555aaa65959999a5aaaaaa [Pause: 11176528 samples]

The decoding is Manchester I.  After decoding, the packets look like this:
	00000000000000000000000000000000000000
	0000a9332fc0a84f1000
	0000a9332fc0a84f1000

The packets could be detected with rtl_433 (compiled on Windows 7 with CMake and MinGW) using the following settings: 
	rtl_433.exe -X "n=Carchet,m=OOK_MC_ZEROBIT,s=50,l=50,r=1000,invert" -s 1M

Here are some samples on BitBench:

Case 1: [Decreasing pressure](https://triq.net/bitbench#c=%5BDEC%20PRESSURE%20SERIES%5D&c=%5BPressure%3A%2031%20PSI%20-%3E%200%20PSI%5D%5BTemp%3A%2084F%5D&c=%5BTiming%3A%20fast%20output%20mode%2C%201.9%20sec%20between%20pulse%20pairs%5D&c=%5B0000%5D%5BChecksum-8%20mod%20256%5D%5BID%3A2b1ade%5D%5BP%3Ahh%5D%5BT%3Ahh%5D%5BF%3Abbbb%5D%5B000%5D&c=0000c12b1ade3f4f1000&c=0000b92b1ade374f1000&c=0000b12b1ade2f4f1000&c=0000a72b1ade254f1000&c=0000a32b1ade214f1000&c=0000992b1ade174f1000&c=00008d2b1ade0b4f1000&c=0000892b1ade074f1000&c=0000892b1ade074f1000&c=0000852b1ade034f1000&c=0000832b1ade014f1000&f=hhhh%20Ck%3Ahh%20ID%3Ahhhhhh%20P%3A8h%20T%3A8h%20Flags%3Abbbb%20hhh&cw=4)
Case 2: [Increasing pressure](https://triq.net/bitbench#c=%5BINC%20PRESSURE%20SERIES%5D&c=%5BPressure%3A%2016%20-%3E%2021%20PSI%2C%2080F%5D&c=%5BTiming%3A%20med%20output%20mode%3B%2011.26s%20between%20pulse%20pairs%5D&c=0000c52b1ade254d3000%20%5B16%20PSI%2C%2080F%5D&c=0000ca2b1ade2a4d3000%20%5B18%20PSI%2C%2080F%5D&c=0000cf2b1ade2f4d3000%20%5B19%20PSI%2C%2080F%5D&c=0000d42b1ade344d3000%20%5B21%20PSI%2C%2080F%5D&c=&c=%5BINC%20PRESSURE%20SERIES%5D&c=%5BPressure%3A%2016%20PSI%20-%3E%2027%20PSI%5D%5BTemp%3A%2079F%5D&c=%5BTiming%3A%20med%20output%20mode%3B%2012s%2C%2016s%2C%2021s%20period%5D&c=%5B0000%5D%5BChecksum-8%20mod%20256%5D%5BID%3A2b1ade%5D%5BP%3Ahh%5D%5BT%3Ahh%5D%5BF%3Abbbb%5D%5B000%5D&c=0000c32b1ade244c3000%20%5B16%20PSI%2C%2079F%5D&c=0000c72b1ade284c3000%20%5B17%20PSI%2C%2079F%5D&c=0000ce2b1ade2f4c3000%20%5B19%20PSI%2C%2079F%5D&c=0000d32b1ade344c3000%20%5B21%20PSI%2C%2079F%5D&c=0000d72b1ade384c3000%20%5B22%20PSI%2C%2079F%5D&c=0000e52b1ade454d3000%20%5B27%20PSI%2C%2081F%5D&f=hhhh%20Ck%3Ahh%20ID%3Ahhhhhh%20P%3A8h%20T%3A8h%20Flags%3Abbbb%20hhh&cw=4)
Case 3: [Decreasing then increasing pressure](https://triq.net/bitbench#c=%5BDEC%20P%20THEN%20INC%20P%20SERIES%5D%5B1.89s%20between%20pulse%20pairs%5D&c=%5B0000%5D%5BChecksum-8%20mod%20256%5D%5BID%3A2b1ade%5D%5BP%3A%2026%20PSI%20-%3E%2022%20PSI%20-%3E%2026%20PSI%5D%5BT%3A%20102F%5D%5Bflags%5D%5B000%5D&c=0000c42b1ade38591000&c=0000c22b1ade36591000&c=0000c32b1ade37591000&c=0000c42b1ade38591000&c=&c=&c=%5BDEC%20P%20THEN%20INC%20P%20SERIES%5D&c=%5B23%20PSI%20-%3E%2024%20PSI%20-%3E%2026%20PSI%5D%5BTemp%3A%2088F%5D&c=%5B0000%5D%5BChecksum-8%20mod%20256%5D%5BID%3A332fc0%5D%5BP%3Ahh%5D%5BT%3Ahh%5D%5BF%3Abbbb%5D%5B000%5D&c=0000c3332fc040511000%20%5B2%20pulses%2C%201%20sec%20between%5D%20%5B23%20PSI%5D&c=0000bf332fc03c511000%20%5B2%20pulses%2C%202%20sec%20after%201st%20pair%5D%5B24%20PSI%5D&c=0000bf332fc03c511000%20%5B2%20pulses%2C%202%20sec%20after%202nd%20pair%5D&c=0000bf332fc03c511000%20%5B2%20pulses%2C%202%20sec%20after%203rd%20pair%5D&c=0000c0332fc03d511000%20%5B2%20pulses%2C%202%20sec%20after%204th%20pair%5D&c=0000c0332fc03d511000%20%5B2%20pulses%2C%202%20sec%20after%205th%20pair%5D&c=0000e6332fc043513000%20%5B2%20pulses%2C%2013%20sec%20after%206th%20pair%5D&c=0000ec332fc049513000%20%5B2%20pulses%2C%2022%20sec%20after%207th%20pair%5D&c=%5Bflags%20%3D%200001%2C%20fast%20output%20for%20pressure%20dec%5D&c=%5Bflags%20%3D%200011%2C%20moderate%2013s%2F22s%20output%20for%20pressure%20inc%5D&f=hhhh%20Ck%3Ahh%20ID%3Ahhhhhh%20P%3A8h%20T%3A8h%20Flags%3Abbbb%20hhh&cw=4)
Case 4: [Additional learning](https://triq.net/bitbench#c=%5BFREEZER%20EXPERIMENTS%20TRIGGERED%20LOW%20BAT%20FLAG%5D&c=00009c2b1ade37329000%20%5BLow%20batt%2C%2020%20PSI%2C%2032F%5D%5Bfails%20cksum%20--%20expected%201C%5D%5Bgood%20cksum%20if%20low%20bat%20flag%20removed%5D&c=&c=%5BLow%20batt%2C%2025%20PSI%2C%2036F%5D&c=0000c92b1ade42343000%20%5Bgood%20cksum%5D&c=0000cd2b1ade4634b000%20%5Bfails%20cksum%20--%20expected%204d%5D%5Bif%20low%20bat%20flag%20omitted%20from%20cksum%2C%20it%20works%5D&c=&c=%5BMISC%20DATA%20POINTS%2C%20ID%20332fc0%5D&c=000083332fc000511000%20%5B0%20PSI%2C%2088F%5D&c=0000db332fc04b4e2000%20%5B27%20PSI%2C%2082F%5D%20%5Bone%20pulse%20pair%2C%20no%20fast%20flag%5D&c=0000d7332fc0474e2000%20%5B26%20PSI%2C%2082F%5D%20%5Bone%20pulse%20pair%2C%20no%20fast%20flag%5D&c=0000d8332fc0484e2000%20%5B26%20PSI%2C%2082F%5D%20%5Bone%20pulse%20pair%2C%20no%20fast%20flag%5D&c=0000bb332fc04c4d0000%20%5B28%20PSI%2C%2081F%5D%20%5Bone%20pulse%20pair%2C%20no%20fast%20flag%5D&c=&c=%5Bfast%20flag%2C%20increasing%20pressure%5D%5B29%20PSI%20-%3E%2030%20PSI%2C%2090F%5D&c=0000d5332fc051521000%20%5Bfirst%20pulse%20pair%5D&c=0000d6332fc052521000%20%5B1.9%20s%20between%20pulse%20pairs%5D&c=0000d6332fc052521000%20%5B1.9%20s%20between%20pulse%20pairs%5D&c=0000d6332fc052521000%20%5B1.9%20s%20between%20pulse%20pairs%5D&c=0000d7332fc053521000%20%5B1.9%20s%20between%20pulse%20pairs%5D&c=&c=0000bb332fc04c4d0000%20%5B28%20PSI%2C%2081F%5D%5Bsingle%20pulse%20pair%5D&c=0000ba332fc04a4e0000%20%5B27%20PSI%2C%2082F%5D%5Bsingle%20pulse%20pair%2C%201%20min%20later%5D&c=&c=%5BRV%20testing%20for%20high%20pressure%20range%5D&c=0000a9332fc09c4b2000%20%5B57%20PSI%2C%2077F%5D%5Bdec%20P%20to%20wake%5D%5Bcksum%20-%20expect%2029%2C%20got%20a9%20(diff%20%3D%200x80%2C%20128%20dec)%5D&c=%5Binc%20pressure%20to%20%3E%2070%20PSI%2C%20dec%20P%20to%20activate%20TPMS%5D&c=0000ae332fc0af4d1000%20%5Bchecksum%20expected%202e%2C%20got%20ae%20(diff%3D0x80)&c=0000d3332fc0b44d3000%20%5Bfast%20mode%2C%201s%20later%5D%5Bchecksum%20expect%2053%2C%20got%20d3%5D&c=0000d6332fc0b74d3000%20%5Bfast%20mode%2C%201s%20later%5D%5B66%20PSI%2C%2081F%5D%5Bchecksum%20expect%2056%2C%20got%20d6%20(diff%20%3D%200x80%2C%20128%20dec)%5D&c=%5B0x56%20%3D%2001010110%5D&c=%5B0xd6%20%3D%2011010110%5D&c=&c=0000de332fc0b7553000%20%5B64%20PSI%2C%2093F%5D%5Bexpect%205e%2C%20got%20de%5D&c=0000cc332fc0ba502000%20%5B65%20PSI%2C%2086F%5D&c=0000cb332fc0b9502000%20%5B67%20PSI%2C%2086F%5D&c=&c=0000c5332fc0b3502000&c=0000b2332fc0b14f1000&c=0000b1332fc0b04f1000%20%5B64%20PSI%2C%2084F%5D&c=&c=0000a9332fc0a84f1000&c=0000a6332fc0a54f1000&c=0000a5332fc0a44f1000&c=0000a4332fc0a34f1000&c=0000a3332fc0a24f1000%20%5B59%20PSI%2C%2084F%5D&f=hhhh%20Flags%3A%20Ck%3Ahh%20ID%3Ahhhhhh%20P%3A8h%20T%3A8h%20%5BLO%20BAT%5D%3Ab%20bbb%20hhh&cw=4)

I wrote and tested a Carchet decoder file, which I called tpms_carchet.c and attached as a text file.  (I have never contributed on GitHub before, so I am not sure who normally adds devices or modifies the rtl_433_devices header file or CMakeLists.txt file when a new device is added.)  I am planning to do some additional testing on the flags, but I wanted to share the code before I went any further.  Please let me know if you have any questions or if I need to provide anything else.  I'll share more on the flags when I get new data.

Thanks!
[tpms_carchet.txt](https://github.com/merbanan/rtl_433/files/8627834/tpms_carchet.txt)

zuckschwerdt, 2022-05-06:
That looks plausible. We usualy see TPMS with that coding and timing.

You would just add to the `include/rtl_433_devices.h` and then run `./maintainer_update.py` and recompile (and run `./maintainer_update.py` again to update the docs).

You also would fork, branch, push your changes, then PR. If it doesn't work we can insert your decoder manually.

zuckschwerdt, 2022-05-06:
- you'll need to cleanup indentation: 4 spaces, no tabs.

kesuhm, 2022-05-06:
Thanks for the review and instructions.  I will do what you recommend once I have a bit (or byte :smiley:) more understanding on the flags. Back soon!

gdt, 2023-10-15:
@kesuhm Any progress?   It seems like you have a working decoder and it would be great to get it into the code base.

kesuhm, 2023-10-20:
Hi, there!  I just submitted a PR. The Carchet TPMS sensor is similar to the EezTire-E618 sensor, which popped up in my display this week while I was testing my code from May of 2022.  While the sensor data appears to be identical to the EezTire-E618 output, I hope that the PSI and deg F readouts and low battery and inflate flags may be of some use to those running rtl_433.

zuckschwerdt, 2023-10-20:
Let's continue this in #2677 now.

closed 2023-10-20 by zuckschwerdt