merbanan/rtl_433 #2064: MQTT or connection?

giros79, 2022-05-06:
Hi everybody,
i succedeed to export data to Home Assistant trough 
MQTT, but sometimes happens that connection is lost and I see data 3-4 hours old. 
Dongle is connected to a raspberry pizero w. 
what i do is launching command from my laptop to the pizero using ssh.......
sometimes happens that connection is broken....sometimes instead everything looks ok but no data is displayed 
Do I need an hardware much better than this or I make something wrong?

this is an example of error I find
[image]

Thanks

zuckschwerdt, 2022-05-06:
Especially the "No route to host" looks like wifi connection problems. But it could also be resource stall.
Did you check `top` or `htop` to see what the load is?

I don't know if anyone tested rtl_433 on the original Zero. But i would suspect it to be about the cpu power of the Raspi 1 or even less? The Raspi 1 can just sustain rtl_433. A big difference would be if you have FSK protocols enabled?
Try to enable only the protocol you really need and also use `-Y squelch` to significantly reduce cpu needed.

giros79, 2022-05-06:
Not significant values. CPU was approx 13%. Probably pizero is not enough for such kind of work

zuckschwerdt, 2022-05-07:
Well at 13% it seems plenty enough. Must be the Wifi then?

giros79, 2022-05-07:
I can try to change position. Zero should be covered by a signal booster....but sometimes I detect temporary disconnections

zuckschwerdt, 2022-05-10:
I have that with a single Raspi among a batch too. Never figured out why the Wifi is dropping. If the Wifi comes back eventually then MQTT should be able to handle that though.

gdt, 2022-07-07:
@giros79 DId you resolve this?  Looking at this issue I don't see anything the project can do unless you are able to figure out a bit more about what's going on.

I would suggest you add some monitoring of whether the network is ok, separately from rtl_433.

giros79, 2022-07-07:
Hi! Yes it's perfect

closed 2022-07-07 by zuckschwerdt