merbanan/rtl_433 #2065: TFA-TwinPlus / BL999

DrPerryCoke, 2022-05-08:
Greetings ! I have 4 bl999. Of the four, 3 pieces work perfectly, one for some reason works fine only if it is placed right next to the antenna.
<details>
  <summary>Does not receive the signal well</summary>
```
time      : 2022-05-08 23:15:44
model     : TFA-TwinPlus Id        : 60
Channel   : 3            Battery   : 1             Temperature: 25.8 C       Humidity  : 62 %
Integrity : CHECKSUM
Analyzing pulses...
Total count:   80,  width: 286.57 ms            (71643 S)
Pulse width distribution:
 [ 0] count:   80,  width:  560 us [560;568]    ( 140 S)
Gap width distribution:
 [ 0] count:   38,  width: 3952 us [3948;3956]  ( 988 S)
 [ 1] count:   39,  width: 1884 us [1876;2000]  ( 471 S)
 [ 2] count:    2,  width: 8832 us [8832;8836]  (2208 S)
Pulse period distribution:
 [ 0] count:   38,  width: 4516 us [4512;4520]  (1129 S)
 [ 1] count:   39,  width: 2448 us [2440;2564]  ( 612 S)
 [ 2] count:    2,  width: 9400 us [9400;9400]  (2350 S)
Pulse timing distribution:
 [ 0] count:   80,  width:  560 us [560;568]    ( 140 S)
 [ 1] count:   38,  width: 3952 us [3948;3956]  ( 988 S)
 [ 2] count:   39,  width: 1884 us [1876;2000]  ( 471 S)
 [ 3] count:    3,  width: 9224 us [8832;10004] (2306 S)
Level estimates [high, low]:   2770,    109
RSSI: -7.7 dB SNR: 14.1 dB Noise: -21.8 dB
Frequency offsets [F1, F2]:    2595,      0     (+9.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
view at https://triq.org/pdv/#AAB010040102300F70075C240881818181828355+AAB02F040202300F70075C24088282818181818181828182828281828282828282818282828281828181828181818181828355
Attempting demodulation... short_width: 1884, long_width: 3952, reset_limit: 8840, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PPM,s=1884,l=3952,g=3960,r=8840'
pulse_demod_ppm(): Analyzer Device
bitbuffer:: Number of rows: 3
[00] { 5} f0             : 11110
[01] {36} 3f 44 08 5b e0 : 00111111 01000100 00001000 01011011 1110
[02] {36} 3f 44 08 5b e0 : 00111111 01000100 00001000 01011011 1110
```
</details>
<details>
  <summary>It receives the signal well</summary>
```
time      : 2022-05-08 23:15:12
model     : TFA-TwinPlus Id        : 27
Channel   : 2            Battery   : 1             Temperature: 23.6 C       Humidity  : 65 %
Integrity : CHECKSUM
Analyzing pulses...
Total count:  104,  width: 2316.60 ms           (579150 S)
Pulse width distribution:
 [ 0] count:    5,  width: 152420 us [140284;179804]    (38105 S)
 [ 1] count:    2,  width: 197164 us [193676;200652]    (49291 S)
 [ 2] count:    2,  width: 319448 us [294288;344608]    (79862 S)
 [ 3] count:    2,  width: 59240 us [54480;64000]       (14810 S)
 [ 4] count:    1,  width: 5976 us [5976;5976]  (1494 S)
 [ 5] count:    1,  width: 23140 us [23140;23140]       (5785 S)
 [ 6] count:    1,  width: 37000 us [37000;37000]       (9250 S)
 [ 7] count:    1,  width: 16580 us [16580;16580]       (4145 S)
 [ 8] count:   82,  width:  496 us [412;528]    ( 124 S)
 [ 9] count:    4,  width:  364 us [340;388]    (  91 S)
 [10] count:    2,  width:  268 us [256;284]    (  67 S)
 [11] count:    1,  width:  152 us [152;152]    (  38 S)
Gap width distribution:
 [ 0] count:   12,  width:   44 us [44;52]      (  11 S)
 [ 1] count:    2,  width:   72 us [72;72]      (  18 S)
 [ 2] count:    3,  width: 8904 us [8876;8948]  (2226 S)
 [ 3] count:   39,  width: 4024 us [3988;4108]  (1006 S)
 [ 4] count:   47,  width: 1976 us [1916;2284]  ( 494 S)
Pulse period distribution:
 [ 0] count:    5,  width: 152468 us [140328;179856]    (38117 S)
 [ 1] count:    2,  width: 201636 us [200724;202552]    (50409 S)
 [ 2] count:    2,  width: 319504 us [294332;344680]    (79876 S)
 [ 3] count:    2,  width: 59284 us [54528;64044]       (14821 S)
 [ 4] count:    1,  width: 6028 us [6028;6028]  (1507 S)
 [ 5] count:    1,  width: 23192 us [23192;23192]       (5798 S)
 [ 6] count:    1,  width: 37044 us [37044;37044]       (9261 S)
 [ 7] count:    1,  width: 16628 us [16628;16628]       (4157 S)
 [ 8] count:   39,  width: 4516 us [4508;4532]  (1129 S)
 [ 9] count:   47,  width: 2452 us [2424;2588]  ( 613 S)
 [10] count:    2,  width: 9392 us [9384;9404]  (2348 S)
Pulse timing distribution:
 [ 0] count:    5,  width: 152420 us [140284;179804]    (38105 S)
 [ 1] count:    2,  width: 197164 us [193676;200652]    (49291 S)
 [ 2] count:    2,  width: 319448 us [294288;344608]    (79862 S)
 [ 3] count:    2,  width: 59240 us [54480;64000]       (14810 S)
 [ 4] count:    1,  width: 5976 us [5976;5976]  (1494 S)
 [ 5] count:    1,  width: 23140 us [23140;23140]       (5785 S)
 [ 6] count:    1,  width: 37000 us [37000;37000]       (9250 S)
 [ 7] count:    1,  width: 16580 us [16580;16580]       (4145 S)
 [ 8] count:   82,  width:  496 us [412;528]    ( 124 S)
 [ 9] count:    4,  width:  364 us [340;388]    (  91 S)
 [10] count:    2,  width:  268 us [256;284]    (  67 S)
 [11] count:    1,  width:  152 us [152;152]    (  38 S)
 [12] count:   12,  width:   44 us [44;52]      (  11 S)
 [13] count:    2,  width:   72 us [72;72]      (  18 S)
 [14] count:    3,  width: 8904 us [8876;8948]  (2226 S)
 [15] count:   39,  width: 4024 us [3988;4108]  (1006 S)
Level estimates [high, low]:  15777,      3
RSSI: -0.2 dB SNR: 37.2 dB Noise: -37.4 dB
Frequency offsets [F1, F2]:   15966,      0     (+60.9 kHz, +0.0 kHz)
Guessing modulation: No clue...
```
</details>

I don't really understand Linux, maybe there is a simple solution, but I couldn't find it... And I also use a translator, maybe I could have missed something, so I'm sorry)

closed 2022-05-10 by DrPerryCoke

zuckschwerdt, 2022-05-10:
Did you figure out what was wrong? Didn't look obvious to me, sorry.

DrPerryCoke, 2022-05-11:
@zuckschwerdt When I wrote, there were problems with reception from one sensor, the next day the problem was solved by itself ... Rebooting the system may have helped, I did not understand)

zuckschwerdt, 2022-05-11:
The usual advice is too change the center frequency slightly. But from the "Frequency offsets" you posted that is not warranted, they are well in range. Mysterious.