merbanan/rtl_433 #2077: Honeywell Hometronic

Ian-Pratt, 2022-05-22:
I have some patches for the honeywell_cm921 decoder that adds support for Hometronic Manager HCM200d controller, HCC80R manifolds and HCW82 thermostats. 

These devices uses the same Ramses II protocol, but required a number of changes:
 * header type 1a is common, which includes two device IDs and a sequence number
 * the HCM200d sends multiple commands in the same radio message, separated by an ff00 sequence. I loop over each command.
 * in my experience, the first nibble of the 55:5f:f0:04 preamble is frequently corrupted in otherwise perfect packets, so I now search for a 26bit preamble instead of 30.
 * I've added decode support for a handful of extra commands. However, several of the commands involve the controller sending information about multiple zones in the same message e.g. one message sends a temperature reading for every configured zone, another sends a setpoint for every configured zone. Following the format of the existing output format, I've added a further 15 entries of the form "temp (zone X)" and the same for setpoints and flags, but this may not be the preferred way of doing things, please advise.

If there's interest in adding support I'd be happy to clean up the patches and post them them

Best,
Ian

zuckschwerdt, 2022-05-23:
The honeywell_cm921 is already quite complex compared to other decoders. But if there is a sender with more information then yes: please PR changes to use them.
A shorter preamble match to ignore the first few bits is always good.
Keys should be of the format "temperature_foo_C" and "setpoint_foo_C" if possible.

gdt, 2023-10-15:
@Ian-Pratt Are you still working on this?  Your changes sound very useful, and I think it's much better to file a PR as soon as you get to the point where you are willing to let other people see your changes, even if it's not ready to be merged (can be marked draft and/or just say that).   This captures the code for posterity and lets others work on it too.

Ian-Pratt, 2023-10-15:
I've been remiss in not posting my patch as promised. I've just created a draft pull request based on what I have, which has been working fine in my home for the last 18 months or so: 

https://github.com/merbanan/rtl_433/pull/2672
 
Apologies that its not against current master, and still contains some additional debug. I will have a go rebasing and cleaning up at a later date, but posting here for reference.

gdt, 2023-10-15:
Thanks!  If amenable to you, I lean to closing the issue and having the PR carry the torch.

Ian-Pratt, 2023-10-15:
Of course.

closed 2023-10-15 by Ian-Pratt

gdt, 2023-10-17:
Sorry, I read it and complained about debug forgetting reading here.    So whenever you are ready please just forcepush and we can go around again.