merbanan/rtl_433 #2079: Adding temp sensor: LaCrosse / technoline / Mobile alerts

emilic14, 2022-05-28:
Hello,

I am trying to add this new set of sensors... They go by different names/companies LaCrosse technology, or Technoline, etc and are part of the MAxxxx system (example MA10001).
The main receiver indicates that it is working on 868.3Mhz and I've actually scanned it to be more around 868.208e6

What I don't know how to do, is to decode it and add it to this database :-)
could someone help me?

Running rtl_433 -f 868.225e6 -a 4 
I get : 
`*** signal_start = 16538003, signal_end = 19388054, signal_len = 2850051, pulses_found = 1
Iteration 1. t: 1415025    min: 0 (0)    max: 2830050 (1)    delta 3567587328
Iteration 2. t: 1415025    min: 0 (0)    max: 2830050 (1)    delta 0
Distance coding: Pulse length 1415025

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 1415025
bitbuffer:: Number of rows: 0 
*** signal_start = 25563970, signal_end = 26255349, signal_len = 691379, pulses_found = 644
Iteration 1. t: 223865    min: 166 (643)    max: 447565 (1)    delta 25921
Iteration 2. t: 223865    min: 166 (643)    max: 447565 (1)    delta 0
Pulse coding: Short pulse length 166 - Long pulse length 447565

Short distance: 2, long distance: 56, packet distance: 40286

p_limit: 223865
bitbuffer:: Number of rows: 3 
[00] { 1} 80                                                                                                                         : 1
[01] {322} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[02] {321} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
*** signal_start = 26677840, signal_end = 27369251, signal_len = 691411, pulses_found = 643
Iteration 1. t: 223898    min: 166 (642)    max: 447630 (1)    delta 4489
Iteration 2. t: 223898    min: 166 (642)    max: 447630 (1)    delta 0
Pulse coding: Short pulse length 166 - Long pulse length 447630

Short distance: 52, long distance: 56, packet distance: 40263`

zuckschwerdt, 2022-06-01:
Don't hit the frequency dead-on, use `-f 868.1M`. Don't use `-a` but use `-A`.

emilic14, 2022-06-01:
> Don't hit the frequency dead-on, use `-f 868.1M`. Don't use `-a` but use `-A`.

Will try, would you mind explaining me why should we not hit the frequency dead-on? and why use 868.1M instead of 868.2M (as the frequency I was hitting is  868.225e6 and the stated by the product is  868.3M ? 
(just trying to understand and learn :) )

zuckschwerdt, 2022-06-02:
That looks way too noisy. Llikely too "loud", hard to say without a sample. You never want to see pulses shorten than 10 or even 20 us.

wittj, 2022-11-26:
Hi, I am looking for help on the integration of a TFA device (see this issue https://github.com/merbanan/rtl_433/issues/1693#issuecomment-1328085878 ). What is interesting here is that the device has a QR code which includes MA10100 as information. So there might be some relation between these devices. I have added some signal files to the other issue.

emilic14, 2023-08-03:
> Hi, I am looking for help on the integration of a TFA device (see this issue [#1693 (comment)](https://github.com/merbanan/rtl_433/issues/1693#issuecomment-1328085878) ). What is interesting here is that the device has a QR code which includes MA10100 as information. So there might be some relation between these devices. I have added some signal files to the other issue.

Indeed MA10100 seems as a similar device group!
see here : [https://github.com/sarnau/MMMMobileAlerts/blob/master/MobileAlertsDevices.markdown](url)

gdt, 2023-09-29:
What's next for this issue?

gedger, 2023-10-21:
I decode the Mobile Alerts using the linked repository. Can be used as a reference for all the demodulator and formats required.

https://github.com/baycom/tfrec

gdt, 2024-06-05:
As I see it, the combination of lack of interest/progress and very limited useful information in this issue does not justify keeping it open.

closed 2024-06-05 by gdt