merbanan/rtl_433 #2080: Decoding Maverick XR-30

jbfunk, 2022-05-29:
I recently tried to use a Maverick XR-30 dual probe BBQ thermometer (similar to the Maverick ET-732 but with "extended range") with rtl_433 and was hoping it would work with one of the preexisting device protocols. Unfortunately it did not.

Based on the FCC filing it's transmitting at 433.92 MHz and using GFSK.

I've tried to figure out a flex decoder using the -A option (and -Y minmax), but cannot get any sensible data using any that I've tried (including the analyzer-suggested 'm=FSK_PCM,s=4,l=4,r=4096' as well as 'm=FSK_MC_ZEROBIT,r=4096' with various short timings based on the analyzed pulse timings). I have a suspicion it may be Manchester encoded but really don't know. It also appears to be repeating the same signal four times similar to the [Maverick ET73x device](https://github.com/merbanan/rtl_433/blob/master/src/devices/maverick_et73x.c) which uses OOK and Manchester encoding. I attached four samples--the temperature of both probes was reading as 19 C at the time.

Any help with trying to decode this device would be greatly appreciated.

[MaverickXR30-Samples.zip](https://github.com/merbanan/rtl_433/files/8794022/MaverickXR30-Samples.zip)

jbfunk, 2022-06-03:
Thank you for the help with that.

I tried again with the transmitter farther away and was able to get better samples.

These are with Temp1 and Temp2 both at 19 C:
[MavXR30-Samples-Both19.zip](https://github.com/merbanan/rtl_433/files/8835588/MavXR30-Samples-Both19.zip)

These are with Temp1 at 20 C and Temp2 at 19 C:
[MavXR30-Samples-20-19.zip](https://github.com/merbanan/rtl_433/files/8835590/MavXR30-Samples-20-19.zip)

I ended up with a flex decoder of:
`-X 'n=MavXR30,m=FSK_PCM,s=360,l=360,reset=4096,bits=104,get=Temp1:@67:{10},get=Temp2:@77:{10}'`

The raw row of data in pdv that is output with 20 C/19 C is:
`55555555a723a72211444e4720aaaaaaab4e474e4422889c8e41555555569c8e9c884511391c82aaaaaaad391d39108a2272390`
In base-2 the first part of that is (the first of four signals quickly sent in succession):
`0101010101010101010101010101010110100111001000111010011100100010000100010100010001001110010001110010000`

However when looking at the ook files in pdv, the first bit output is a 0, but before that is a rapidly pulsing signal that almost looks like a 1 if it was just a solid pulse. Adding the 1 to the start of the base-2 yields:
`10101010101010101010101010101010110100111001000111010011100100010000100010100010001001110010001110010000`
In base-16:
`aaaaaaaad391d39108a22723900a`

Which you can then put in bitbench, and, using the [ET73x code](https://github.com/merbanan/rtl_433/blob/master/src/devices/maverick_et73x.c) to make some educated guesses, input
`BURST: 32h PRE: 32h STATE: 4h TA:10d TB:10d POST: 24h`
to yield
`BURST: aaaaaaaa PRE: d391d391 STATE: 0 TA:0552 TB:0551 POST: 23900a`
which is 20 C and 19 C with an offset of 532 (just like the ET73x).

Though it works fine to extract the temps based on the bit position as in the flex decoder above, adding the 1 at the start seems to align the nibbles better and indeed reveals an identical signal transmitted four times.

Is there any way to do math in a flex decoder so that it could subtract the offset (532) from the reported temp values?

Also, I have tried using "unique" and "gap" with a very large "reset" value (to force rows instead of separate signals) to try to suppress 3 of the 4 identical signals, but that didn't work. Any help on suppressing the repeats would also be very much appreciated.

zuckschwerdt, 2022-06-03:
The `d391d391` is a sync word we have seen before (rojaflex.c marlec_solar.c archos_tbh.c), so that is very plausible.
Sorry, there is currently no way for calculation in flex decoders. This should be a full decoder, and I guess the "POST" is a checksum?

jbfunk, 2022-06-04:
My guess is that the "POST" may be a checksum or the digest similar to [ET73x](https://github.com/merbanan/rtl_433/blob/master/src/devices/maverick_et73x.c), but figuring that out for sure is something I have zero experience with... Here are a few readouts (adding a 1 bit at the start as above) with different temperatures (same set of temperatures lead to same last four nibbles):

`BURST... , SYNC.... , ? , Ta/Tb , ?... : Ta C/Raw & Tb C/Raw`
`aaaaaaaa , d391d391 , 0 , 89e27 , 7ee6 : 19 C/551 & 19 C/551`
`aaaaaaaa , d391d391 , 0 , 8a227 , 2390 : 20 C/552 & 19 C/551`
`aaaaaaaa , d391d391 , 0 , 8a629 , ed3d : 21 C/553 & 21 C/553`
`aaaaaaaa , d391d391 , 0 , 8c229 , 4e24 : 28 C/560 & 21 C/553`
`aaaaaaaa , d391d391 , 0 , 8ce29 , fd80 : 31 C/563 & 21 C/553`

Interestingly, if I press RESYNC (while the temps are both at 21 C), for one set of four repeated transmissions it switches to:
`aaaaaaaa , d391d391 , 5 , b4368 , 1b0e : 720 & 872 (not 21 C/553)`
Then back to:
`aaaaaaaa , d391d391 , 0 , 8a629 , ed3d : 21 C/553 & 21 C/553`

Given the similarity to, and seeming variation on, the ET73x, perhaps @gismo2004 (who is credited as authoring [ET73x](https://github.com/merbanan/rtl_433/blob/master/src/devices/maverick_et73x.c)) may be able to shed some light on this...

jbfunk, 2022-06-05:
Here is what I have come up with so far for a decoder. Note the "???" lines which pertain to the digest. I do not know if those are correct. Please be aware I do not know C at all, so there may be errors...

```
/** @file
    Maverick XR-30 BBQ Sensor.

    Copyright (C) 2022 jbfunk
    Heavily derived from Maverick ET-73x BBQ Sensor (maverick_et73x.c), Copyright (C) 2016 gismo2004

    This program is free software; you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation; either version 2 of the License, or
    (at your option) any later version.
*/

/**
Maverick XR-30 BBQ Sensor.

The thermometer transmits 4 identical messages every 12 seconds at 433.92 MHz,

Each message consists of 26 nibbles (104 bits total) but the first (non-data) bit is getting dropped in reception, so the payload is shifted 7 bits left to align the bytes

Payload:

- R =  7 bit removed (binary: 0101010)
- B = 24 bit Burst (0xaaaaaa)
- P = 32 bit Preamble (0xd391d391)
- F =  4 bit device state (0=default; 5=init)
- T = 10 bit temp1 (degree C, offset by 532)
- t = 10 bit temp2 (degree C, offset by 532)
??? - D = 16 bit digest (over FTt, includes non-transmitted device id renewed on a device reset) gen 0x8810 init 0xdd38

    shifted byte:   0   1   2   3   4   5     6     7     8     9     10    11
    shifted nibble: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23
    msg:            B B B B B B P P P P P  P  P  P  F  T  T  Tt t  t  D  D  D  D
    BUR:24h PRE: 32h FLAG:4h TA:10d TB:10d | DIGEST:16h

*/

#include "decoder.h"

static int maverick_xr30_callback(r_device *decoder, bitbuffer_t *bitbuffer)
{
    data_t *data;

    if (bitbuffer->num_rows != 1)
        return DECODE_ABORT_EARLY;

    //check correct data length
    if (bitbuffer->bits_per_row[0] != 104) // 104 bits
        return DECODE_ABORT_LENGTH;

    uint8_t b[12];
    bitbuffer_extract_bytes(bitbuffer, 0, 7, b, 96);

    //check for correct burst/preamble (0xaaaaaad391d391)
    if ((b[0] != 0xaa ) || b[1] != 0xaa || b[2] != 0xaa || b[3] != 0xd3 ) || b[4] != 0x91 || b[5] != 0xd3 || b[6] != 0x91)
        return DECODE_ABORT_EARLY; // burst/preamble missing

    int pre    = b[3] << 24 | b[4] << 16 | b[5] << 8 | b[6];
    int flags  = (b[7] & 0xf0) >> 4;
    int temp1  = (b[7] & 0x0f) << 6 | (b[8] & 0xfc) >> 2;
    int temp2  = (b[8] & 0x03) << 8 | b[9];
    int digest = b[10] << 8 | b[11];

    float temp1_c = temp1 - 532.0f;
    float temp2_c = temp2 - 532.0f;

    char *status = "unknown";
    if (flags == 0)
        status = "default";
    else if (flags == 5)
        status = "init";

    uint8_t chk[3];
    bitbuffer_extract_bytes(bitbuffer, 0, 63, chk, 24);

    //??? digest is used to represent a session. This means, we get a new id if a reset or battery exchange is done.
    int id = lfsr_digest16(chk, 3, 0x8810, 0xdd38) ^ digest;

    decoder_logf(decoder, 1, __func__, "pre %08x, flags %0x, t1 %d, t2 %d, digest %04x, chk_data %02x%02x%02x, digest xor'ed: %04x",
                pre, flags, temp1, temp2, digest, chk[0], chk[1], chk[2], id);

    /* clang-format off */
    data = data_make(
            "model",            "",                     DATA_STRING, "Maverick-XR30",
            "id",               "Session_ID",           DATA_INT,    id,
            "status",           "Status",               DATA_STRING, status,
            "temperature_1_C",  "TemperatureSensor1",   DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp1_c,
            "temperature_2_C",  "TemperatureSensor2",   DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp2_c,
            NULL);
    /* clang-format on */

    decoder_output_data(decoder, data);
    return 1;
}

static char *output_fields[] = {
        "model",
        "id",
        "status",
        "temperature_1_C",
        "temperature_2_C",
        "mic",
        NULL,
};

r_device maverick_xr30 = {
        .name        = "Maverick XR-30 BBQ Sensor",
        .modulation  = FSK_PULSE_PCM,
        .short_width = 360,
        .long_width  = 360,
        .reset_limit = 4096,
        .decode_fn   = &maverick_xr30_callback,
        .fields      = output_fields,
};
```

zuckschwerdt, 2022-06-06:
Pretty sure we are just missing the first bit and it's theoretically there :)
You should document the data as:
```
- P = 32 bit preamble (0xaaaaaaaa)
- S = 32 bit sync-word (0xd391d391)
...
```
The  digest for the samples above seems to be: generator 8810 key 0d42 (fixed value/id here is 45c0).

jbfunk, 2022-06-10:
I added in your changes below. The issue with the missing first bit (which, regardless of distance, seems to oddly only be received temporarily after I press RESYNC) is accounted for with a quick check of the first byte to see if it's 55 or aa and then extracting the bytes from the bitbuffer with a shift of 7 or 8 bits left for analysis. I don't know if that's the best or correct way to address the missing bit, but it does take care of the issue.

```
/** @file
    Maverick XR-30 BBQ Sensor.

    Copyright (C) 2022 jbfunk
    Heavily derived from Maverick ET-73x BBQ Sensor (maverick_et73x.c), Copyright (C) 2016 gismo2004
    Additional assistance from zuckschwerdt

    This program is free software; you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation; either version 2 of the License, or
    (at your option) any later version.
*/

/**
Maverick XR-30 BBQ Sensor.

The thermometer transmits 4 identical messages every 12 seconds at 433.92 MHz,

Each message consists of 26 nibbles (104 bits total) but the first (non-data) bit (1) is getting dropped sometimes in reception, so for analysis the payload is shifted 7 bits left to align the bytes (or 8 bits if 0xaa is observed rather than 0x55 as the first byte received)

Payload:

- P = 32 bit preamble (0xaaaaaaaa; 7 or 8 bits shifted left for analysis)
- S = 32 bit sync-word (0xd391d391)
- F =  4 bit device state (0=default; 5=init)
- T = 10 bit temp1 (degree C, offset by 532)
- t = 10 bit temp2 (degree C, offset by 532)
- D = 16 bit digest (over FTt, includes non-transmitted device id renewed on a device reset) gen 0x8810 init 0x0d42

    byte (after shift):       0   1   2   3   4   5     6     7     8     9     10    11
    nibble (after shift):     0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23
    msg:                  P P P P P P P P S S S S S  S  S  S  F  T  T  Tt t  t  D  D  D  D
    PRE:32h SYNC: 32h FLAG:4h T:10d t:10d | DIGEST:16h

*/

#include "decoder.h"

static int maverick_xr30_callback(r_device *decoder, bitbuffer_t *bitbuffer)
{
    data_t *data;

    if (bitbuffer->num_rows != 1)
        return DECODE_ABORT_EARLY;

    //check correct data length
    if (bitbuffer->bits_per_row[0] != 104) // 104 bits
        return DECODE_ABORT_LENGTH;

    //check for correct preamble/sync (0xaaaaaad391d391)
    if (bitbuffer->bb[0][0] == 0x55) {
        uint8_t b[12];
        bitbuffer_extract_bytes(bitbuffer, 0, 7, b, 96); // shift in case first bit was not received properly
    } else if (bitbuffer->bb[0][0] == 0xaa) {
        uint8_t b[12];
        bitbuffer_extract_bytes(bitbuffer, 0, 8, b, 96); // shift in case first bit was received properly
    } else {
        return DECODE_ABORT_EARLY; // preamble/sync missing
    }
    if (b[0] != 0xaa || b[1] != 0xaa || b[2] != 0xaa || b[3] != 0xd3 || b[4] != 0x91 || b[5] != 0xd3 || b[6] != 0x91)
        return DECODE_ABORT_EARLY; // preamble/sync missing

    int sync   = b[3] << 24 | b[4] << 16 | b[5] << 8 | b[6];
    int flags  = (b[7] & 0xf0) >> 4;
    int temp1  = (b[7] & 0x0f) << 6 | (b[8] & 0xfc) >> 2;
    int temp2  = (b[8] & 0x03) << 8 | b[9];
    int digest = b[10] << 8 | b[11];

    float temp1_c = temp1 - 532.0f;
    float temp2_c = temp2 - 532.0f;

    char *status = "unknown";
    if (flags == 0)
        status = "default";
    else if (flags == 5)
        status = "init";

    uint8_t chk[3];
    chk[0] = b[7];
    chk[1] = b[8];
    chk[2] = b[9];

    //digest is used to represent a session. This means, we get a new id if a reset or battery exchange is done.
    int id = lfsr_digest16(chk, 3, 0x8810, 0x0d42) ^ digest;

    decoder_logf(decoder, 1, __func__, "sync %08x, flags %0x, t1 %d, t2 %d, digest %04x, chk_data %02x%02x%02x, digest xor'ed: %04x",
                sync, flags, temp1, temp2, digest, chk[0], chk[1], chk[2], id);

    /* clang-format off */
    data = data_make(
            "model",            "",                     DATA_STRING, "Maverick-XR30",
            "id",               "Session_ID",           DATA_INT,    id,
            "status",           "Status",               DATA_STRING, status,
            "temperature_1_C",  "TemperatureSensor1",   DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp1_c,
            "temperature_2_C",  "TemperatureSensor2",   DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp2_c,
            NULL);
    /* clang-format on */

    decoder_output_data(decoder, data);
    return 1;
}

static char *output_fields[] = {
        "model",
        "id",
        "status",
        "temperature_1_C",
        "temperature_2_C",
        "mic",
        NULL,
};

r_device maverick_xr30 = {
        .name        = "Maverick XR-30 BBQ Sensor",
        .modulation  = FSK_PULSE_PCM,
        .short_width = 360,
        .long_width  = 360,
        .reset_limit = 4096,
        .decode_fn   = &maverick_xr30_callback,
        .fields      = output_fields,
};
```

zuckschwerdt, 2022-06-10:
Should I commit your code as new decoder, or do you want to try a PR?

jbfunk, 2022-06-10:
I tried a PR: https://github.com/merbanan/rtl_433/pull/2090

gdt, 2023-10-02:
This PR looks merged.  @jbfunk What's the status of this issue?  Should we just close it?  Something else?

jbfunk, 2023-10-03:
Yes, it's merged.

closed 2023-10-03 by jbfunk