merbanan/rtl_433 #2084: Support for ambient weather remote sensor WH2B

maxmib, 2022-06-06:
Is wh2b supported by rtl433 ? Thanks .

zuckschwerdt, 2022-06-06:
The code is in [fineoffset.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c).
Currently we known about WH2 and WH2A. Post some samples and we'll add support for WH2B/C.
https://triq.org/rtl_433/ANALYZE.html

zuckschwerdt, 2022-06-06:
Is the device that old? What's noted there should fit with the WH2 code :)

maxmib, 2022-06-06:
I think wh2b is  at least eight years old according to comments on Amazon.I bought some at very cheap price . When I get them ,I will check if they can be recognized by rtl433.Thanks.

maxmib, 2022-06-18:
WH2b is supported by rtl433 ,it fits with WH2 code .so issue closed.

closed 2022-06-18 by maxmib

zuckschwerdt, 2022-06-18:
Thanks for the feedback!