merbanan/rtl_433 #2085: Aggregator UI

tb0hdan, 2022-06-07:
Hi guys,

It would be really nice to have some kind of aggregator UI for a community that is running RTL_433.  I am not much of a front-end guy,
but I've played around Google Maps and created a prototype that uses my own RTL-SDR dongle.

You can see it here: https://rtl433.0x21h.net

If you would like to submit your own sensor data (currently only temperature devices are supported) you can use a simple script:

```bash
rtl_433 -F json -K token=somegarbage -K lat=123.4 -K lon=23.4|curl --retry 3 --retry-connrefused -T- https://rtl433.0x21h.net/submit
```
In any case, this is a PoC and I really would like to see something more nice/mature.

I'll gladly discuss any details.

Thanks!

zuckschwerdt, 2022-06-07:
That's really great! User donated data is something I wanted to try someday too.

Once the threaded input #1978 is finished we could allow an output to block on, timeout, and retry HTTP push. Your current suggestion with curl is the best option for now though.

I'd also like options to donate raw data (pulse data, not I/Q) to maybe analyse and decode new devices easier and quicker. But let's start small ;)

tb0hdan, 2022-06-07:
Donating raw data might be dangerous (even illegal) in some countries. Having decoded stream for historical purposes/analytics is good, though. Let's see how this thread develops.

rct, 2022-06-07:
For the weather domain, there are some existing aggregators, like Weather Underground, PWS weather, Windy, ...

Off the top of my head there is WeeWX which can take use rtl_433 data instead of proprietary weather station software and submit it to various weather aggregators. There are probably other implementations.

You probably would want to make sure you can select what data is made public. For example, you wouldn't want to be collecting data from security sensors, TPMS, or even internal temperature sensors that might disclose too much info.

tb0hdan, 2022-06-07:
Is there any kind of list that tells what types of sensors should be excluded?

zuckschwerdt, 2022-06-07:
There is no usable typing-scheme in rtl_433. It would be really hard to do as many sensors are "kind of like $class but ...".
We would have to manually review the whole list, or better allow use only with explicit `-R`.

My envisioned use-case for "raw data" wasn't a global intake but a way for users to get help more effeciently. I.e. some option like "push samples from an new device to a public pdv instance". Maybe with accounts or at least a key/password.

tb0hdan, 2022-06-07:
One of the ways to make this more compliant is to show the map to logged-in users and their own sensors only.

merbanan, 2022-06-07:
For an aggregator I would envision that we need a config file that lists matching json key/values and key/values that are added if a match occurs. This way you can propagate locally unique devices and then add the static co-ordinates the sensor is at (or other key/values).

What would also be fun is an Android port that could be used for sensor location purposes. Just drive around slowly and aggregate sensor data together with the current gps location from the phone.

Regarding the legality of this it will always be up to the user to make sure they are within the legal bounds of the law.

zuckschwerdt, 2022-06-07:
I have never tried an Android port, but people did talk about that. The GPS thing is there :) The question is if gpsd (or equivalent) runs on Android or some other provider can deliver by TCP.
```
$ rtl_433 -K
		= Data tags option =
  [-K FILE | PATH | <tag> | <key>=<tag>] Add an expanded token or fixed tag to every output line.
	If <tag> is "FILE" or "PATH" an expanded token will be added.
	The <tag> can also be a GPSd URL, e.g.
		"-K gpsd,lat,lon" (report lat and lon keys from local gpsd)
		"-K loc=gpsd,lat,lon" (report lat and lon in loc object)
		"-K gpsd" (full json TPV report, in default "gps" object)
		"-K foo=gpsd://127.0.0.1:2947" (with key and address)
		"-K bar=gpsd,nmea" (NMEA default GPGGA report)
		"-K rmc=gpsd,nmea,filter='$GPRMC'" (NMEA GPRMC report)
	Also <tag> can be a generic tcp address, e.g.
		"-K foo=tcp:localhost:4000" (read lines as TCP client)
		"-K bar=tcp://127.0.0.1:3000,init='subscribe tags\r\n'"
		"-K baz=tcp://127.0.0.1:5000,filter='a prefix to match'"
```

merbanan, 2022-06-07:
Well, I think a wrapper app would be the best way forward. Then you could use the regular Location api and let the wrapper app inject the position to the mail rtl_433 process. If you configure the rtlsdr device with a fixed gain it should be possible to generate a heat map of unknown sensors.

gdt, 2022-06-09:
It would be nice to separate:
  * support for uploading
  * a software implementation of a server
  * operating an instance of a server

The first two are just Free Software issues, but the third raises issues about Open Data.   Also, it would be good to have the server operate without tracking (and thus without google) and without non-Free javascript.

tb0hdan, 2022-06-09:
I'm just going to share code I currently have after doing some cleanup. Hopefully, that'll inspire some developers to do a better job than I did :)

We'll need to have a discussion here to make it more relevant though

gdt, 2022-06-09:
>     * Support for uploading seems to be already there with curl and some shell magic :)

OK, so this is basically all out of scope.

>     * Would it make sense to use, for instance, OpenStreetMap or are you referring to a whole `log-in` concept?

I mean that loading maps from google is not ok, both because they track, and because it needs the user to run non-Free javascript.   That leads to using leaflet, and then you need a tile provider. OSM data is fine, but you shouldn't use their tile servers, which are for helping mappers and promoting the project, not for providing people free tiles.

>     * Legal matters. Perhaps can be solved by using proper ToS. Still, someone has to define what can be PII in terms of sensor data

I really just meant to have a license on the aggregated data, that presumably those uploading would consent to, sort of ODbL, CC-BY-SA, etc.   Often sites have terms granting lots of rights to the site and none to users.  I don't want to have anything to do with those (I've never even thought about sending to WU), and I don't think the project should endorse or accomdate those.

There is also the privacy issue.    Part 15 devices in the US don't have any legal protection (I have no idea about other countries), but there is a larger issue about what's polite and meets open source community standards.     Personally I think it's only ethical to publish/share data that is 1) known to be from your own sensors (or with permission of sensor owner) or 2) data that is known to be "weather", meaning information about outside environmental conditions that is not affected by nearby human activity, and perhaps 3) data that is already widely treated as public (perhaps AIS).   That leads to uploading being configured by device type/id only, and default off.    On top of that there are privacy considerations for one's own data, and there I think the wise choice is to upload only environmental/weather data.

gdt, 2023-10-02:
I think the conclusion is that an aggregator project is firmly out of scope, but that if there was a project, someone could (me making this up a bit now) create a python script to read json over syslog, like everything else does, and report.   So until there is a project with code and a spec and all that, I don't see that having this open is useful to rtl_433.  And, nothing has been happening for a really long time.

closed 2023-10-02 by gdt