merbanan/rtl_433 #2088: Sporadic reception of data from Acurite/TFA 30.3221.02

cristianlivella, 2022-06-08:
Hi there! I recently bought a pair of sensor for freezer/fridge ([Amazon link](https://www.amazon.it/gp/product/B0894NVTQF)).

I couldn't find much information about that specific model online, but it looked very similiar to some Acurite/TFA products, so I gave it a try anway, and in fact it seems it use the same protocol, as rtl_433 detect it as TFA 30.3221.02 (and sometimes as LaCrosse-TX141THBv2).

The issue is that the reception of the data is very sporadic, it is actually a rarity to be able to receive messages (less than one per hour, leaving rtl_433 running).

I don't think it's a transmission issue, as I can see the data updated about every minute on their display, and it doesn't seem a generic issue with my receiver (I'm using the RTL-SDR blog one), as I regularly receive updates from some alarm sensors, probably of some neighbour.

I also tried removing the sensors from the freezer/fridge, placed at a distance of a few metres from the antenna, without any obstacles, but even so I see no improvement.

I recorded a couple of samples (one per sensor) using `rtl_433 -S unknown`, in case anyone can look at them. Thanks

https://drive.google.com/drive/folders/1Ji026_LObdBc55eqA4oBRrgSRFFzpM9L

zuckschwerdt, 2022-06-08:
Huh, I wanted to say "likely the signal is at the very edge of the band, try slightly different center frequencies", but those samples look good and are well inside the band.

The trouble is something else: this model sends 4 repeats and we demand 3 to be intact. But there is a noise pulse at the start which gets registered and thus it looks like a 12 repeats model ("more than 4 packets") and we demand 5 intact repeats -- which of course are not there.
You are only seeing the bad transmissions with one packet missing ;)

Change this line https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_tx141x.c#L132 from
`r = bitbuffer_find_repeated_row(bitbuffer, bitbuffer->num_rows > 4 ? 5 : 3, 32); // 32`
to
`r = bitbuffer_find_repeated_row(bitbuffer, bitbuffer->num_rows > 5 ? 5 : 3, 32); // 32`
recompile and it should work.

cristianlivella, 2022-06-08:
This seems to work perfectly! I receive all transmissions every minute now (even with the sensors in the fridge on a different floor).

Thanks very much!

zuckschwerdt, 2022-06-09:
We only find these edge-cases with constant curiosity and help from the community. Thank *you*.

I'll do more tests and add this fix soon.

zuckschwerdt, 2022-06-09:
Are there any markings on the sensor, a name or model number?

commit 5c17e4e3 (2022-06-10): minor: Fix ORIA WA50B (closes #2088)
closed 2022-06-10 by zuckschwerdt