merbanan/rtl_433 #2091: Build Error

Sineos, 2022-06-12:
Trying to build on a RPi / Ubuntu and getting following error message:

```
ubuntu@ubuntu:~/01_src/rtl_433/build$ PKG_CONFIG_PATH=/usr/local/lib/pkgconfig cmake .. && cmake --build .
-- Build type not specified: defaulting to release.
-- Found Git version: refs/heads/master commit 5c17e4e36c2e9df071925c89d67f25a2a89316cb from 2022-06-10T09:19:22+0200
-- Using Git version tag: 21.12-108-g5c17e4e3 on master at 202206100919
-- IPv6 support enabled.
-- Threads support will be compiled.
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR)
-- OpenSSL development files not found, TLS won't be possible.
-- RTL-SDR device input will be compiled. Found version 0.6.0
-- libusb-1.0 error messages are available. Found version 1.0.23
-- Could NOT find SoapySDR (missing: SoapySDR_DIR)
-- SoapySDR development files not found, SoapySDR device input won't be possible.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/01_src/rtl_433/build
[  1%] Built target data
[ 92%] Built target r_433
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/librtlsdr.so', needed by 'src/rtl_433'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:1012: src/CMakeFiles/rtl_433.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

`librtlsdr.so` is here:

```
/usr/local/lib/librtlsdr.a
/usr/local/lib/librtlsdr.so
/usr/local/lib/librtlsdr.so.0
/usr/local/lib/librtlsdr.so.0.8git
/usr/local/lib/pkgconfig/librtlsdr.pc
```

`pkconfig` should be searching this path:

```
ubuntu@ubuntu:~/01_src/rtl_433/build$ pkg-config --variable pc_path pkg-config
/usr/local/lib/aarch64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/aarch64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig
```

What am I doing wrong?

zuckschwerdt, 2022-06-12:
What does this report?
`PKG_CONFIG_PATH=/usr/local/lib/pkgconfig pkg-config --libs librtlsdr`

And why are you setting `PKG_CONFIG_PATH`? Isn't `/usr/local` already a default pkg-config path?

Sineos, 2022-06-12:
Thanks for your reply.

```
ubuntu@ubuntu:~/01_src/rtl_433/build$ PKG_CONFIG_PATH=/usr/local/lib/pkgconfig pkg-config --libs librtlsdr
-L/usr/local/lib -lrtlsdr -lusb-1.0
```

> And why are you setting `PKG_CONFIG_PATH`? Isn't `/usr/local` already a default pkg-config path?

Was an attempt to get rid of the error. Happens without as well.

zuckschwerdt, 2022-06-12:
Can you show the entries from the CMake cache? `grep LibRTLSDR_ CMakeCache.txt`

Sineos, 2022-06-12:
This did the trick. It had old entries. Removing it and building worked. Many thanks, really appreciated.

I was running `make clean` before but that apparently does not remove the `CMakeCache.txt`. A `distclean` target seems not existing.

closed 2022-06-12 by Sineos