merbanan/rtl_433 #2092: Skylink HA-434TL motion sensor - identifying specific sensor

cgmckeever, 2022-06-14:
Just started playing with the `rtl_433` library and trying to figure out what Im seeing with my 1 `Skylink HA-434TL motion sensor`
It seems like the ID is always changing. Is that the the expected behaviour of this particular device? 

I only have 1, but I guess I could be picking up a rogue sensor from a neighbor. Or am I just not interpreting these packets correctly?

```
model: "Skylink HA-434TL motion sensor"
motion: "true"
id: "04000"
raw: "a4000"
protocol: "Skylink HA-434TL motion sensor"
rssi: -70
duration: 194022
```

```
model: "Skylink HA-434TL motion sensor"
motion: "false"
id: "0c000"
raw: "4c000"
protocol: "Skylink HA-434TL motion sensor"
rssi: -71
duration: 161025
```

```
model: "Skylink HA-434TL motion sensor"
motion: "false"
id: "00110"
raw: "40110"
protocol: "Skylink HA-434TL motion sensor"
rssi: -71
duration: 76024
```

```
model: "Skylink HA-434TL motion sensor"
motion: "false"
id: "08000"
raw: "48000"
protocol: "Skylink HA-434TL motion sensor"
rssi: -70
duration: 342028
```

zuckschwerdt, 2022-06-14:
That looks like random noise, not data. Capture and analyze a packet, then build a flex decoder first. https://triq.org/rtl_433/ANALYZE.html

cgmckeever, 2022-06-15:
thanks .. will unravel the noise! Appreciate the tip!

closed 2022-06-15 by cgmckeever