merbanan/rtl_433 #2093: Davis Vantage Pro 2

MidwestTechnologies, 2022-06-17:
Is there any interest in adding the Davis Vantage Pro 2 or is rtldavis still recommended?  

I have samples and several successful decodes although the frequency hopping may pose a problem.  I just need someone to help me push this over the goal line.

Please let me know.

zuckschwerdt, 2022-06-18:
It's a weather station, so rtl_433 seems like a fit. How wide is the hopping and do you know the timing or protocol parameters (modulation, coding)?

zuckschwerdt, 2022-06-18:
Oh, I just browsed rtldavis. The hopping is ~500k steps from 902M to 927M? We would only be able to monitor a single transmission out of every ~50 then :/

gdt, 2022-06-18:
The point of rtldavis is to synchronize to the hopping pattern and follow it.  The point of the Davis hopping is, I think, robustness against  persistent interference at any one frequency.    So implementing the Davis protocol properly would require a decoder to be able to sync and then give hopping commands.  (That's perhaps not a crazy thing to do; I had contemplated tracking repeat intervals for N devices on M frequencies and trying to catch most of them.  But now I am running 3 RTL-SDR dongles and not dealing with the issue at all.)

However, it probably makes sense to decode the transmissions, and then transmissions that happen to be within the rx bandwidth would get decoded.   I think it's every 2.5s so getting one out of 50 might still be quite useful.  (Although I think they are different with T in each and rain less often.)

Given rtldavis, though, I'm not sure it's worthwhile and I haven't been inclined do to the work.

MidwestTechnologies, 2022-06-18:
Well I pretty much figured as much on the hopping.  I appreciate the feedback.  This was a good exercise for me as this is the first sensor I have attempted.  I’ve gotten familiar with the tools on triq.org so all is not lost.

Wind speed and wind direction are sent with every packet.  The other sensors are sent one packet at a time with temperature, humidity, and rain being more frequent.  I can’t make heads or tails of the humidity information that I’m decoding.

The hops are about 2.5625 seconds dwell time if I remember correctly.  I attempted 2 or 3 seconds with rtl_433 and got the first 2 packets in the sequence but eventually things do get out of sync.

Anyway, if you still want to give it a go I can post what I have so far.  I thought it would be wise to see if there was a need for it before I dig any deeper.

gdt, 2022-07-05:
@MidwestTechnologies not sure what you want to do.    I more or less personally think that Davis VP2 is a bit outside what rtl_433 aims at, because it requires dedicating a hopping radio, but it's not out of the question.   And then there is rtldavis which I hear works.   If you want to submit a draft PR and ask on the mailinglist for others to test, that sounds fine, and if you don't want to spend the effort I totally understand.   I'm not quite clear on rtl_433 issue tracker culture, but my view is that there are a lot of open issues and this one staying open after resolving the discussion isn't that useful, as I really don't expect anyone to dig in from the issue.  If you see it that way you could hit close (I don't want to close your issue unilaterally).

MidwestTechnologies, 2022-07-05:
@gdt Thanks for your input.  It’s frustrating that Davis doesn’t tx all information in each packet.  I understand the benefits of frequency hopping and why they would use it but it would have been ideal for rtl_433 to send all data each packet.  And I suppose real estate in each packet is limited with checksums and the like.  rtldavis doesn’t work well for me.  I see data maybe once every 2 minutes.  I think we should close this thread though.  I really set out to decode the Orion Badger water meter but was distracted by all of these pesky Vantage Pro 2 packets.  I’m not seeing any Badger packets so my meter must be using cellular mode only.  Anyway, thanks for the help!

closed 2022-07-05 by MidwestTechnologies

gdt, 2022-07-05:
Davis has designed for their system and with their console it works very well.  It is a design decision in rtl_433 to just decode packets and be stateless.  That's fair and a useful separation, but I'd say that most real use leads one to want to have a stateful post-processor.   I will give rtldavis a try sometime.  You may just not have a good enough signal.