merbanan/rtl_433 #2094: Acurite 5n1 Wind Speed decodes to high value

n7qnm, 2022-06-18:
In rtl_433 version 21.05-143-g4ca3cb0d branch master at 202111171123 inputs file rtl_tcp RTL-SDR with TLS, packets from my Acurite 5n1 are occasionally (1 or 2 times in 24 hours) decoding outrageous values for wind_avg_km_h  - Here's a snip from a log

{"time" : "2022-06-18 12:25:06", "model" : "Acurite-5n1", "message_type" : 49, "id" : 1225, "channel" : "A", "sequence_num" : 0, "battery_ok" : 1, "wind_avg_km_h" : 5.139, "wind_dir_deg" : 315.000, "rain_in" : 27.810, "mic" : "CHECKSUM"}
{"time" : "2022-06-18 12:25:06", "model" : "Acurite-5n1", "message_type" : 49, "id" : 1225, "channel" : "A", "sequence_num" : 1, "battery_ok" : 1, "wind_avg_km_h" : 5.139, "wind_dir_deg" : 315.000, "rain_in" : 27.810, "mic" : "CHECKSUM"}
{"time" : "2022-06-18 12:25:06", "model" : "Acurite-5n1", "message_type" : 49, "id" : 1225, "channel" : "A", "sequence_num" : 2, "battery_ok" : 1, "wind_avg_km_h" : 107.786, "wind_dir_deg" : 225.000, "rain_in" : 120.000, "mic" : "CHECKSUM"}
{"time" : "2022-06-18 12:25:24", "model" : "Acurite-5n1", "message_type" : 56, "id" : 1225, "channel" : "A", "sequence_num" : 0, "battery_ok" : 1, "wind_avg_km_h" : 4.311, "temperature_F" : 51.600, "humidity" : 80, "mic" : "CHECKSUM"}

In rtl_433 version unknown inputs file rtl_tcp RTL-SDR (that I downloaded with RTL_SDR) several years ago; the same issue does NOT exist.  That version feeding my "real" weather web page, I'd rather not take it offline for 24 hours to get independent logs; but i can if needed.

zuckschwerdt, 2022-06-19:
There were some minor changes https://github.com/merbanan/rtl_433/commits/master/src/devices/acurite.c
But the wind speed wasn't change in three years https://github.com/merbanan/rtl_433/blame/master/src/devices/acurite.c#L908-L916
The value you see should be raw 129, that might have significance.

n7qnm, 2022-06-19:
OK – definitely suspicious; my 5n1 is 6 or 7 years old.

 

I’ll try backing that change out later today and will let you know.

 

Thanks for the quick response!

 

Clay

zuckschwerdt, 2022-06-19:
To be clear: the wind decoding wasn't ever changed since introduction in 520aad3 from Jan 2017.
Must be something else, like an error condition or problem in receiving / slicing the bits.
Best case would be to grab a sample that shows a bad value and then analyze that.
https://triq.org/rtl_433/ANALYZE.html#grab-a-sample

n7qnm, 2022-06-19:
OK – I looked at the old code and the new, and the code you highlighted is there and identical.

 

BUT, I did find that in the call to data_make at line 841, in the OLD code, the 14th parameter is “wind_speed_kph” and in the NEW code, it’s wind_speed.  I changed that line, did a make and am trying the new code right now.

 

I’ll keep you posted – if history holds, I should see (or not 😉) a “bad” high speed in 24 hours or less.

 

Here’s a git diff –

 

diff --git a/src/devices/acurite.c b/src/devices/acurite.c

index 8ae813ea..5ad80caf 100644

--- a/src/devices/acurite.c

+++ b/src/devices/acurite.c

@@ -841,7 +841,7 @@ static int acurite_txr_decode(r_device *decoder, bitbuffer_t *bitbuffer)

                         "channel",      NULL,   DATA_STRING,    channel_str,

                         "sequence_num",  NULL,   DATA_INT,      sequence_num,

                         "battery_ok",       "Battery",      DATA_INT,    !battery_low,

-                        "wind_avg_km_h",   "wind_speed",   DATA_FORMAT,    "%.1f km/h", DATA_DOUBLE,     wind_speed_kph,

+                        "wind_avg_km_h",   "wind_speed_kph",   DATA_FORMAT,    "%.1f km/h", DATA_DOUBLE,     wind_speed_kph,

                         "wind_dir_deg", NULL,   DATA_FORMAT,    "%.1f", DATA_DOUBLE,    wind_dir,

                         "rain_in",      "Rainfall Accumulation",   DATA_FORMAT, "%.2f in", DATA_DOUBLE, raincounter * 0.01f,

                         "mic",                  "Integrity",    DATA_STRING, "CHECKSUM",

 

What do you think?

 

Thanks!

Clay

n7qnm, 2022-06-20:
Just did a bit more research – even tried moving the old RTL to the new system and tried it.   No joy.   

 

So, I looked at the radio/antenna setup.  The old one (that works) has a relatively expensive radio with reasonable specs (.5PPM accuracy, TXCO).  The NEW one is a cheaper model, 1PPM decode, no claim of a TXCO).   So, I’m ordered the next gen of the .5PPM/TXCO one and we’ll see what happens.

 

No offense intended, but I was really hoping on was a software issue 😊.

 

Thanks again for your response, research and pointers!

 

Clay

N7QNM

 

https://triq.org/rtl_433/ANALYZE.html#grab-a-sample

zuckschwerdt, 2022-06-21:
Yes, that might be part of the trouble: the signal is degraded but somehow gets the checksum right.

closed 2024-06-04 by gdt