merbanan/rtl_433 #2096: Bresser 5-in-1 / 6-in-1 issues

mczakk, 2022-06-23:
Hi, I have two bresser stations, both of which were purchased as 5-in-1 models and are the same model.
One is in a forest 2 hours away so i cant check the model number but am sure that they are the same.
This unit is being decoded by rtl_433 (probably an old build, but i cant check the .config file right now) and reports the following via mqtt
{"time":"2022-06-23 11:57:11","model":"Bresser-5in1","id":132,"battery_ok":1,"temperature_C":21.6,"humidity":63,"wind_max_m_s":4.0,"wind_avg_m_s":3.6,"wind_dir_deg":315.0,"rain_mm":2.4,"mic":"CHECKSUM"}

The new unit is being decoded by the latest version of rtl_433 and does not find any messages when using protocol 119
protocol 172 however gives a confusing mix of messages, not the alternating 12 and 24 second periods as mentioned in other posts, although it did do that when i first tried it!
attached is the terminal output and the config file
[breser data.txt](https://github.com/merbanan/rtl_433/files/8966750/breser.data.txt)
[rtl_433-conf.txt](https://github.com/merbanan/rtl_433/files/8966757/rtl_433-conf.txt)
 here is a snippet of the output file:
time      : 2022-06-23 12:04:35
model     : Bresser-6in1 id        : 38053505
channel   : 0            Sensor type: 1            Wind Gust : 0.6 m/s       Wind Speed: 0.6 m/s
Direction : 270          Rain      : 0.8 mm        Flags     : 1             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-23 12:04:47
model     : Bresser-6in1 id        : 38053505
channel   : 0            Battery   : 1             Temperature: 25.1 C       Humidity  : 52
Sensor type: 1           Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 270
UV        : 0.0          Flags     : 0             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-23 12:05:59
model     : Bresser-6in1 id        : 38053505
channel   : 0            Battery   : 1             Temperature: 25.1 C       Humidity  : 51
Sensor type: 1           Wind Gust : 0.8 m/s       Wind Speed: 0.8 m/s       Direction : 300
UV        : 0.0          Flags     : 0             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-23 12:06:23
model     : Bresser-6in1 id        : 38053505
channel   : 0            Battery   : 1             Temperature: 25.1 C       Humidity  : 52
Sensor type: 1           Wind Gust : 1.1 m/s       Wind Speed: 1.1 m/s       Direction : 300
UV        : 0.0          Flags     : 0             Integrity : CRC

as you can see the unit reports as Bresser-6-in-1, and has a value for UV, but as far as i know the unit does not have a uv sensor

Is there any way to get the old decoder (protocol 119) working again?

MacH-21, 2022-06-26:
Hi mczakk,

1. Could you give a link to your weather station user manual.
2. Does your weather station have a solar panel?
3. Could you try this command for a couple of minutes
 
rtl_433 -f 868300000 -s 1M -X "n=Bresser/mczakk Weather Center 5-in-1,m=FSK_PCM,s=120,l=120,r=2500"

should look like this:

time      : 2022-06-26 16:22:45
model     : Bresser/Youshiko Weather Center 6-in-1 count     : 1             num_rows  : 1             rows      :
len       : 204          data      : 55555555545ba9cae83220001e31f76ff842b1fffffffe0317f
codes     : {204}55555555545ba9cae83220001e31f76ff842b1fffffffe0317f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-26 16:22:45
model     : Bresser-6in1 id        : 1910000f
channel   : 0            Battery   : 1             Sensor type: 1            Wind Gust : 4.4 m/s       Wind Speed: 3.8 m/s
Direction : 215          Rain      : 0.0 mm        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-26 16:22:57
model     : Bresser/Youshiko Weather Center 6-in-1 count     : 1             num_rows  : 1             rows      :
len       : 204          data      : 55555555545ba96dd43220001e31f76df833302c68c7ffe16bf
codes     : {204}55555555545ba96dd43220001e31f76df833302c68c7ffe16bf
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-26 16:22:57
model     : Bresser-6in1 id        : 1910000f
channel   : 0            Battery   : 1             Temperature: 16.3 C       Humidity  : 63            Sensor type: 1            Wind Gust : 4.4 m/s
Wind Speed: 3.9 m/s      Direction : 199           Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-26 16:23:09
model     : Bresser/Youshiko Weather Center 6-in-1 count     : 1             num_rows  : 1             rows      :
len       : 204          data      : 55555555545ba901c23220001e31f6f5f64111fffffffe03340
codes     : {204}55555555545ba901c23220001e31f6f5f64111fffffffe03340
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-06-26 16:23:09
model     : Bresser-6in1 id        : 1910000f
channel   : 0            Battery   : 1             Sensor type: 1            Wind Gust : 4.8 m/s       Wind Speed: 4.5 m/s
Direction : 208          Rain      : 0.0 mm        Integrity : CRC

mczakk, 2022-06-27:
Could you give a link to your weather station user manual. - no, i have a paper copy of it! but the both units have the same model number - 7002510 one was purchased last may, the other last week
Does your weather station have a solar panel? - no neither of them do
Could you try this command for a couple of minutes

OK, I tried that and got the following:
[test1.txt](https://github.com/merbanan/rtl_433/files/8992021/test1.txt)
If I use a slightly different frequency (868.27)I get more stable results, as shown here:
[test2.txt](https://github.com/merbanan/rtl_433/files/8992030/test2.txt)
To amend my OP, the old unit is running on the latest build of rtl_433, as I only installed it on it's pi last month!

MacH-21, 2022-06-28:
I think you are on the right track.
I use SDR Console to see the frequencies (my system is windows 10 x64)
See attached screen shot-
[image]

You can see two frequencies 868.23M and 868.35M product manual says 868.3M
I can select either frequency but tend to go middle for diddle
rtl_433-rtlsdr  -f 868230000 -s 1M -X "n=Bresser 6-in-1,m=FSK_PCM,s=120,l=120,r=2500,preamble=0x2dd4" -R 172
rtl_433-rtlsdr  -f 868350000 -s 1M -X "n=Bresser 6-in-1,m=FSK_PCM,s=120,l=120,r=2500,preamble=0x2dd4" -R 172

rtl_433-rtlsdr  -f 868290000 -s 1M -X "n=Bresser 6-in-1,m=FSK_PCM,s=120,l=120,r=2500,preamble=0x2dd4" -R 172

You should receive data every 12 seconds Bresser data usually consists of
aaaaaaaaaa 2dd4 475e1910000f18fdbafe2158ffffffff0184  (5 bytes preamble, 2 bytes sync, 18 bytes data) = 25 bytes total
I like to filter on the sync bytes, 2dd4 seems to be solid so far!

[MyTest4_mczakk.txt](https://github.com/merbanan/rtl_433/files/8998077/MyTest4_mczakk.txt)
As you can see in my txt file I have a sample every 12 seconds no errors.

Your last test with 868.27M must be pretty close, still a few errors, maybe try 
rtl_433-rtlsdr  -f 868270000 -s 1M -X "n=Bresser 6-in-1,m=FSK_PCM,s=120,l=120,r=2500,preamble=0x2dd4" -R 172
may filter out the junk

Good luck, let me know how you get on.

mczakk, 2022-06-28:
I've been running your command and getting some interesting results, though I'm not sure how the command works with my config file. Is there a way to enter the -X statement in the config? I ask, as ultimately I will need to have the ability to jsut plug the pi in and have it all run automatically.
I've attached my current config file, which i'm running with the command rtl_433 (no additions) This is giving pretty stable results, but does occasionally report a-b-a-a-b-a-b (if a is the short message and b the long one) and also a couple of gqrx screenshots, with a time period in the waterfall of ~80 seconds, as that was how long it took to fill the screen. 
Next step is to get another rtl2838 and see if i can match up the recieved messages with the waterfall, such a pain that you cant run gqrx and rtl_433 at the same time with only one rtl2838 - hopefully it will work!
config file:
[rtl_433.txt](https://github.com/merbanan/rtl_433/files/9001868/rtl_433.txt)
screenshot 1
[image]
screenshot 2

[image]

mczakk, 2022-06-29:
I ran <rtl-433-rtlsdr -f 868.26M -S unknown -T 120> and got a csv which matched the kv output: 
[image]
one thing is different - the id value, any idea why?

I also ran again and produced a .cu8 file, but even compressed it is too big to upload here.
I have put it here : https://www.transfernow.net/dl/20220629EPq9TvCV I think it will be available there for 7 days

I also had a converstaion with one of the technical guuys at Bresser UK. He told me that there was a change in the boards put into the 5-in-1 sensors due to the semiconductor shortage at the moment. All new 5-in-1 will have this new board which has a different reporting regime. the change point is reflected in the lot number found on the bottom of the unit. anything with a lot number higher than 1156742 will have the new board in.

MacH-21, 2022-06-30:
Hi,
 Interesting re-bresserUK,
 
 Re; the ID: change rtl_433 reports the correct ID, MQTT Event displays incorrect ID.
 You seem to have had this from day one your OP states
 via mqtt
{"time":"2022-06-23 11:57:11","model":"Bresser-5in1","id":132,"battery_ok":1,"temperature_C":21.6,"humidity":63,"wind_max_m_s":4.0,"wind_avg_m_s":3.6,"wind_dir_deg":315.0,"rain_mm":2.4,"mic":"CHECKSUM"}
ID-132 is odd as usually Bresser ID = 4 bytes as your 38,05,35,05.

and your  test2.txt
time      : 2022-06-23 13:52:59
model     : Bresser-6in1 id        : 38053505
channel   : 0            Sensor type: 1            Wind Gust : 1.0 m/s       Wind Speed: 1.0 m/s
Direction : 144          Rain      : 0.8 mm        Startup   : 1             Flags     : 1             Integrity : CRC
{"time" : "2022-06-23 13:52:59", "model" : "Bresser-6in1", "id" : 939865349, "channel" : 0, "sensor_type" : 1, "wind_max_m_s" : 1.000, "wind_avg_m_s" : 1.000, "wind_dir_deg" : 144, "rain_mm" : 0.800, "startup" : 1, "flags" : 1, "mic" : "CRC"}
expand_topic: unknown token "data"
testpi@raspberrypi:~/rtl_433 $ rtl_433 -F json  -F mqtt://localhost,retain=0,events=bresser

Have you ever had 5 weather station signals per minute (1 every 12 Seconds).

I do not use mqtt at present (will be soon). 
I use an ESP32 + Si4463 transceiver (bit cheaper than PI and SDR).

I have changed rtl_433.conf to stop using mqtt and also adjusting the levels Lines  106,110,114,203,205,206,510 all have ###.
Also 3 x frequencies lins 57,58,59 which you could test and see if any run stable, giving you a result every 12 seconds.
[Test-rtl_433.conf.txt](https://github.com/merbanan/rtl_433/files/9022986/Test-rtl_433.conf.txt)

MacH-21, 2022-07-05:
HI,
one thing is different - the id value, any idea why?

 ID value 939865349 is Your station in Decimal format  -  ID (38053505)hex
[image]
.

gdt, 2023-09-26:
Please retest with current git  master.  A lot has been improved.

closed 2024-06-04 by gdt