merbanan/rtl_433 #2097: "time" field invisible

gvanem, 2022-06-25:
My shell (4NT from [jpsoft.com](jpsoft.com)) uses a blue background. 
So when using the `-M time` parameter the console looks like:
[image]

invisible! So after this patch:
```diff
--- a/src/output_file.c 2022-05-16 12:21:09
+++ b/src/output_file.c 2022-06-25 15:01:07
@@ -155,7 +155,7 @@
     if (!key || !*key)
         return TERM_COLOR_RESET;
     if (!strcmp(key, "tag") || !strcmp(key, "time"))
-        return TERM_COLOR_BLUE;
+        return TERM_COLOR_BRIGHT_WHITE;
     if (!strcmp(key, "model") || !strcmp(key, "type") || !strcmp(key, "id"))
         return TERM_COLOR_RED;
     if (!strcmp(key, "mic"))
```

it looks OK for me:

[image]

I think many people have other background shell colours than plain black. I have yet to see someone with bright-white.
So the above patch should be pleasing for most.

zuckschwerdt, 2022-06-25:
Bright (9x codes) doesn't seem to work in my terminals. The default white isn't great on a bright background (we should consider bright and dark bg's, and also blue then).

[image]

Maybe we need to add the `bold` attribute for this?

gvanem, 2022-06-25:
Or a reverse;`TERM_COLOR_BRIGHT_BLUE_ON_WHITE`. Possible?

gdt, 2022-07-03:
I would expect there is prior art on how to deal with this (besides the curmudgeon approach of not using colors).   rtl_433 doesn't seem particularly special in doing this sort of thing.

gdt, 2023-09-29:
I'm inclined to close this; it seems intractable and if someone wants to create a PR that adds some general approach that's fine.  This could be support for multiple palettes, or a config file for colors.     But I can't see us tweaking colors to fit particular people's setups, and this has been idle for a long time.

(I can also see an easy way to disable colorization.  This is proof that's needed, but that's for another issue.)

closed 2023-09-29 by gdt