merbanan/rtl_433 #2098: writing gibberish to json at high-speed

suciokhan, 2022-06-30:
When writing results to a json file, the json file fills up at lightning speed (~1MB per second) even though there don't appear to be any active signals, and the file gets filled with what appear to be random characters, rather than decoding results.

### Recreation

`rtl_433 -w test_output.json -F json`

[image]

What could be causing this? If I simply run `rtl_433` I don't see any outputs on the command line. This is being remotely executed via ssh on an RPI 4 8GB, if that makes a difference.

gdt, 2022-06-30:
`-w` outputs the raw data, not the json, so that is likely behaving as documented.   (You'll get help debugging startup on the mailinglist.)

gdt, 2022-06-30:
Please close your bug then.