merbanan/rtl_433 #2099: WH45 Combi Sensor

JaraLowell, 2022-07-01:
I own a DP250 (aka as WH45 from Ecowitt) the 5-in-1 Air Quality Sensor, with PM2.5 / PM10 / CO2 / Temperature / Humidity Detection Sensor. Seems tough rtl_433 does not see this one. So was wondering if there anything I can do to help for this to be added.

JaraLowell, 2022-07-01:
```
time      : 2022-07-01 04:39:35
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 141          data      : 555555555545ba8ae81e266087e4f71ba810
codes     : {141}555555555545ba8ae81e266087e4f71ba810

time      : 2022-07-01 04:39:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 163          data      : 00002aaaaaaaaaaa8b75381ca18ed3c3e08000000
codes     : {163}00002aaaaaaaaaaa8b75381ca18ed3c3e08000000

time      : 2022-07-01 04:39:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 49           data      : 0000000000000
codes     : {49}0000000000000

time      : 2022-07-01 04:39:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 181          data      : ffff800015555555555545ba9c0e50c764f0f820000000
codes     : {181}ffff800015555555555545ba9c0e50c764f0f820000000

time      : 2022-07-01 04:39:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 26           data      : 0000000
codes     : {26}0000000

time      : 2022-07-01 04:39:44
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 188          data      : 555555555545ba88a006ea005287086e9075a04238eb681
codes     : {188}555555555545ba88a006ea005287086e9075a04238eb681

time      : 2022-07-01 04:40:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 162          data      : 00002aaaaaaaaaaa8b75381ca18ec9e2b10000000
codes     : {162}00002aaaaaaaaaaa8b75381ca18ec9e2b10000000

time      : 2022-07-01 04:40:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 26           data      : 0000000
codes     : {26}0000000

time      : 2022-07-01 04:40:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 26           data      : 0000000
codes     : {26}0000000

time      : 2022-07-01 04:40:38
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 193          data      : 0001fffe000055555555555516ea7039431d93c5620000000
codes     : {193}0001fffe000055555555555516ea7039431d93c5620000000

time      : 2022-07-01 04:40:54
model     : DP250        count     : 1             num_rows  : 1             rows      :
len       : 139          data      : 555555555545ba8ae81e266087e4f71ba82
codes     : {139}555555555545ba8ae81e266087e4f71ba82
```
Also note that for bat it can return 6 for DC imput rather then battery use.

zuckschwerdt, 2022-07-01:
What timings did you grab the codes with?
Here is [a BitBench with that data](https://triq.net/bitbench#c=%7B141%7D555555555545ba8ae81e266087e4f71ba810&c=%20%7B163%7D00002aaaaaaaaaaa8b75381ca18ed3c3e08000000&c=%20%7B181%7Dffff800015555555555545ba9c0e50c764f0f820000000&c=%20%7B188%7D555555555545ba88a006ea005287086e9075a04238eb681&c=%20%7B162%7D00002aaaaaaaaaaa8b75381ca18ec9e2b10000000&c=%20%7B193%7D0001fffe000055555555555516ea7039431d93c5620000000&c=%20%7B139%7D555555555545ba8ae81e266087e4f71ba82&f=8h8h%20&a=Preamble&m=aa2dd4&cw=4). It looks like different (at least two) devices?

zuckschwerdt, 2022-07-01:
If you can grab some samples https://triq.org/rtl_433/ANALYZE.html and upload here as zip.
Preferably, as mentioned without antenna so you only get the DP250.

JaraLowell, 2022-07-01:
From the Manual is shows,.
```PM2.5 0 ~ 999 ug/m3
pm10 0 ~999 ug/m3
co² 0 ~ 40000 ppm
temp -40 ~ 60 °C
humidity 0 ~ 99%```
Plus bat wish i presume can be 0 ~ 3 and 6 for DC

anthyz, 2022-07-01:
What a coincidence. I have a WH45 and have been working on decoding it. Here's what I have:
`Packet layout:

     0  1  2  3  4  5  6  7  8  9 10 11 12 13 14
    YY II II II 0T TT HH Bp pp BP PP CC CC XX AA
    45 00 36 60 02 7e 36 40 23 00 29 02 29 07 4f

- Y: 8 bit fixed sensor type 0x45
- I: 24 bit device ID
- T: 11 bit temperature, offset 40, scale 10
- H: 8 bit humidity
- B: 1 bit MSB of battery bars out of 5 (a value of 6 indicates external power via USB)
- p: 14 bit PM2.5 reading in ug/m3 * 10
- B: 2 bits LSBs of battery bars out of 5
- P: 14 bit PM10 reading in ug/m3 * 10
- C: 16 bit CO2 reading in ppm
- X: 8 bit CRC
- A: 8 bit checksum`

I can prepare a PR if that makes things easier.

gdt, 2022-07-01:
I don't have commit access, but if you can make a PR, that's great because not only is it code that could be merged, but it is code that other people can run. 

I have a WH41, which is PM2.5/PM10 only.  However, the PM10 is faux; it's a formula from PM2.5.  From memory the PM2.5/PM10 encoding looks famiiar.

anthyz, 2022-07-01:
Okay, a PR it is.

Since many Fineoffset devices share the same timings what is the preferred approach for adding a new device? Add it in fineoffset.c as part of an existing protocol #?  Or add it as a new protocol either in fineoffset.c or its own file? A new protocol seems best to me, but I thought I'd ask first. I'll add that I'm also decoding two other Fineoffset devices, WN34 (temperature probe) and WH55 (leak detector), and could add those later.

zuckschwerdt, 2022-07-01:
Running a device decoder is cheap, create a new file with a fresh decoder.

Having multiple decoders in one source file is reserved for protocols that share e.g. a custom coding or checksum.
Having multiple devices in one decoder is reserved for protocols tha that can't easily distinguish the devices.

gdt, 2022-07-01:
However, I suspect there is significant shared decoding with this and the WH41.  A similar situation is acurite which has various sensors with similar encodings.   So do think about if you have to cut and paste or are reimplementing the same thing.

zuckschwerdt, 2022-07-01:
Yes, I was talking about *nested* decoders like https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c#L474-L478
We should avoid that if possible. Ideally WH25, WH24, WH0290 would be three individual decoders.

anthyz, 2022-07-01:
The data layout is similar between WH41 and WH45 in that the battery bits and PM2.5/PM10 are encoded the same way, but they are at different offsets. Due to a bigger ID and the addition of temperature, humidity, and CO2, the only data that is aligned between the two is the family code :-).

gdt, 2022-07-01:
My point is really only that the code to unpack battery and PM2.5/PM10 should exist  once and not be duplicated.   That could well be a function called by both with the multiple words and address of variables.

zuckschwerdt, 2022-07-02:
Yes, for complex situations: DRY ;)

It's just a few bits here, no worries: https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c#L415-L416

closed 2022-07-05 by gdt

gdt, 2022-07-27:
@JaraLowell @anthyz Thank you both very much for your efforts.   I just got a WH45 today and plugged it in, and (after remembering to restart rtl_433 after building the new code) saw nice json showing up in MQTT.    A few minutes of hand-writing HA mqtt sensors later, and I have data  being displayed!

gdt, 2022-07-27:
And if you are listening Ecowitt: I bought it because it worked with rtl_433.  And thanks for answering questions about the WH0290 earlier in email.

anthyz, 2022-07-28:
Awesome! Glad it's working for you!