merbanan/rtl_433 #2100: Tuning to other frequency fails

kord1e, 2022-07-02:
I tried to monitor the 868 MHz range but when I tried to tune to 868.4 MHz, it didn't work. It still tuned to 433.92 MHz. Here's the console output:

```
$ rtl_433 -v -d 0 -g 42 -f 868400000 -s 2400000
rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/user/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Reading conf from "/etc/rtl_433/rtl_433.conf".

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 185 out of 217 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
[R82XX] PLL not locked!
Sample rate set to 2400000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to 42.100000 dB.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 2400000 Hz at cutoff 240000 Hz, 4.2 us
```

pswiatki, 2022-07-02:
Question is: did it really tuned to 433.92 MHz, or just faked this status but in fact tuned to your desired frequency(?)

kord1e, 2022-07-02:
@pswiatki 
I believe it tuned to 433.92 MHz because I can see some devices that transmit at about 433 MHz.

Looks like hopping to other frequencies works but it's also inconsistent. I can see several issues:

- it tunes to 433.92 MHz even when it's not listed in the frequencies
- there is a 600-second hop interval even when it's not included in the options
- hopping is needed to tune to other frequencies in the first place

```
$ rtl_433 -v -d 0 -g 42 -H 120 -f 868400000 -H 120 -f 868500000 -H 120 -s 2400000 -E hop
rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/user/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Reading conf from "/etc/rtl_433/rtl_433.conf".

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 185 out of 217 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
[R82XX] PLL not locked!
Sample rate set to 2400000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to 42.100000 dB.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 2400000 Hz at cutoff 240000 Hz, 4.2 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-02 18:47:41
<<<<<<<<<some devices were found here>>>>>>>>>>>>>>>>>
Tuned to 868.400MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 868400000                        frequencies: 433920000, 868400000, 868500000            hop_times : 600, 120, 120, 120
Allocating 15 zero-copy buffers
Tuned to 868.500MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 868500000                        frequencies: 433920000, 868400000, 868500000            hop_times : 600, 120, 120, 120
Allocating 15 zero-copy buffers
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 433920000                        frequencies: 433920000, 868400000, 868500000            hop_times : 600, 120, 120, 120
Allocating 15 zero-copy buffers
```

pswiatki, 2022-07-02:
I see. Interesting.

zuckschwerdt, 2022-07-02:
> Reading conf from "/etc/rtl_433/rtl_433.conf".

Check the contents of your config.

kord1e, 2022-07-02:
These options (in addition to the decoders) were not commented out. But shouldn't these be just the defaults that are overridden by the command line options? That's what I would expect.

```
gain          0
frequency     433.92M
hop_interval  600
ppm_error     0
sample_rate   250k

```

zuckschwerdt, 2022-07-02:
All config settings (you can have multiple files, also includes) and the command line stack.

But good point, `frequency` and `hop_interval` allow multiple values for hopping and that's confusing. We should remove that from the example config, likely.

pswiatki, 2022-07-02:
> All config settings (you can have multiple files, also includes) and the command line stack.
Hmmm..... Would be good to know the order of stacking, I think. Id est, what overwrites what.

zuckschwerdt, 2022-07-02:
Strictly left to right (or top to bottom in files). If there is a default config then that's first.

gdt, 2023-10-14:
This seems like an answered question, and thus closable.   Specific issue mentioned in #2670.

closed 2023-10-14 by gdt