merbanan/rtl_433 #2103: New device ELV combo sensor KS300

JanSchmidt3, 2022-07-04:
I own a combo sensor which is currently not supported.

For a picture see 
[https://de.elv.com/elv-funk-kombi-wettersensor-ks-300-inkl-hochwertigem-2-m-edelstahlmast-und-qualitaetsbatterien-065390](url)

It measures wind speed, temperature, humidity, rain (with additional real time marker).

I recorded some sample like

rtl_433 -f 868.3M -S unknown

and got as output

Detected OOK package    @0.201057s
Analyzing pulses...
Total count:   91,  width: 110.23 ms            (110229 S)
Pulse width distribution:
 [ 0] count:   49,  width:  846 us [844;849]    ( 846 S)
 [ 1] count:   42,  width:  358 us [356;361]    ( 358 S)
Gap width distribution:
 [ 0] count:   49,  width:  373 us [372;376]    ( 373 S)
 [ 1] count:   41,  width:  862 us [860;865]    ( 862 S)
Pulse period distribution:
 [ 0] count:   90,  width: 1220 us [1218;1224]  (1220 S)
Pulse timing distribution:
 [ 0] count:   90,  width:  853 us [844;865]    ( 853 S)
 [ 1] count:   91,  width:  366 us [356;376]    ( 366 S)
 [ 2] count:    1,  width: 10001 us [10001;10001]       (10001 S)
Level estimates [high, low]:  15944,    419
RSSI: -0.1 dB SNR: 15.8 dB Noise: -15.9 dB
Frequency offsets [F1, F2]:   -3880,      0     (-59.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
view at https://triq.org/pdv/#AAB1030355016E27118181818181818181818190909090819090818181908190908190908190819081908181909090908190818190819081818181908181818190818181819090819081908190818190908190909081909081908181819090909090819255
Attempting demodulation... short_width: 358, long_width: 846, reset_limit: 866, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=358,l=846,r=866,g=0,t=195,y=0'
pulse_slicer_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {91} 00 3d 8b 6a 9e 94 21 0d 53 76 8f a0 
>

I could reproduce the flex decoder
rtl_433 -s 1000k -X 'n=ELV KS300,m=OOK_PWM,s=358,l=846,r=866,g=0,t=195,y=0' g044_868.3M_1000k.cu8 

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.201057s
model     : ELV KS300    count     : 1             num_rows  : 1
rows      : 
len       : 91           data      : 003d8b6a9e94210d53768fa
codes     : {91}003d8b6a9e94210d53768fa

>

The corresponding decoding is
https://triq.net/bitbench#c=%7B91%7D003D8B6A9E94210D53768FA%5BTYPE%3DCombo%20Sensor%2C%20RAINING%3DNo%2C%20TEMP%3D%2B25.6%20C%2C%20HUMIDITY%3D47%25%2C%20WIND%3D0%20km%2Fh%2C%20TOTAL%20RAIN%20COUNT%3D3365%5D&f=TYPE%3A%5E4h%20x%20x%20IS_RAINING%3Ab%20x%20TEMP_IS_NEG%3Ab%20x%20TEMP_BCD_TENTHS%3A%5E4h%20x%20TEMP_BCD_ONES%3A%5E4h%20x%20TEMP_BCD_TENS%3A%5E4h%20x%20HUMIDITY_BCD_ONES%3A%5E4h%20x%20HUMIDITY_BCD_TENS%3A%5E4h%20x%20WIND_BCD_TENTHS%3A%5E4h%20x%20WIND_BCD_ONES%3A%5E4h%20x%20WIND_BCD_TENS%3A%5E4h%20x%20RAIN_COUNT_LSN%3A%5E4h%20x%20RAIN_COUNT_MID%3A%5E4h%20x%20RAIN_COUNT_MSN%3A%5E4h%20x%20UNKNOWN%3A%5E4h%20x%20CHK_XOR%3A%5E4h%20x%20SUM_PLUS_FIVE_LSN%3A%5E4h%20x%20&a=Preamble&m=0y00000001&cw=4

JanSchmidt3, 2022-07-04:
sample file.
[file.zip](https://github.com/merbanan/rtl_433/files/9040190/file.zip)

JanSchmidt3, 2022-07-04:
I searched at source...

I fount an existing decoder for that group of devices (No. 7 "ELV WS 2000", devices/elv.c) In principle I can add the combo sensor.

In that group 8 different sensors can be supported, two of them are already implemented. But not all deliver all values and sometimes the resolution of some values differ. That means, depending of the subtype the array of the output fields needs to be changed. Schould I copy the driver and add a new one or enlarge the output filed array?

JanSchmidt3, 2022-07-04:
this works for me (for the recorded samples)
diff --git a/src/devices/elv.c b/src/devices/elv.c
index 16fbebb9..a4e4c20c 100644
--- a/src/devices/elv.c
+++ b/src/devices/elv.c
@@ -118,11 +118,11 @@ static int ws2000_callback(r_device *decoder, bitbuffer_t *bitbuffer)
 {
     data_t *data;
     bitrow_t *bb = bitbuffer->bb;
-    uint8_t dec[13]={0};
+    uint8_t dec[16]={0};
     uint8_t nibbles=0;
     uint8_t bit=11; // preamble
-    char* types[]={"!AS3", "AS2000/ASH2000/S2000/S2001A/S2001IA/ASH2200/S300IA", "!S2000R", "!S2000W", "S2001I/S2001ID", "!S2500H", "!Pyrano", "!KS200/KS300"};
-    uint8_t length[16]={8, 8, 5, 8, 12, 9, 8, 8, 8};
+    char* types[]={"!AS3", "AS2000/ASH2000/S2000/S2001A/S2001IA/ASH2200/S300IA", "!S2000R", "!S2000W", "S2001I/S2001ID", "!S2500H", "!Pyrano", "KS200/KS300"};
+    uint8_t length[16]={8, 8, 5, 8, 12, 9, 8, 14, 8};
     uint8_t check_calculated=0, sum_calculated=0;
     uint8_t i;
     uint8_t stopbit;
@@ -169,10 +169,21 @@ static int ws2000_callback(r_device *decoder, bitbuffer_t *bitbuffer)
     int code       = dec[1] & 7;
     float temp     = ((dec[1] & 8) ? -1.0f : 1.0f) * (dec[4] * 10 + dec[3] + dec[2] * 0.1f);
     float humidity = dec[7] * 10 + dec[6] + dec[5] * 0.1f;
+    float wind     = 0;
+    int it_rains   = 0;
+    int rainsum    = 0;
     int pressure   = 0;
+    int unknown    = 0;
     if (dec[0] == 4) {
         pressure = 200 + dec[10] * 100 + dec[9] * 10 + dec[8];
     }
+    if (dec[0] == 7) {
+      it_rains = (dec[1] & 2) ? 1 : 0;
+      humidity = dec[6] * 10 + dec[5];
+      wind = dec[9] * 10 + dec[8] + dec[7] * 0.1f;
+      rainsum = (dec[12] << 8) + (dec[11] << 4) + dec[10];
+      unknown = dec[13];
+    }
 
     /* clang-format off */
     data = data_make(
@@ -182,6 +193,11 @@ static int ws2000_callback(r_device *decoder, bitbuffer_t *bitbuffer)
             "temperature_C",    "", DATA_FORMAT, "%.1f C", DATA_DOUBLE, (double)temp,
             "humidity",         "", DATA_FORMAT, "%.1f %%", DATA_DOUBLE, (double)humidity,
             "pressure_hPa",     "", DATA_FORMAT, "%d hPa", DATA_INT, pressure,
+            //KS200 / KS300
+            "wind_avg_km_h",    "", DATA_FORMAT, "%d kmh", DATA_DOUBLE, (double)wind,
+            "rain_count",       "", DATA_FORMAT, "%d", DATA_INT, rainsum,
+            "is_raining",     "", DATA_FORMAT, "%d", DATA_INT, it_rains,
+            "unknown",     "", DATA_FORMAT, "%d", DATA_INT, unknown,
             NULL);
     /* clang-format on */
 
@@ -196,6 +212,11 @@ static char *elv_ws2000_output_fields[] = {
         "temperature_C",
         "humidity",
         "pressure_hPa",
+        //KS200 / KS300
+        "wind_avg_km_h",
+        "rain_count",
+        "is_raining",
+        "unknown",
         NULL,
 };
 
@@ -206,6 +227,6 @@ r_device elv_ws2000 = {
         .long_width  = 854,  // no repetitions
         .reset_limit = 1000, // Longest pause is 854us according to link
         .decode_fn   = &ws2000_callback,
-        .disabled    = 1,
+        .disabled    = 0,
         .fields      = elv_ws2000_output_fields,
 };

zuckschwerdt, 2022-07-04:
Make a new file / copy (without the "elv_em1000" part).
Add documentation, like this PR: https://github.com/merbanan/rtl_433/pull/2101/files

JanSchmidt3, 2022-07-05:
I'm not familiar with git, so I created a diff.

```diff --git a/include/rtl_433_devices.h b/include/rtl_433_devices.h
index c692dc87..6ad8ce7f 100644
--- a/include/rtl_433_devices.h
+++ b/include/rtl_433_devices.h
@@ -226,6 +226,7 @@
     DECL(ant_antplus) \
     DECL(emos_e6016_rain) \
     DECL(hcs200_fsk) \
+    DECL(elv_ws2000_combo) \
 
     /* Add new decoders here. */
 
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index 1a5cfbcd..7ee5cdee 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -87,6 +87,7 @@ add_library(r_433 STATIC
     devices/efth800.c
     devices/elro_db286a.c
     devices/elv.c
+    devices/elv_combo.c
     devices/emontx.c
     devices/emos_e6016.c
     devices/emos_e6016_rain.c
diff --git a/src/devices/elv_combo.c b/src/devices/elv_combo.c
new file mode 100644
index 00000000..ccb4aaf5
--- /dev/null
+++ b/src/devices/elv_combo.c
@@ -0,0 +1,191 @@
+/** @file
+    ELV WS 2000.
+
+    This program is free software; you can redistribute it and/or modify
+    it under the terms of the GNU General Public License as published by
+    the Free Software Foundation; either version 2 of the License, or
+    (at your option) any later version.
+*/
+
+#include "decoder.h"
+
+static uint16_t AD_POP(uint8_t *bb, uint8_t bits, uint8_t bit)
+{
+    uint16_t val = 0;
+    uint8_t i, byte_no, bit_no;
+    for (i=0;i<bits;i++) {
+        byte_no=   (bit+i)/8 ;
+        bit_no =7-((bit+i)%8);
+        if (bb[byte_no]&(1<<bit_no)) val = val | (1<<i);
+    }
+    return val;
+}
+
+/**
+
+   based on http://www.dc3yc.privat.t-online.de/protocol.htm
+
+   - added support for comob sensor (subtyp 7)
+   - sensor 1 (Thermo/Hygro) and 4 (Thermo/Hygro/Baro) supported as well
+   - other sensors could be detected, if the length is defined correct
+     but will not receive correct values
+
+   - rain_count counts the ticks of a seesaw, the amount of water per
+     tick has to be calibrated. As shown in user manual the default is
+     295ml / m²
+
+   Protocol version V1.2
+
[... 129 lines cut ...]
+        "id",
+        "subtype",
+        "temperature_C",
+        "humidity",
+        "pressure_hPa",
+        //KS200 / KS300
+        "wind_avg_km_h",
+        "rain_count",
+	"rain_mm",
+        "is_raining",
+        "unknown",
+        NULL,
+};
+
+r_device elv_ws2000_combo = {
+        .name        = "ELV WS 2000",
+        .modulation  = OOK_PULSE_PWM,
+        .short_width = 366,  // 0 => 854us, 1 => 366us according to link in top
+        .long_width  = 854,  // no repetitions
+        .reset_limit = 1000, // Longest pause is 854us according to link
+        .decode_fn   = &ws2000_combo_callback,
+        .disabled    = 0,
+        .fields      = elv_ws2000_combo_output_fields,
+};
```

zuckschwerdt, 2022-07-05:
That's fine. I'll test and commit with your name.

This decoder is actually for the KS200/KS300 right? "WS2000" is just the name of the display panel /gateway?

JanSchmidt3, 2022-07-05:
Yes, new is the decoder for KS200/KS300. I tested with KS300.
WS2000 is one of the compatible receivers with display. I have a WS 300 PC. Sadly there is no complete overview of possible combinations.

SilentDude80, 2022-08-05:
Hi, 

I would also like to read out a "eQ-3 S 300 TH"  temperature/humidity sensor, which should be the same as the one above - I compiled and installed RTL_433 on my Raspberry Pi today, but there is no data incoming from the sensor ... is this sensor already integrated ? Do I need to supply special command line parameters ?

zuckschwerdt, 2022-08-06:
I still need to commit this decoder to rtl_433, wait a few days, sorry.

commit 40f05c70 (2022-08-06): Add support for ELV WS2000 (closes #2103)
commit b7b0f853 (2022-08-06): Add support for KS200/KS300 to ELV WS2000 (closes #2103)
closed 2022-08-06 by zuckschwerdt