merbanan/rtl_433 #2110: feature request: read data from an I Q stream for openwebrx

hubertmeier, 2022-07-13:
Hi,

it would be cool to see your work integrated in openwebrx, a very nice web-based receiver which supports many input devices and could display decoded data in the Webinterface. Openwebrx delivers I/Q-Streams to the decoder and gets back the output of the decoder to display it in the Webinterface.

rtl_433 seems not to accept I/Q-Streams, therefore your project can not be used in openwebrx...

you will find more informations:

https://github.com/jketterl/openwebrx

Thank you

Hubert

merbanan, 2022-07-13:
Well we do support rtl_tcp input. I concider that beeing streamad I/Q.

zuckschwerdt, 2022-07-13:
Good point about `rtl_tcp`. Also we have streaming input from stdin with e.g. `CU8:-` as filename (https://triq.org/rtl_433/OPERATION.html#loaders-and-dumpers) and the JSON output should be suitable.

Some `Popen` module like https://github.com/jketterl/openwebrx/blob/develop/csdr/module/drm.py should be workable.
Or even better use some networked I/O, e.g. rtl_tcp in and UDP out.

I don't have time to implement something, but let us know if anything should be improved to make integration easier.

gdt, 2023-10-02:
It sounds like this can be done with rtl_tcp or stdin, and this issue does not have a pointer to a specific protocol spec and/or library that is not implemented.  I think everybody at rtl_433 would like to see  something happen, but we need a concrete spec, and preferably that should be some semi-widely-implemented lingua franca.  Please update to be precise and actionable.

merbanan, 2026-11-08:
This is done: OpenWebRX(+) integrates rtl_433 by streaming I/Q to it on stdin, `rtl_433 -r cs16:- -s <rate> -F json` (see #3312). Also note the default FSK detection now runs all detectors in parallel (`-Y auto`) instead of choosing one by frequency, which should help the WMBus vs. other ISM devices case from #3312. Closing.

closed 2026-11-08 by merbanan