merbanan/rtl_433 #2115: influxdb output: use a single measurement

azrdev, 2022-07-20:
Currently when using the influxdb (v1) output, the data_model (i.e. model/make of the sender, according to the decoded information/protocol) is used as influx "measurement". This makes the data hard to explore in influxdb, since different measurements cannot simply be queried together -- influx assumes that all data belonging together is in a single measurement, separated by tags.

So I propose to (optionally? if you want to add/extend a cli parameter) write to a single measurement, and add data_model as a tag.

The relevant code is probably:
https://github.com/merbanan/rtl_433/blob/10891fb732e65f3c6962a72ac05dd9656f07638e/src/output_influx.c#L286-L293

zuckschwerdt, 2022-07-20:
In essence you propose to use a given fixed measurement and add the model as tag? Could be a useful case, yes.
That would the give many (and mostly empty) fields, right? It would happen when you have different sensors reporting many distinct fields.

What little I can find in the [InfluxDB docs about the role of "measurement"](https://docs.influxdata.com/influxdb/v2.3/reference/key-concepts/data-elements/#measurement) I seem to understand that a measurement can indeed have a varied set of random fields?

azrdev, 2022-07-20:
Yes, that's what I propose.

I don't find it in the docs, but https://github.com/influxdata/influxdb-python/issues/655 suggests that data points can simply omit fields, and i can confirm on CLI:
~~~
> insert measure f1=1,f2=2
> insert measure f1=1,f2=2,f3=3
> insert measure f2=2
> select  * from measure
name: measure
time                f1 f2 f3
----                -- -- --
1658354281756369335 1  2
1658354290249673637 1  2  3
1658354298986028596    2
~~~

zuckschwerdt, 2022-07-20:
I imagine the data table somewhat like the CSV output. About 350 fields when the default protocols are enabled. Or more likely the subset you actually receive I guess. My concern is if that's too many empty columns on most points?

azrdev, 2022-07-25:
AIUI there's no reason do worry about performance with 350 fields: https://community.influxdata.com/t/maximum-number-of-fields-and-tags/9472/3

It would be lacking overview to look at the whole rtl433 measurement with that sparse data, but that's what tags are for. Plus in my use case it would actually improve overview, as described, since I could query for e.g. `temperature_C` and get all values regardless of source (would have to differentiate them by model= tag then, but I think this should be doable).

zuckschwerdt, 2022-07-25:
We should note however that it's likely an abuse of the format. Tags are surely meant to describe the source, not the value.
I.e. tags would usually be "machine-a", "database-5", ... as opposed to "tire temperature", "oven thermometer", "soil probe".
Or in other words, we can't guarantee that all instances of a given field can be reasonably plotted in the same graph.

gdt, 2023-10-01:
My quick read is that we probably shouldn't do this.   What if someone is storing data about 10 machines, such as disk usage %, cpu load, temperature, and 20 other things?   How will they organize that?   Why is rtl_433 reporting about 10 temp/humidity sensors different?

gdt, 2023-10-01:
ticket gardening: What's the plan?  This has been dormant, and it seems if we want to consider it we need an implementation on the table with a trip report about using it.   I have marked it discuss and inactive for now.