merbanan/rtl_433 #2119: New device InFactory PT310 (ZX-7074/7073) pool thermometer

unverbraucht, 2022-07-21:
I bought this pool thermometer that's not currently decoded. I added samples and what I think might be the decoder in the [tests repo](https://github.com/merbanan/rtl_433_tests/compare/master...unverbraucht:rtl_433_tests:master), along with some samples:

Test output:
Data can be decoded by `-X 'n=pool3,m=OOK_PPM,s=1000,l=2008,g=2024,r=4064'`. Data packets get repeated 6 times with an empty packet between - maybe I'm missreading the encoding?

Decoding: 
```
Test mode active. Reading samples from file: ch1-27.7.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package    @0.247600s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.247600s
model     : pool3        count     : 11            num_rows  : 11            rows      : 
len       : 37           data      : 8e8115f6e0, 
len       : 0            data      : , 
len       : 37           data      : 8e8115f6e0, 
len       : 0            data      : , 
len       : 37           data      : 8e8115f6e0, 
len       : 0            data      : , 
len       : 37           data      : 8e8115f6e0, 
len       : 0            data      : , 
len       : 37           data      : 8e8115f6e0, 
len       : 0            data      : , 
len       : 37           data      : 8e8115f6e0
codes     : {37}8e8115f6e0, {0}, {37}8e8115f6e0, {0}, {37}8e8115f6e0, {0}, {37}8e8115f6e0, {0}, {37}8e8115f6e0, {0}, {37}8e8115f6e0
Analyzing pulses...
Total count:  233,  width: 463.29 ms            (115823 S)
Pulse width distribution:
 [ 0] count:  227,  width:  472 us [424;500]    ( 118 S)
 [ 1] count:    6,  width:  108 us [104;116]    (  27 S)
Gap width distribution:
 [ 0] count:  113,  width: 1968 us [1948;2052]  ( 492 S)
 [ 1] count:  114,  width:  984 us [980;1000]   ( 246 S)
 [ 2] count:    5,  width: 3916 us [3912;3924]  ( 979 S)
Pulse period distribution:
 [ 0] count:  113,  width: 2428 us [2140;2452]  ( 607 S)
 [ 1] count:  114,  width: 1464 us [1452;1476]  ( 366 S)
 [ 2] count:    5,  width: 4348 us [4344;4360]  (1087 S)
Pulse timing distribution:
 [ 0] count:  227,  width:  472 us [424;500]    ( 118 S)
 [ 1] count:    6,  width:  108 us [104;116]    (  27 S)
 [ 2] count:  113,  width: 1968 us [1948;2052]  ( 492 S)
 [ 3] count:  114,  width:  984 us [980;1000]   ( 246 S)
 [ 4] count:    5,  width: 3916 us [3912;3924]  ( 979 S)
 [ 5] count:    1,  width: 10004 us [10004;10004]       (2501 S)
Level estimates [high, low]:   1592,     16
RSSI: -10.1 dB SNR: 20.0 dB Noise: -30.1 dB
Frequency offsets [F1, F2]:   -6487,      0     (-24.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB035060501D8006C07B003D80F4C271482838383828282838283838383838382838383828382838282828282838282838282828383928455+AAB034060101D8006C07B003D80F4C2714828383838282828382838383838383828383838283828382828282828382828382828283839555

```

The `8e8115f6e0` would be for 27.7 degrees centigrade.

zuckschwerdt, 2022-07-21:
A safe margin should be added to gap and reset. Sync isn't supported -- it's gap in the case of PPM.

- `-X 'n=pool3,m=OOK_PPM,s=1000,l=2000,g=3000,r=5000'`

> Data packets get repeated 6 times with an empty packet between - maybe I'm missreading the encoding?

That's just the gap being to close and registering as sync I guess.

unverbraucht, 2022-07-21:
I've collected some more samples for testing. Still can't understand the last two nibbles before the last 0000 which seems to be some kind of checksum :(
[infactory-XZ-7034-tests.zip](https://github.com/merbanan/rtl_433/files/9162326/infactory-XZ-7034-tests.zip)

unverbraucht, 2022-07-21:
> * -X 'n=pool3,m=OOK_PPM,s=1000,l=2000,g=3000,r=5000'

Thanks @zuckschwerdt, that parses them more clearly. Then also all the strange 0's disappear - maybe the CRC will be more clear now.

zuckschwerdt, 2022-07-21:
Best to stick many codes into [a BitBench like this](https://triq.net/bitbench#c=%7B38%7De18106f334%5B27.7%20C%3F%5D&c=%7B38%7D4b810cf474&c=%7B38%7Db2810ef404&c=%7B38%7D7e810ffb74&c=%7B38%7Da08111f884&c=%7B38%7D8e8115f6e4&f=4b%204b%208h%208h%208h%204b%202b&co=8&cl=24&cw=4) and then look how they change and relate ;)

unverbraucht, 2022-07-22:
The protocol looks quite similar to nexus, which are also sold as other sensors under the Infactory brand: [nexus.c](https://github.com/merbanan/rtl_433/blob/c93aebfab6c6e2d5a72e189339c24e115b50b954/src/devices/nexus.c#L25). This would fit both the channel encoding and also that the battery still shows as good. Temp is also encoded the same way.

zuckschwerdt, 2022-07-22:
Seems very likely that this is a variant of the Rubicson /TFA /Kjell then. Change to allow 38 bits and add notes about the device and the difference and this issue number.
Also add ", TFA 30.3197" and ", InFactory PT310" to `.name`. Then run `maintainer_update.py` -- that should be all ;)

unverbraucht, 2022-07-22:
Great :) I opened a PR.

unverbraucht, 2022-07-22:
I also updated the existing Rubicson tests and added a test set for the PT-310 in a [rtl_433_tests PR](https://github.com/merbanan/rtl_433_tests/pull/427)

zuckschwerdt, 2022-08-08:
Closed by #2123

closed 2022-08-08 by zuckschwerdt