merbanan/rtl_433 #2121: Ecowitt WN34L Support

err53, 2022-07-21:
I'm going to be honest, this is my first time delving into SDR decoding and I don't know what I'm doing. If someone more experienced wants to hop in and lend a hand, that would be greatly appreciated.

I've made a PR for sensor data here: https://github.com/merbanan/rtl_433_tests/pull/426

After some poking around, it looks like PCM at 58 µs seems to produce something usable. I was able to isolate the sync word and the ID of the transmitter, but I haven't been able to find the temperature readings yet.

https://triq.net/bitbench#c=AA%20AA%20AA%20AA%20AA%20A8%20B7%2050%20D0%2000%20EF%2034%2008%2075%2011%20A8%2024%2082%204&c=55%2055%2055%2055%2055%2051%206E%20A1%20A0%2001%20DE%2068%2014%2062%2026%209F%2009%2004%208&c=55%2055%2055%2055%2055%2051%206E%20A1%20A0%2001%20DE%2068%201F%2042%2022%201D%2071%2001%200&c=55%2055%2055%2055%2055%2051%206E%20A1%20A0%2001%20DE%2068%2015%206A%2027%20D9%2051%2004%208&f=hh%20hh%20hh%20hh%20%7C%20ID%3Ahh%20hh%20%7C%20hh%20hh%20hh%20hh%20hh%20hh%20hh&a=Sync&m=2dd4&cw=4

err53, 2022-07-21:
Some progress: It seems like the temperature is sent as an integer between 0 and 1000, to get the real temperature in Celsius subtract 400 and shift the decimal point to the left by 1

https://triq.net/bitbench#c=AA%20AA%20AA%20AA%20AA%20A8%20B7%2050%20D0%2000%20EF%2034%2008%2075%2011%20A8%2024%2082%204&c=55%2055%2055%2055%2055%2051%206E%20A1%20A0%2001%20DE%2068%2014%2062%2026%209F%2009%2004%208&c=55%2055%2055%2055%2055%2051%206E%20A1%20A0%2001%20DE%2068%201F%2042%2022%201D%2071%2001%200&c=55%2055%2055%2055%2055%2051%206E%20A1%20A0%2001%20DE%2068%2015%206A%2027%20D9%2051%2004%208&f=SYNC%3Ahhhh%20%7C%20%3F%3Ahh%20hh%20%7C%20ID%3Ahhhh%20%7C%20bbbb%20bb%20%7C%20TEMP_C%3A10d%20%7C%20%3F%3Ahh%20hh%20hh%20hh%20hh%20&a=Sync&m=2dd4&cw=4

anthyz, 2022-07-21:
I've been working on a decoder for this. I'll create a PR shortly.

err53, 2022-07-21:
> I've been working on a decoder for this. I'll create a PR shortly.

Thanks, that sounds great! Let me know if you need any more data.

anthyz, 2022-07-21:
Look at other fineoffset decoders with the same PCM at 58 µs timings for hints on decoding the data.

anthyz, 2022-07-22:
I created PR https://github.com/merbanan/rtl_433/pull/2122.

closed 2022-07-22 by zuckschwerdt