merbanan/rtl_433 #2126: Fish Finder

Gadgeteering, 2022-07-24:
Anybody decoded a fish finder yet ?

[image]

Gadgeteering, 2022-07-25:
I have uploaded some samples data
[samples.zip](https://github.com/merbanan/rtl_433/files/9184126/samples.zip)

zuckschwerdt, 2022-07-25:
`rtl_433 -X 'n=fishfinder,m=OOK_PWM,s=175,l=575,r=700'`

There are three different signals: 20 bits `c0000`, 16 bits `b3cc` or `b3cc` and one long pulse (6.6 ms, shown as `{1}0`).
Data could be in that `b3..` packet, or if those don't vary then the data could be in the timing? Faster signals -> more fish close by? Or what should the signal indicate?

Gadgeteering, 2022-07-25:
I think the 20bits might be a sync code as this changes when you switch the unit on and off

The 16bits might be the water temperature and the other data packets are the returns

[Sample.zip](https://github.com/merbanan/rtl_433/files/9184387/Sample.zip)

Gadgeteering, 2022-07-29:
I don't think that it is pure digital signal there maybe an analogue echo signal, the temperature maybe transmitted as digital signal.

[Fish finder sonar: circuits, protocols and transducers](https://www.eevblog.com/forum/projects/fish-finder-sonar-circuits-protocols-and-transducers/)

Gadgeteering, 2022-08-03:
I have now managed to decode the 20 Bits, the first 16 bits are the temperature, strangely the bits are doubled up, odd and even bits are the same. So I needed to extract every other bit to decode the temperature. I still need to calibrate the algorithm to get the correct temperature.

[image]

gdt, 2023-10-14:
@Gadgeteering Are you making progress towards a PR to add a decoder?

gdt, 2024-06-04:
This seems abandoned and it's really iffy if there is enough info in the PR to take up cognitive space in the tracker.

closed 2024-06-04 by Gadgeteering