merbanan/rtl_433 #2132: R900 and SCM Not Found

robertscheib, 2022-07-29:
Hello,

I just installed this on a Pi0W but I'm unable to get it to see anything. I know at least my water meter is broadcasting as I can see it using rtlamr. When I run "rtl_443 -R 149" it just sits at the below.

My end goal is to locate my water, power and gas meters and pass that data along to Home Assistant. As a second goal, I'd like to see if I can grab any appliances in my house that are broadcasting or a local weather station.

I'm still a novice at both linux and SDR, so any help and support is appreciated.

```pi@raspberrypi:~ $ rtl_433 -R 149
rtl_433 version 21.12-125-g9d2e0b7f branch master at 202207281452 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 219 device decoding protocols [ 149 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
```

If I try and run "rtl_443" to see everything it just hangs here

```
pi@raspberrypi:~ $ rtl_433
rtl_433 version 21.12-125-g9d2e0b7f branch master at 202207281452 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 187 out of 219 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-219 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

Mudary, 2022-07-29:
rtl_443  -f 915M -R 149

On 7/29/2022 3:22 PM, robertscheib wrote:
> rtl_443 -R 149

robertscheib, 2022-07-30:
> rtl_443  -f 915M -R 149
> 
> On 7/29/2022 3:22 PM, robertscheib wrote:
> > rtl_443 -R 149
> 

Thanks, I'll give that a shot later tonight. Any idea why I can't run it wide open to find everything?

Mudary, 2022-07-30:
If you just run rtl_433 -f 915M , you should fine SCM stuff, water meters are R900 here in Canada.
Maybe weather stations even most of those are 433.92.
Some TPMS tire pressure measuring on 315M.
Security (Honeywell) at 345M.

robertscheib, 2022-07-30:
No joy. Tried both commands, let them sit for 5-10 minutes, still hanging.

Ran "rtl_433 -f 915M and got the following

```
pi@raspberrypi:~ $ rtl_433 -f 915M
rtl_433 version 21.12-125-g9d2e0b7f branch master at 202207281452 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 187 out of 219 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-219 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 915.000MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
```

```
pi@raspberrypi:~ $ rtl_433  -f 915M -R 149
rtl_433 version 21.12-125-g9d2e0b7f branch master at 202207281452 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 219 device decoding protocols [ 149 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 915.000MHz.
Allocating 15 zero-copy buffers
```

Mudary, 2022-07-30:
What kind of antenna are you using. And do you receive any readings with 
'rtl_433' by itself?

robertscheib, 2022-07-30:
I'm using a NooElec R820T. The second example from my original post is what I get when I try to run "rtl_433" by itself. I know the antenna is working fine, because when I use "[amrscm2mqtt](https://community.home-assistant.io/t/get-your-smart-electric-water-and-gas-meter-scm-readings-into-home-assistant-with-a-rtl-sdr/93707)" I'm able to see my R900.

Mudary, 2022-07-31:
"rtl_433" does not decode R900 only ERT-SCM.

Mudary, 2022-07-31:
"rtl_433" does not decode R900 only ERT-SCM and ERT-IDM/network.

robertscheib, 2022-08-01:
I just went back and looked at ticket #977 and realized that I misread it. Only SCM+ was added, by @merbanan R900 was not added at that time. That was my mistake for not reading that correctly.

Is the output I'm receiving below expected if there is nothing in my area that RTL_433 can receive?

```
pi@raspberrypi:~ $ rtl_433
rtl_433 version 21.12-125-g9d2e0b7f branch master at 202207281452 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 187 out of 219 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-219 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```

Mudary, 2022-08-01:
Unless you live on a farm and have no weather stations nearby, guess a 
couple of blocks in the city, you should pick something up. Try rtl_433 
-f 915M, gas, water, electric meters are frequency hopping between 
902Mhz-924Mhz.

What are you using for a computer, PC, Mac, RasberryPI, if you are using 
a laptop, go around the area to see if you can pick anything up. Another 
choice is to get an outside antenna.

robertscheib, 2022-08-12:
I was using a Pi3B+, if I run it on my Pixelbook, I'm getting the visualization from the readme and I'm getting TPMS reports. I think from the cars driving by my house. 

I tried the other commands your provided and I'm not getting anything. I was able to get the R900 values from a HA addon, so, I'm good to go. I'll just putz around with this on my pixelbook.

I appreciate the help.

closed 2022-08-12 by robertscheib

frevoli, 2022-08-12:
Hello guys.

I’ve noticed that if you use flag 
-W <filename>.ook 
the date and time is only written at the top of the file once. Each subsequent detection records…

;pulses
;freq1
;centerfreq
;samplerate
;sampledepth
;range
;rssi
;snr
;noise

But there is no time stamp.

Is it possible to add this with another flag somehow?

Many thanks
Matt

zuckschwerdt, 2022-08-12:
> noticed that if you use flag -W <filename>.ook the date and time is only written at the top of the file once.

Good point, thanks. I'll fix that. Probably a `;received DATETIME` line.

zuckschwerdt, 2022-08-12:
> Good point, thanks. I'll fix that. Probably a `;received DATETIME` line.

done with 8c7806b.

frevoli, 2022-08-12:
Thanks Christian, that was super quick!
How do I get the revised code? Via a git update?
Thanks
Matt

zuckschwerdt, 2022-08-12:
Yes, clone a fresh copy, or run `git pull` then build and install.