merbanan/rtl_433 #2133: Watchman Plus

arfond, 2022-07-30:
Ref - https://groups.google.com/g/rtl_433/c/VJSPl7h0848

Hello all, I'm a newbie to rtl_433 since I moved into a house with an oil tank.... Searching for ways to monitor oil levels in openhab I came across rtl_433. I'm all setup, sucessfully reading a temperature sensor and what appears to be 3 watchman sonic sensors from my neighbours, but not my own... I've worked out it's a watchman plus. Searching has brought me to this thread and I'm sorry to say I'm struggling to follow. Any step by step instructions anyone can provide would be much appreciated. My other option is to remove the watchman plus and replace it with a sonic unit, which I'm able to read just fine. Thank you in advance.

zuckschwerdt, 2022-07-30:
Thanks! Yery likely we can work out the protocol for this "Plus"model. As mentioned, anybody please dump a range of codes to figure this out.

arfond, 2022-07-31:
Here is the 1st capture I think, on the basis it's full of f0f0f0f0f0 or f83e0f83e0. I'll leave it running and based on previous posts there should be another coming in an hour.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:05:16
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
codes     : {2482}f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

zuckschwerdt, 2022-07-31:
We need to know what's expected to be in that data so we can find the fields. What depth value and maybe temperature would this be?

zuckschwerdt, 2022-07-31:
Note for reference: the following are equivalent:
- `11111` from the PWM codes, `f0f0f0f0f0` from the PCM codes, a logical `1` bit
- `0000` from the PWM codes, `f83e0f83e0` from the PCM codes, a logical `0` bit

Alternate representations thus are:

- `f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0`

- `f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0`

- `11111111111101010011110111011101000101000100101000010011000011`
- `111111111111010100111101110111010001010001010110000100101000010100111101`

Converted to hex this is:
- `fff5 3ddd 144a 1303 [F with Bund alarm ON]`
- `fff5 3ddd 1456 1285 3d [F with Bund alarm OFF]`

The code above is `11111111111101010001010000100001001101000101100000010010110011` thus
- `fff5 1421 3458 1233 [2022-07-31 09:05:16]`

arfond, 2022-07-31:
> We need to know what's expected to be in that data so we can find the fields. What depth value and maybe temperature would this be?

I'll do a bit of research and calcs to come back to you in this. The Watchman Plus has a sensor pole that drops from the top of the tank into the oil I think, so not sure what it will be transmitting (air depth or oil depth). Will look at details of tank and what's shown on the plug / screen and come back to you later. Thank you for your support.

arfond, 2022-07-31:
Just to share, i removed the battery pack from the transmitter, and re-inserted it. Below is what a go, a stream of two codes every 5 seconds or so. I removed the other devices / codes from this, except at the end.

Link to Watchman Plus (Niveau) here - https://www.commercialfuelsolutions.co.uk/downloads/manuals/oil_watchman.pdf - which shows the calibration / measurement pole.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:47:37
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 00
codes     : {6}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:47:37
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
codes     : {2482}f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:52:24
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 00
codes     : {6}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:52:24
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
codes     : {2482}f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:52:29
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 00
codes     : {6}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 09:52:29
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
[... cut ...] f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 10:01:29
model     : Plus         count     : 1             num_rows  : 1

zuckschwerdt, 2022-07-31:
After removing dups and unrelated or broken codes we have:
`11111111111101010001010000100001001101000101100000010010110011` = `fff5 1421 3458 1233`
which we already saw above as reading.

arfond, 2022-07-31:
Ok, hopefully what I'm posting is useful. Next step I understand is for me to give you an idea of what this reading relates to, in terms of oil or air depth, is that right? Just so I provide useful info. Thanks.

zuckschwerdt, 2022-07-31:
Yes, that would be great. Perhaps we can spot those values in the codes, otherwise we need a range of codes where those values change slowly, e.g. moving the sensor so it detects different oil height/distance.

Scoff123, 2022-07-31:
With my plug in receiver LCD showing '0' as the level, the distance from the u/s sensor to the oil level in my case was 84cm. I would expect this to be transmitted as 'depth_cm: 84' in a similar fashion to the sensors that are already decoded by RTL_433.
I expect the plug in receiver to do the calculation that this distance measurement from the u/s sensor equates to an oil level of 0, based on the original calibration/setup (previous home owner). My transmitter and plastic tube show 900mm on, and the plug in receiver shows 900+ as the model. The Niveau model linked is also the closest I could also find to the Plus model. 

Due to the age of the Plus being 2004, I don't believe it has much in the way of data that it actually transmits, unlike the newer Watchman Sonic devices that I believe transmit tank temperature and 'binding countdown' also. All I'm expecting within the transmitted data are:

device id:
alarm: (expecting this to be either 0 or 1 depending on state of the bund alarm_
depth_cm: expecting this to be the distance reading from the u/s sensor to the surface of the oil

While my plug in receiver shows '0', and with the bund alarm OFF, I get

{280}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f000000000f800007c03ff

which using BitBench equates to 

1111111111111111111111111111111111111111111111111111111111110000111110000111110000000011111111111111111111000011111111111111100001111111111111110000111110000000000001111100001111100000000000011111000000000000000000000000000000000000111110000000000000000000011111000000001111111111

which is the equivalent of 

f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0

How do you convert the binary form being used to hex?

zuckschwerdt, 2022-07-31:
The `f0f0f0...` I convert to bits using text search replace, here `11111111111101010011110111011101000101000100000000010000010011` then put that in BitBench with a leading `y` for binary: `y11111111111101010011110111011101000101000100000000010000010011` and choose `8h8h ` as format.

Scoff123, 2022-07-31:
Thanks for explaining :) OK so i've put the information i've gathered so far into a spreadsheet and attached it here. Once @arfond confirms the sensor distance, and what was displayed on the plug in receiver when the fff5 1421 3458 1233 [2022-07-31 09:05:16] reading was captured, then this can hopefully go into the spreadsheet on the appropriate data row
[Watchman Plus testing.xlsx](https://github.com/merbanan/rtl_433/files/9228788/Watchman.Plus.testing.xlsx)

arfond, 2022-07-31:
Sharing my results:

The sesnor tube that drops into the oil measures 111cm. I removed it, cleaned it,  put it back in, and took it out again to get a measure on the oil depth. This showed the lower 33cm of the tube is in oil. The plug shows 3. This corresponds to the previous data I've sent. This makes sense as 33/111 is 0.3 (or 3 on a scale of 0 - 10 assuming 111cm is 10).

I then tried adding some wood under the sensor, lifting it up by 5cm and 10cm. Meaning tube would be 28cm in oil and 23cm in oil.  Then I removed the wood, and removed the tube as far as I could (cable connecting the sensor not long enough to remove it completly, then I gradually lowered it until it was in it's final position. Data as follows:

5 cm lift (28cm in oil, or 78cm not in oil)

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 12:43:34
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 00
codes     : {6}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 12:43:34
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
codes     : {2482}f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 12:43:39
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 00
codes     : {6}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 12:43:39
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
codes     : {2482}f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 12:43:44
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 6            data      : 00
codes     : {6}00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-07-31 12:43:45
model     : Plus         count     : 1             num_rows  : 1             rows      :
len       : 2482         data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e [... cut ...] 83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f83e0f83e0f0f0f0f0f0f0f0f0f0f0f83e0f83e0f83e0f83e0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

arfond, 2022-07-31:
Also to say, I agree this sensor probably doesn't transmit as much data as the more recent sonic variants, but I'd expect it to transmit a low battery alarm (which shows up on the display, in addition to the items @Scoff123 mentioned above.

Scoff123, 2022-07-31:
Good point about the low battery alarm. I have an adjustable DC power supply, so should be able to feed the transmitter with various voltages below the required 6v until I trigger the low battery alarm and we can hopefully identify that within the data as well.

zuckschwerdt, 2022-07-31:
Cleaned up and converted we see:
- 5 cm lift (28cm in oil, or 78cm not in oil)

`fff5 1421 3458 1266 [2022-07-31 12:43:34]`
...

- 10 cm lift (23cm in oil, or 83cm not in oil) - plug shows 2 in this position.

`fff5 1421 3448 1666 [2022-07-31 12:47:58]`
...

- gradually lifting as far as possible (which hopefully should show a zero), then gradually lowering it into final position. The last bit of data in this chunk should correspond to the 1st data I posted, equal to 33cm in oil / 78 not in oil, 3 on the plug

`fff5 1421 3448 1666 [2022-07-31 12:52:01]`
`fff5 1421 3448 1666 [2022-07-31 12:52:06]`
`fff5 1421 3448 1666 [2022-07-31 12:52:11]`
`fff5 1421 3450 1166 [2022-07-31 12:52:17]`
`fff5 1421 3450 1166 [2022-07-31 12:52:22]`
`fff5 1421 3440 1566 [2022-07-31 12:52:28]`
`fff5 1421 3440 1566 [2022-07-31 12:52:33]`
`fff5 1421 3440 1566 [2022-07-31 12:52:38]`
`fff5 1421 3440 1566 [2022-07-31 12:52:44]`
`fff5 1421 3440 1566 [2022-07-31 12:52:49]`
`fff5 1421 3440 1566 [2022-07-31 12:53:00]`
`fff5 1421 3440 1566 [2022-07-31 12:53:05]`
`fff5 1421 3440 1566 [2022-07-31 12:53:11]`
`fff5 1421 3440 1566 [2022-07-31 12:53:16]`
`fff5 1421 3440 1566 [2022-07-31 12:53:22]`
`fff5 1421 3440 1566 [2022-07-31 12:53:27]`
`fff5 1421 3440 1566 [2022-07-31 12:53:32]`
`fff5 1421 3450 1166 [2022-07-31 12:53:38]`
`fff5 1421 3448 1666 [2022-07-31 12:53:43]`
`fff5 1421 3448 1666 [2022-07-31 12:53:49]`
`fff5 1421 3448 1666 [2022-07-31 12:53:54]`
`fff5 1421 3458 1266 [2022-07-31 12:53:59]`
`fff5 1421 3458 1266 [2022-07-31 12:54:05]`
`fff5 1421 3458 1266 [2022-07-31 12:54:10]`
`fff5 1421 3458 1266 [2022-07-31 12:54:16]`
`fff5 1421 3458 1266 [2022-07-31 12:54:21]`
`fff5 1421 3458 1266 [2022-07-31 12:54:26]`

It's not really clear how those changes compose a value.
Here is [a BitBench with those codes](https://triq.net/bitbench#c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A52%3A01%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A52%3A06%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A52%3A11%5D&c=fff5%201421%203450%201166%20%5B2022-07-31%2012%3A52%3A17%5D&c=fff5%201421%203450%201166%20%5B2022-07-31%2012%3A52%3A22%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A28%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A33%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A38%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A44%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A49%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A00%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A05%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A11%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A16%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A22%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A27%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A32%5D&c=fff5%201421%203450%201166%20%5B2022-07-31%2012%3A53%3A38%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A53%3A43%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A53%3A49%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A53%3A54%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A53%3A59%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A05%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A10%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A16%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A21%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A26%5D&f=PRE%3F8h8h%20ID%3F8h8h8h%20DEPTH%3F8b%208b%20TRAILER%3F8h&cw=4), but the alignment isn't at all clear.

arfond, 2022-07-31:
Summarising what we've seen so far, based on below I don't think the last 2 digits are a trailer as suggested in the BitBench.

no lift (32cm in oil, 79cm not in oil) - plug shows 3 in this position

fff5 1421 3458 1233 [2022-07-31 09:05:16]

5 cm lift (28cm in oil, or 78cm not in oil)

fff5 1421 3458 1266 [2022-07-31 12:43:34]

10 cm lift (23cm in oil, or 83cm not in oil) - plug shows 2 in this position

fff5 1421 3448 1666 [2022-07-31 12:47:58]

arfond, 2022-07-31:
Sorry, correction, bad maths above

Summarising what we've seen so far, based on below I don't think the last 2 digits are a trailer as suggested in the BitBench.

no lift (32cm in oil, 79cm not in oil) - plug shows 3 in this position

fff5 1421 3458 1233 [2022-07-31 09:05:16]

5 cm lift (27cm in oil, or 84cm not in oil)

fff5 1421 3458 1266 [2022-07-31 12:43:34]

10 cm lift (22cm in oil, or 89cm not in oil) - plug shows 2 in this position

fff5 1421 3448 1666 [2022-07-31 12:47:58]

Scoff123, 2022-07-31:
Are you able to capture a transmission when your plug in receiver shows F for Full, either with the bund alarm on or off? Just curious if it will be identical to the code I captured? I'm going to try to get transmissions for a reading of 2 and 3 on my receiver in a minute
It's a slow process as I seem to have to power off my transmitter for hours at a time in order to start capturing transmissions on power on.

arfond, 2022-08-01:
Hi @Scoff123 - not sure I can get a measurement when Full as the Watchman is installed in the tank. Looking at the tank label I'd need 200 litres of oil to do that - pricey experiment :-) Any joy with getting readings with 2 and 3 on the plug?

arfond, 2022-08-02:
Thinking about this in the dog walk this morning. Maybe the transmission doesn't contain a depth reading, it might just contain what is displayed on the screen: 0-9 or F, then something else for low battery, something else for the alarm. The plug for the Sonic needs to know the depth as it has dip switches on the back to set tank depth. For the Plus device the measuring pole in the tank is cut to suit. The plug for the Plus probably doesn't do any calculations at all.

Scoff123, 2022-08-02:
> Hi @Scoff123 - not sure I can get a measurement when Full as the Watchman is installed in the tank. Looking at the tank label I'd need 2000 litres of oil to do that - pricey experiment :-) Any joy with getting readings with 2 and 3 on the plug?

Hi @arfond - sorry for the delay in replying, I didn't get a chance yesterday. It's probably a little bit extreme asking you to fill the tank up, especially when it holds 2000 litres :) My tank is 1300 and I'm dreading the cost of refilling when the time comes!

I'm actually cheating with my setup - I've removed the whole unit form my tank and I have it rigged up with a bucket of water instead. I've successfully managed to get readings for level 2 and level 3 now, but in order to get the remaining levels, "we're gonna need a bigger bucket" :)

With the plug in receiver showing level 2, I captured a transmission of 
{283}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f00f8000007c03fff8007fe
or
11111111111101010011110111011101000101000100100000010011100011
with a hex of
fff5 3ddd 1448 1323

With the plug in receiver showing level 3, I captured a transmission of 
{284}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0ffc000003e1f0ffc003ff
or
11111111111101010011110111011101000101000101100000010101100011
with a hex of
fff5 3ddd 1458 1523

Once I've rigged up a deeper bucket I can hopefully generate the remaining levels 4-9, and also the low battery warning.

You could be right in regards to the plug in receiver not doing any calculations - the transmissions could simply be the level number, as calculated by either the u/s sensor itself, based on the distance it reads to the oil surface, or the transmitter does this before sending the level to display on the plug in receiver. Hopefully once we have a full data set we can identify if this is the case.

arfond, 2022-08-02:
Straight away I see xxxx xxxx xx58 and 48 in your data, same as mine for levels 3 and 2 respectively.... Not sure if it's this simple....

arfond, 2022-08-02:
You @Scoff123 appear to have this for zero

fff5 3ddd 1440 1013 i.e 40

Scoff123, 2022-08-02:
That's correct - this is what I have so far:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Watchman   Plus Receiver status | Distance to Oil   surface | Hex
-- | -- | --
F with Bund   alarm ON | <20cm | fff5 3ddd 1456 1285   3d
F with Bund   alarm OFF | <20cm | fff5 3ddd 144a 1303
9 |   |  
8 |   |  
7 |   |  
6 |   |  
5 |   |  
4 |   |  
3 | 67-66cm | fff5 3ddd 1458 1523
2 | 73cm | fff5 3ddd 1448 1323
1 | 78-75cm | fff5 3ddd 1450 1723
0 | 84cm | fff5 3ddd 1440 1013

</body>

</html>

zuckschwerdt, 2022-08-02:
Just to note: e.g. xx48 might very well not be the actual field. There is no indication to the alignment yet.
It could be shifted by up to 3 bits, e.g. could be xx90, xx20, xx40 if you change the alignment.

I would expect that for one alignment the field will show a plain linear matching to the level numbers.

Scoff123, 2022-08-04:
I've finally been able to complete my testing and have a full data set now. Hopefully this will shed some light on things. I've managed to capture a transmission for each of the levels 0-9 and Full. I've also captured a 'connection error', where a 'c' flashes alternately with the current level number to indicate a bad connection somewhere between the transmitter and bund alarm or u/s sensor. Also, low battery warning which flashes 'Bat Low' underneath the number on the plug in receiver when the supply voltage is 4.10v or below. Just converting and completing the spreadsheet and I will upload the data...

arfond, 2022-08-04:
Thank you @Scoff123 - look forward to seeing the results.

Scoff123, 2022-08-05:
I had to retake a couple of measurements as they turned out to be from a different neighbourhood watchman device after generating the hex code, but all done now. Here is the short table - I've attached the full spreadsheet as well

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Watchman   Plus Receiver status | Distance to Oil   surface | Hex
-- | -- | --
F with Bund   alarm ON | <18cm | fff5 3ddd 1456 1285   3d
F with Bund   alarm OFF | <18cm | fff5 3ddd 144a 1303
9 | 18cm | fff5 3ddd 1452 1703
8 | 21cm | fff5 3ddd 1442 1023
7 | 31cm | fff5 3ddd 145c 1423
6 | 40cm | fff5 3ddd 144c 1223
5 | 47cm | fff5 3ddd 1454 1623
4 | 61cm | fff5 3ddd 1444 1123
3 with 'Bat   Low'  flashing (4.10v or below) | 67-66cm | fff5 3ddd 1458 3521
3 alternating   with c (connection error) | 67-66cm | fff5 3ddd 1456 1604   03
3 | 67-66cm | fff5 3ddd 1458 1523
2 | 73cm | fff5 3ddd 1448 1323
1 | 78-75cm | fff5 3ddd 1450 1723
0 | 84cm | fff5 3ddd 1440 1013

</body>

[Watchman Plus testing.xlsx](https://github.com/merbanan/rtl_433/files/9268199/Watchman.Plus.testing.xlsx)

</html>

zuckschwerdt, 2022-08-05:
When looking at neighbourhood data, did you spot where the ID is? E.g. is it `53ddd14`, `3ddd14`, `3ddd` -- which bits seem fixed or do change?

The plain level (sans Bund, which seems to add "3") and the battery flag are visible now.
The bits are reversed (LSB first).
The alignment seems to off by 1 (i.e. the preamble is 15bit not 16).

Here is [a BitBench with those results](https://triq.net/bitbench#c=fff53ddd145612853d%5BF%20with%20Bund%20alarm%20ON%20%3C18cm%5D&c=fff53ddd144a1303%5BF%20with%20Bund%20alarm%20OFF%20%3C18cm%5D&c=fff53ddd14521703%5B9%2018cm%5D&c=fff53ddd14421023%5B8%2021cm%5D&c=fff53ddd145c1423%5B7%2031cm%5D&c=fff53ddd144c1223%5B6%2040cm%5D&c=fff53ddd14541623%5B5%2047cm%5D&c=fff53ddd14441123%5B4%2061cm%5D&c=fff53ddd14583521%5B3%20with%20'Bat%20Low'%20%20flashing%20(4.10v%20or%20below)%2067-66cm%5D&c=fff53ddd1456160403%5B3%20alternating%20with%20c%20(connection%20error)%2067-66cm%5D&c=fff53ddd14581523%5B3%2067-66cm%5D&c=fff53ddd14481323%5B2%2073cm%5D&c=fff53ddd14501723%5B1%2078-75cm%5D&c=fff53ddd14401013%5B0%2084cm%5D&f=PRE%3F7h8h%20ID%3F8h8h8h4h%20LEVEL%3F%5E4d%20FLAGS%3F4b%204b%204b%205b%208b&cw=4).

zuckschwerdt, 2022-08-05:
Also try `PRE?7h8h ID?^8h^8h^8h^4h LEVEL?^4d FLAGS?^6b ^8h ^8h ^8h` format to see a pattern in the last field(s).

Scoff123, 2022-08-05:
Thanks. I noticed, what I believe to be neighbourhood Watchman Devices, have different starting bits, for example:

{281}fffffffffffffff0fc3e01ffffe1fffc3fff87c003e1f000f8000000007c00003e01ff8
fff6 9eee 8a20 0813

I don't know if this device is a newer version which rtl_433 is currently able to decode with one of the existing protocols

Both my device and arfond's start with fff5 xxxx

With the alignment being off by 1, it may not relate to this at all, however either before or after every transmission there is a single {1}8 transmission:

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

05/08/2022 11:57 |   | {1}8
-- | -- | --
05/08/2022 11:57 |   | {284}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f807c0003e1fffc0003ff
05/08/2022 11:57 |   | {1}8
05/08/2022 11:57 |   | {284}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f807c0003e1fffc0003ff
05/08/2022 11:57 |   | {1}8
05/08/2022 11:57 |   | {284}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f807c0003e1fffc0003ff
05/08/2022 11:57 |   | {1}8

</body>

</html>

arfond, 2022-08-05:
We're making progress! Thanks @Scoff123 for your testing. I'm receiving data from 3 other Watchman devices, all of them decoded by default. I thought one of them was mine until one day the depth went up massively - one of the neighbors had filled up their tank... This led me to posting on the forum... I have no other plus devices around me (not surprising given its age).

zuckschwerdt, 2022-08-05:
Paste the `fff6 9eee 8a20 0813` into the BitBench and it shows neatly as empty tank. Could be a Plus model.

Scoff123, 2022-08-05:
It does indeed! I am leaving my rtl_433 -R 0 -F csv:log.csv -X 'n=name,m=OOK_PWM,s=3200,l=4000,g=5000,r=9000' command running for a few hours as this should pick up my hourly reading, which I have left as Full at the moment, as well as hopefully any other Plus devices. I'll share any findings.
What else do we need/where do we go from here in order to progress this further?

zuckschwerdt, 2022-08-05:
I can add this as a prototype/debug decoder so you can run this for longer and check if the fields work.
If one of you is up to that, then you could also code and PR that decoder.

I would like to pin down where the ID bits are and if that last field is maybe a checksum.
The  "level" field is mostly well defined now, the ID would be some number of bits to the left, e.g. all 24 (or 28) bits.
The "flags" field is unlikely to be 6 bits wide. We only see the 0's and the one bit für battery low. The remainder is some kind of value that fits the level, could be a checksum (addition with carry).

Scoff123, 2022-08-05:
I'm happy to do some more testing if you are able to add this as a prototype/debug decoder.
I'm not sure if I can code and PR the decoder but can see what's involved/required in order to do that, as I don't have any experience with PR. 
Regarding the ID, is this likely to just be some hard coded code on both transmitter and receiver in order to sync/handshake the two devices? I don't imagine it would be the serial number of the device would it. My serial number, which flashes one digit at a time on the LCD when you turn on the plug in receiver at the wall socket, is 05073745  - just in case that number is lurking somewhere in my captured transmissions?

arfond, 2022-08-05:
I'm also running to confirm if I don't actually have another plus model. Whilst running I've noticed mine now has connection error for some reason....

fff5 1421 3456 1605 630

arfond, 2022-08-05:
Sorted it, hasn't seated the batteries correctly after testing the other night. Now have

fff5 1421 3458 1266

PRE?7ffa ID?5108594 LEVEL?03 FLAGS?010000 9a 1

My plug has the following code after powering up : 05 40 41 05

arfond, 2022-08-05:
> Paste the `fff6 9eee 8a20 0813` into the BitBench and it shows neatly as empty tank. Could be a Plus model.

I did the above and noticed the 1st number in the flags is 1, whereas with mine and @Scoff123 's device it's always a 0.

Last digit in the PRE changes from 7ffa for mine and @Scoff123 , to 7ffb for this other device.

[BitBench link](https://triq.net/bitbench#c=y111111111111010100010100001000010011010001011000000100101100110%20%5Barfon%20level%203%5D&c=fff53ddd14581523%20%5Bscott%20level%203%5D&c=fff53ddd14401013%20%5Bscott%20level%200%5D&c=fff6%209eee%208a20%200813%20%5Bscott%20other%20device%20level%200%5D&f=PRE%3F7h8h%20ID%3F%5E8h%5E8h%5E8h%5E4h%20LEVEL%3F%5E4d%20FLAGS%3F%5E6b%20%5E8h%20%5E8h%20%5E8h&cw=4)

zuckschwerdt, 2022-08-05:
>  ID, is this likely to just be some hard coded code on both transmitter and receiver... is 05073745

It's likely (part of) that is encoded in the transmission. But 05073745 has a very different bit pattern to 3ddd14 -- encoded as BCD it might roughly fit. @arfond what ID is printed on your device?

> noticed the 1st number in the flags is 1, whereas with mine and @Scoff123 's device it's always a 0.

Preamble usually ends with a fixed pattern (e.g. fff5) but some protocols have two different pattern, depending on the first payload bit -- could be the case here.

arfond, 2022-08-05:
> It's likely (part of) that is encoded in the transmission. But 05073745 has a very different bit pattern to 3ddd14 -- encoded as BCD it might roughly fit. @arfond what ID is printed on your [...]

These plugs have nothing printed on them. A code flashes on the screen when it's turned off and on again. Mine is 05404105.

Scoff123, 2022-08-05:
Both my transmitter and receiver have the following sticker with serial number on, and it's this number (minus one of the initial 0's) which flashes on the LCD of the plugin receiver when you turn it off and back on again. It's likely your stickers may have come off during their lifetime
[image]

arfond, 2022-08-05:
[image]

Found mine... Last digit rubbed off at some point.

Scoff123, 2022-08-05:
Well done! I've been playing around with hex and BCD but so far I've not been able to find anything close to the serial number. @arfond - your other Watchman devices, do they have different formats of serial number to our Plus devices? Just curious what format they use, as the ID field in the transmissions of the already 'known' watchman devices have ID'S like this:

[image]
 so it would be helpful, I think, to see if these devices transmit the serial number or if they use a different number as the ID

zuckschwerdt, 2022-08-05:
A [BitBench with ID's and codes](https://triq.net/bitbench#c=y000000%20x05404105%20%5Barfon%20ID%5D&c=fff5142134581266%20%5Barfon%20level%203%5D&c=&c=y000000%20x05073745%20%5Bscott%20ID%5D&c=fff53ddd14581523%20%5Bscott%20level%203%5D&f=v&cw=4). @arfond ID is "lighter" (fewer 1 bits) and @Scoff123 is "heavier" (more 1 bits) and that is the same in the codes. And we know bits should be reversed... somehow. Quite the puzzle though.

arfond, 2022-08-06:
Attached is one of the neighbouring watchman's I'm seeing:

[image]

arfond, 2022-08-06:
I've been reading a bit this morning, trying to work out how to define a flex decoder in the conf file for this plus device... I'm getting somewhere, but not sure how exactly to use the get and match options....

zuckschwerdt, 2022-08-06:
> trying to work out how to define a flex decoder in the conf file for this plus device

Test best flex decoder is already listed above. The custom coding can only be decoded manually. I'll add a proper decoder soon.

Scoff123, 2022-08-06:
In the meantime, just for completeness, I wanted to see exactly what intervals the Watchman Plus transmits - once the initial 10-15 minute of 5 second transmissions is over. The Interval seems to vary a bit - I don't know what controls this timing but thought it would perhaps be a little more regular. Anyway, ignoring the gap where one reading wasn't picked up, it sends a reading on average every 46 minutes, which maybe over a longer period would average out to 45 minutes

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/THINKP~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Time | Interval | Codes
-- | -- | --
05/08/2022   18:50 |   | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
05/08/2022   19:30 | 00:40:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
05/08/2022   20:12 | 00:42:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
05/08/2022   21:49 | 01:37:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
05/08/2022   22:41 | 00:52:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
05/08/2022   23:32 | 00:51:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   00:23 | 00:51:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   01:12 | 00:49:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   01:59 | 00:47:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   02:41 | 00:42:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   03:24 | 00:43:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   04:13 | 00:49:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   05:01 | 00:48:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   05:44 | 00:43:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   06:33 | 00:49:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   07:21 | 00:48:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   08:04 | 00:43:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   08:53 | 00:49:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8
06/08/2022   09:39 | 00:46:00 | {325}fffffffffffffff0f87c03ffffc3fff87fff0f8007c3e001f0f87fe0001f0ffc000003e00000001ff8

</body>

</html>

arfond, 2022-08-30:
Hi @zuckschwerdt - hope you're doing well, and don't mind me asking if you've managed to find some time to write the decoder for this Watchman Plus device? Many thanks in advance.

zuckschwerdt, 2022-08-30:
Thanks for pinging, yes I need to get around to this. Can you provide a same for testing? `rtl_433 -S unknown` then look for a good file, zip, upload here.

zuckschwerdt, 2022-08-30:
Oh, wait, we already have captures in the Groups post. Nevermind then :)

arfond, 2022-08-30:
> Oh, wait, we already have captures in the Groups post. Nevermind then :)

OK, thanks. Let me know if you need anything else.

jacobstim, 2022-09-03:
Also stumbled across this thread in an attempt to read out my Sonic Watchman Plus device. Will be happy to provide any help, data, captures or test in case that is useful!

arfond, 2022-11-06:
Hi @zuckschwerdt - I read through the wiki to see if I could have a go at adding a decoder for this Watchman Plus device, but I'm afraid it's completely beyond me..... If you have time to look at this it would be very much appreciated.

zuckschwerdt, 2022-11-06:
> If you have time to look at this it would be very much appreciated.

I'm really busy currently, it'll take some time.

klohner, 2022-11-07:
Just took another look at these signals and may have found the encoding for the ID code and a couple other things.

First, a decode from @arfond's device:

~~~
fff5142134581266 [arfond ID code is 05404105]
0y1111111111110101000101000010000100110100010110000001001001100110 [binary]
               10100010  00010000  01101000 [reverse these bits, in octal: 05404105]
~~~

That looks promising.  How about a decode from @Scoff123's device:

~~~
fff53ddd14581523 [Scoff123 ID code is 05073745]
0y1111111111110101001111011101110100010100010110000001010100100011 [binary]
               10100111  11101110  00101000 [reverse these bits, in octal: 05073745]
~~~

Great!  Another pattern I noticed here is that between each of those 8-bit sections are the two bits `10`

Now, let's look at the decode from Scoff123's suspected neighbor:

~~~
fff69eee8a200813 [Scoff123 suspected neighbor's device has unknown ID code]
0y1111111111110110100111101110111010001010001000000000100000010011
               11010011  01110111  00010100 [reversed, in octal: 12167313]
~~~

Hmm, that looks a little strange.  It doesn't start with `05` like the others and the two bits between the bytes aren't `01`.  So, just to be safe, I tried to convert the original signal of `{281}fffffffffffffff0fc3e01ffffe1fffc3fff87c003e1f000f8000000007c00003e01ff8` and noticed that there's an extra `1` symbol in there that apparently caused a bad decode.  Here's a corrected decode:

~~~
fff53ddd14401026 [fixed decode of neighbor]
0y1111111111110101001111011101110100010100010000000001000000100110
               10100111  11101110  00101000 [reversed, in octal: 05076765]
~~~

That looks more like what I expected.  ID code starts with `05` and we now have the `10` bits between those bytes.

With a little more experimenting, I think I found the Battery Low indicator and the depth digit.   Here's a bitbench format string (note that I've reversed several bit sections since I think values are least significant bit first):

~~~
PREAMBLE:12hb ID_BITS_REVERSED_OCTAL:8bxx8bxx8bxx LEVEL_CODE:^4h ??:3b BAT_LOW:1v XX:bb ??:^4b ??:^4b ???:2b ??:^4b ^4b bb ^4x
~~~

And here's a [bitbench](https://triq.net/bitbench#c=%5Bzuckschwerdt%20commented%20on%20Jul%2031%20https%3A%2F%2Fgithub.com%2Fmerbanan%2Frtl_433%2Fissues%2F2133%23issuecomment-1200382214%5D&c=fff5%203ddd%20144a%201303%20%5BF%20with%20Bund%20alarm%20ON%5D&c=fff5%203ddd%201456%201285%203d%20%5BF%20with%20Bund%20alarm%20OFF%5D&c=fff5%201421%203458%201233%20%5B2022-07-31%2009%3A05%3A16%5D&c=&c=%5Bzuckschwerdt%20commented%20on%20Jul%2031%20https%3A%2F%2Fgithub.com%2Fmerbanan%2Frtl_433%2Fissues%2F2133%23issuecomment-1200419534%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A43%3A34%205%20cm%20lift%20(28cm%20in%20oil%2C%20or%2078cm%20not%20in%20oil)%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A47%3A58%2010%20cm%20lift%20(23cm%20in%20oil%2C%20or%2083cm%20not%20in%20oil)%20-%20plug%20shows%202%20in%20this%20position.%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A52%3A01%20gradually%20lifting...%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A52%3A06%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A52%3A11%5D&c=fff5%201421%203450%201166%20%5B2022-07-31%2012%3A52%3A17%5D&c=fff5%201421%203450%201166%20%5B2022-07-31%2012%3A52%3A22%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A28%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A33%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A38%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A44%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A52%3A49%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A00%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A05%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A11%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A16%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A22%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A27%5D&c=fff5%201421%203440%201566%20%5B2022-07-31%2012%3A53%3A32%5D&c=fff5%201421%203450%201166%20%5B2022-07-31%2012%3A53%3A38%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A53%3A43%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A53%3A49%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A53%3A54%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A53%3A59%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A05%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A10%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A16%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A21%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A54%3A26%5D&c=&c=%5Barfond%20commented%20on%20Jul%2031%20https%3A%2F%2Fgithub.com%2Fmerbanan%2Frtl_433%2Fissues%2F2133%23issuecomment-1200436431%5D&c=fff5%201421%203458%201233%20%5B2022-07-31%2009%3A05%3A16%20no%20lift%20(32cm%20in%20oil%2C%2079cm%20not%20in%20oil)%20-%20plug%20shows%203%20in%20this%20position%5D&c=fff5%201421%203458%201266%20%5B2022-07-31%2012%3A43%3A34%205%20cm%20lift%20(27cm%20in%20oil%2C%20or%2084cm%20not%20in%20oil)%5D&c=fff5%201421%203448%201666%20%5B2022-07-31%2012%3A47%3A58%2010%20cm%20lift%20(22cm%20in%20oil%2C%20or%2089cm%20not%20in%20oil)%20-%20plug%20shows%202%20in%20this%20position%5D&c=&c=%5BScoff123%20commented%20on%20Aug%202%20https%3A%2F%2Fgithub.com%2Fmerbanan%2Frtl_433%2Fissues%2F2133%23issuecomment-1202355318%5D&c=0y11111111111101010011110111011101000101000100100000010011100011%20%5BWith%20the%20plug [... cut ...] fff5%201421%203458%201266%20%5BSorted%20it%2C%20hasn't%20seated%20the%20batteries%20correctly%20after%20testing%20the%20other%20night.%20My%20plug%20has%20the%20following%20code%20after%20powering%20up%20%3A%2005%2040%2041%2005%5D&f=ID_BITS_REVERSED_OCTAL%3A8bxx8bxx8bxx%20LEVEL_CODE%3A%5E4h%20%3F%3F%3A3b%20BAT_LOW%3A1v%20XX%3Abb%20%3F%3F%3A%5E4b%20%3F%3F%3A%5E4b%20%3F%3F%3F%3A2b%20%3F%3F%3A%5E4b%20%5E4b%20bb%20%5E4x&a=Preamble&m=0y1111111111110&cw=4) with most of the data from this issue thread.

klohner, 2022-11-07:
Oops, I decoded the suspected neighbor's code improperly (I guess I'm not great at reading octal bits backwards).  It should be:

~~~
fff53ddd14401026 [fixed decode of neighbor]
0y1111111111110101001111011101110100010100010000000001000000100110
               10100111  11101110  00101000 [reversed, in octal: 05073745]
~~~

Which means that it's actually just Scoff123's device.

klohner, 2022-11-07:
And one more device code I found going back even further in the previous thread:

~~~
[Carl S https://groups.google.com/g/rtl_433/c/VJSPl7h0848/m/a9Odej4YAgAJ]
fff774d75c5912853d [Watchman Plus 2021-02-27 The serial number of the transmitter is 007353167]
~~~

The bitbench format string was able to confirm that ID code as well.

[11 more replies]