merbanan/rtl_433 #2135: Unable to demodulate/decode FSK

bkralik, 2022-08-03:
Hi,
I am trying to run the rtl_433 to catch communication of our building's smart light/windows shade switches. However, I am not able to get any reasonable output either using -A or specifying flex decoder with -X. 
On the spectrum, I can clearly see it is FSK, and using [Inspectrum](https://github.com/miek/inspectrum) tool I can clearly see the demodulated signal. Do you have any idea, what I am doing wrong? Playing with -Y (trying classic/min-max demodulators, different detection levels) doesn't do anything.
Thanks!

rtl_433 command:
`rtl_433 -r g003_868.2M_1000k.cu8 -Y classic -R 0 -X n=test,m=FSK_PCM,short=26,long=26,r=500 -vvvv`

output with analyzer (-A): [here](http://bin.bkralik.cz/?5c0d80d4c4621f7e#yk30p8w8eThUqUgAeYhRnkCt5JvFJv+aWRzIF5ENTTY=)

Inspectrum screenshots (spectrograms on top, demodulated frequency on bottom)
[image]

Zoom at the pulse sequence beginning
[image]

Captured file:
[g003_868.2M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/9252587/g003_868.2M_1000k.cu8.zip)

zuckschwerdt, 2022-08-03:
The signal is clipping (too loud) and the signal could be GFSK, something the demod doesn't handle well.
[image]

bkralik, 2022-08-03:
That's a good point! I somehow didn't think of looking at signal waveform directly.
I found one captured pulse which probably came from my colleague's office (it is not clipped), however, even this one doesn't work properly - so probably really that GFSK is the issue. Do you know, if there is any plan to implement the GFSK demodulator in rtl_433?

merbanan, 2022-08-03:
Well GFSK is supported by the min/max demodulation code but it cant handle to short symbols. So maybe raise the sample rate and make sure the signal doesnt clip.

Generic demodulation will always perform worse then fixed feature decoding but often the sensors are close enough for it to not matter.

Jabieraru142, 2023-03-09:
I have:
time      : 2023-03-09 16:00:06
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 431          data      : fffe603fffffffffffff01f8ffffff3ff807ffc03fff0ffffffffe00003fff000187ff00f1ff87ffffffffffffffffffffffffff8ffe
codes     : {431}fffe603fffffffffffff01f8ffffff3ff807ffc03fff0ffffffffe00003fff000187ff00f1ff87ffffffffffffffffffffffffff8ffe
 with Bresser 5 in1. How is possible correct decode ???
Many tnx de Jabi, ea2aru

zuckschwerdt, 2023-03-13:
Like we already said: verify a good transmission, grab a sample: https://triq.org/rtl_433/ANALYZE.html

gdt, 2023-09-29:
Seems like this was a help request and it's done.   Please clarify/restate if that's not the case and we can re-open.

closed 2023-09-29 by gdt