merbanan/rtl_433 #2136: Running WSL version of rtl_433

gvanem, 2022-08-04:
I've been experimenting  briefly with *Window Subsystem for Linux* under Win-10. 
I've built `rtl_433` in the WSL Ubuntu package okay (but w/o Soapy).

But now I'd like to have this WSL program connect to the Win-10 `rtl_433.exe` version
to receive the same decoded data as I do in Win-10.

What I do now:
 * On the Win-10 side: `rtl_tcp.exe -s 1M -a 10.0.0.10 -p 1234 -f 434M`
 * On the WSL side:  `rtl_433 -d rtl_tcp://10.0.0.10:1234`
  * This works fine. In WSL's bash: [image]  except for some issues with `rtl_tcp: Interrupted system call`.

So I'd like to do this without `rtl_tcp` as the server, but fails to understand how. Anyone?

gvanem, 2022-08-04:
I figured it out a bit more with:
 * on the Win-side: `rtl_433.exe -f 434.00M -s 1M -g 40 -d0 -M protocol -M time -F http://10.0.0.10:80`
 * on the WSL-side: `python3 examples/rtl_433_http_stream.py`: 
[image]

 
But this gives a small picture. Would it be possible (with colorama) to show everything like on a local console?

zuckschwerdt, 2022-08-04:
You could run `rtl_433 -F rtl_tcp` on the Windows side (add `-R 0` if you don't want processing), but this just replaces `rtl_tcp` with basically the same thing.

If you do want the processing on Windows, but the display in WSL (or anywhere) then extending `rtl_433_http_stream.py` to look like the `kv` output whould be interesting. (effectively porting https://github.com/merbanan/rtl_433/blob/master/src/output_file.c#L151-L354)

gvanem, 2022-08-04:
A `WSAStartup()` is missing in `output_rtltcp.c`:
```diff
--- a/src/output_rtltcp.c 2022-05-16 12:21:09
+++ b/src/output_rtltcp.c 2022-08-04 16:38:39
@@ -485,12 +485,20 @@

 struct raw_output *raw_output_rtltcp_create(const char *host, const char *port, r_cfg_t *cfg)
 {
+#ifdef _WIN32
+    WSADATA wsa;
+
+    if (WSAStartup(MAKEWORD(2,2),&wsa) != 0) {
+        perror("WSAStartup()");
+        return NULL;
+    }
+#endif
+
     raw_output_rtltcp_t *rtltcp = calloc(1, sizeof(raw_output_rtltcp_t));
```

The help-page in `rtl_433.exe -F` does not mention this option but I assume the syntax is `-F rtl_tcp://10.0.0.10:1234`.
So patching and trying:
* On the Win-10 side: `rtl_433.exe  -F rtl_tcp://10.0.0.10:1234 -R 0`
* On the WSL side: `rtl_433 -d rtl_tcp://10.0.0.10:1234`

doesn't work. The WSL (Ubuntu 22.04) version:
```
root@AMD-PC:/usr/src/rtl_433# rtl_433 -d rtl_tcp://10.0.0.10:1234
rtl_433 version 21.12-126-g34d6eb4c branch master at 202208021325 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 187 out of 219 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 11
9 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-219 ]
rtl_tcp input from 10.0.0.10 port 1234
rtl_tcp connected to 10.0.0.10:1234 (Tuner: Unknown)
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
rtl_tcp: No such file or directory

Library error 0, exiting...
```

Seems to be an error on the Win-10 side:
```
rtl_tcp client connected from 10.0.0.10 port 62109
rtl_tcp received command SET_SAMPLE_RATE with 250000
rtl_tcp received command SET_GAIN_MODE with 0
rtl_tcp received command SET_FREQ with 433920000
```
when sending the first chunk; a whopping 262144 bytes!

gvanem, 2022-08-04:
> Do you want to PR the WSA startup, otherwise I'll just commit that.

Just do it.

> Is it because MSG_WAITALL doesn't work and we get an empty read from recv()?

Perhaps.

zuckschwerdt, 2022-08-04:
WSAStartup fixed with 300fcc3

gvanem, 2022-08-05:
Perhaps `mongoose.c` should be used for **all** such network stuff? I use it in my [Dump1090](https://github.com/gvanem/Dump1090/) project and it is rock solid. Although the API is in a flux.

zuckschwerdt, 2022-08-05:
I evaluated some different models to implement this (where / how to lock / stream / signal / block...).
Sadly Mongoose dropped out first, the callback nature and buffer shuffling for sending isn't suitable to higher volume streams.

gdt, 2023-10-02:
Looks like fixed in August 2022.  I'm therefore closing, but please feel free to speak up and explain why that's wrong if it is and we can reopen, start over, have a Discussion,etc.

closed 2023-10-02 by gdt