merbanan/rtl_433 #2140: Bresser Lightning Sensor.

AndreyMihaylov85, 2022-08-09:
Hello, 

recently i bought mentioned lightning sensor, but not supported yet. 

i followed the suggestion by recording sample here is the result. 

est mode active. Reading samples from file: g045_868M_1000k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected FSK package @0.088888s
Analyzing pulses...
Total count: 57, width: 71.66 ms (17915 S)
Pulse width distribution:
[ 0] count: 6, width: 1500 us [1456;1712] ( 375 S)
[ 1] count: 47, width: 484 us [480;492] ( 121 S)
[ 2] count: 4, width: 976 us [972;980] ( 244 S)
Gap width distribution:
[ 0] count: 47, width: 484 us [480;492] ( 121 S)
[ 1] count: 4, width: 1456 us [1456;1460] ( 364 S)
[ 2] count: 1, width: 1936 us [1936;1936] ( 484 S)
[ 3] count: 3, width: 968 us [968;972] ( 242 S)
[ 4] count: 1, width: 2428 us [2428;2428] ( 607 S)
Pulse period distribution:
[ 0] count: 10, width: 2116 us [1944;2432] ( 529 S)
[ 1] count: 41, width: 968 us [968;976] ( 242 S)
[ 2] count: 3, width: 1452 us [1452;1456] ( 363 S)
[ 3] count: 2, width: 2912 us [2908;2916] ( 728 S)
Pulse timing distribution:
[ 0] count: 10, width: 1484 us [1456;1712] ( 371 S)
[ 1] count: 94, width: 484 us [480;492] ( 121 S)
[ 2] count: 7, width: 972 us [968;980] ( 243 S)
[ 3] count: 1, width: 1936 us [1936;1936] ( 484 S)
[ 4] count: 1, width: 2428 us [2428;2428] ( 607 S)
[ 5] count: 1, width: 36060 us [36060;36060] (9015 S)
Level estimates [high, low]: 15927, 424
RSSI: -0.1 dB SNR: 15.7 dB Noise: -15.9 dB
Frequency offsets [F1, F2]: 22601, 2388 (+86.2 kHz, +9.1 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
view at https://triq.org/pdv/#AAB031060105CC01E403CC0790097C8CDC8191919191919191919191919191919191919191919191919091A1819190918091819355+AAB014060105CC01E403CC0790097C8CDCA1A28291919455+AAB01E060105CC01E403CC0790097C8CDC91919192A0919191819191919191919555
Attempting demodulation... short_width: 484, long_width: 484, reset_limit: 495616, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=484,l=484,r=495616'
pulse_demod_pcm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {222} f5 55 55 55 55 55 51 6e a2 e2 e8 6c e5 41 54 c5 5d 55 50 00 00 00 00 00 00 00 00 00

Flex decoder data:

X 'n=test,m=FSK_PCM,s=484,l=484,r=495616

time : 2022-08-08 18:26:49
model : test count : 1 num_rows : 1 rows :
len : 35 data : 6ae500000
codes : {35}6ae500000

is there already decoder which i can use or someone who have decode it ? 

thank you in advance,

zuckschwerdt, 2022-08-09:
It does not look like the Acurite lightning detector we have and does no exactly match the FineOffset-W57.

Get raw codes with
`-X 'n=Bresser-Lightning,m=FSK_PCM,s=484,l=484,r=5000,preamble=aa2dd4'`

The lenght is 8 to 10 bytes, not sure with the trailer in that codes above.

Record a few different codes and post here.

AndreyMihaylov85, 2022-08-09:
no data with -X 'n=Bresser-Lightning,m=FSK_PCM,s=484,l=484,r=5000,preamble=aa2dd4'

but following without preamble=aa2dd4

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-09 17:12:44
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 26           data      : 6ae5000
codes     : {26}6ae5000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-09 17:13:43
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 26           data      : 6ae5000
codes     : {26}6ae5000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-09 17:14:41
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 16           data      : 6ae5
codes     : {16}6ae5

zuckschwerdt, 2022-08-09:
Can you upload `g045_868M_1000k.cu8` as zip here?

AndreyMihaylov85, 2022-08-09:
here is, 

[g045_868M_1000k.zip](https://github.com/merbanan/rtl_433/files/9291581/g045_868M_1000k.zip)

zuckschwerdt, 2022-08-09:
I see. It was a confusion about the sample rate. Use this:
`-X 'n=Bresser-Lightning,m=FSK_PCM,s=120,l=120,r=2000,preamble=aa2dd4'`

zuckschwerdt, 2022-08-09:
This is only always the same code `5c5d0d9ca82a98abaaaa0000` we need a few different ones. Then we can check what changes, if there is a checksum and such.
Try to collect codes for a few days and then only show distinct (different) codes.

AndreyMihaylov85, 2022-08-09:
its also 5c5d0d9ca82a98abaaa8, but i will collect more codes and paste them here. 

thank you a lot !

AndreyMihaylov85, 2022-08-10:
Hello, 

i collect some more codes last night, not sure whether these help for further checks or not.   

5c5d0d9ca82a98abaaaa0000000
5c5d0d9ca82a98abaaa8
5c5d0d9ca855315755540000
78ba1b39505531575550
5c5d0d9ca82a98affffc
5c7ffffffffffffffffffffffffffffffffffe
5c5d0d9ca82a98bfffffffffffffffffffffff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-09 18:19:02
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 105          data      : 5c5d0d9ca82a98abaaaa0000000
codes     : {105}5c5d0d9ca82a98abaaaa0000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-09 18:44:23
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 78           data      : 5c5d0d9ca82a98abaaa8
codes     : {78}5c5d0d9ca82a98abaaa8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 03:00:38
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 94           data      : 5c5d0d9ca855315755540000
codes     : {94}5c5d0d9ca855315755540000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 03:09:25
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 77           data      : 78ba1b39505531575550
codes     : {77}78ba1b39505531575550
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 03:32:49
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 78           data      : 5c5d0d9ca82a98affffc
codes     : {78}5c5d0d9ca82a98affffc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 03:33:47
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 151          data      : 5c7ffffffffffffffffffffffffffffffffffe
codes     : {151}5c7ffffffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 03:43:32
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 153          data      : 5c5d0d9ca82a98bfffffffffffffffffffffff8
codes     : {153}5c5d0d9ca82a98bfffffffffffffffffffffff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

regards,

zuckschwerdt, 2022-08-10:
It might be two codes:
`5c5d0d9ca82a98abaaaa0`
`5c5d0d9ca82a98abaaa8`
but they could be the same, not sure.

Maybe simulate a lightning? Use a heavy motor near the sensor. Like a power drill or vacuum cleaner?

@MksRasp and @rct might know some trick to trigger a lightning sensor.

AndreyMihaylov85, 2022-08-10:
i follow your advice and use power drill which trigger a lighting sensor here is the results. 

time      : 2022-08-10 16:52:45
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 95           data      : c87e0d9ca9ca98a2aaaa0000
codes     : {95}c87e0d9ca9ca98a2aaaa0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:53:03
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 95           data      : 3ac80d9ca92a98afaaaa0000
codes     : {95}3ac80d9ca92a98afaaaa0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:53:12
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 95           data      : 216f0d9ca93a98afaaaa0000
codes     : {95}216f0d9ca93a98afaaaa0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:53:15
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 95           data      : 4ce00d9caeaa98abaaaa0000
codes     : {95}4ce00d9caeaa98abaaaa0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:54:13
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 95           data      : 4ce00d9caeaa98abaaaa0000
codes     : {95}4ce00d9caeaa98abaaaa0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:55:18
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 153          data      : 57470d9caeba98afffffffffffffffffffffff8
codes     : {153}57470d9caeba98afffffffffffffffffffffff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:56:41
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 95           data      : 7bae0d9cae8a98abaaaa0000
codes     : {95}7bae0d9cae8a98abaaaa0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:57:25
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 113          data      : 60090d9cae9a98abaaaa000000000
codes     : {113}60090d9cae9a98abaaaa000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:59:22
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 153          data      : 6009ffffffffffffffffffffffffffffffffff8
codes     : {153}6009ffffffffffffffffffffffffffffffffff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-10 16:59:32
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      :
len       : 94           data      : 227c0d9caeeab15755540000
codes     : {94}227c0d9caeeab15755540000

zuckschwerdt, 2022-08-10:
Very good. Here is [a BitBench with that data](https://triq.net/bitbench#c=c87e0d9ca9ca98a2aaaa0%5B16%3A52%3A45%5D&c=3ac80d9ca92a98afaaaa0%5B16%3A53%3A03%5D&c=216f0d9ca93a98afaaaa0%5B16%3A53%3A12%5D&c=4ce00d9caeaa98abaaaa0%5B16%3A53%3A15%5D&c=4ce00d9caeaa98abaaaa0%5B16%3A54%3A13%5D&c=7bae0d9cae8a98abaaaa0%5B16%3A56%3A41%5D&c=60090d9cae9a98abaaaa0%5B16%3A57%3A25%5D&c=227c0d9caeeab15755540%5B16%3A59%3A32%20broken%20code%3F%5D&f=8h8h%20&co=16&cl=64&cw=8).
It seems like there is a checksum in the first two bytes. Strange that it's in the front. It does not work as CRC but seems like one.

Maybe @rct can spot which fields encode things like distance and count.

anthyz, 2022-08-10:
This appears to follow the protocol of Bresser 7-in-1 messages (both OEM'd by CCL Electronics). It looks like a 16 bit digest followed by a 16 bit ID followed by a 12? bit BCD count and then the rest.

zuckschwerdt, 2022-08-10:
@anthyz oh, of course! very good find! After undoing the whitening (0xaa) it's LFSR-16 gen 8810 key abf9 final xor 899e
(link update with whitening)

anthyz, 2022-08-10:
From reading the manual apparently the sensor does some basic loud noise detection, too. No indication it's reported in a message, though. Might be good to experiment with that as well as triggering messages with different sensitivity levels selected.

zuckschwerdt, 2022-08-11:
I've update the BitBench link above. At least an event counter is visible.
There could be a strength or distance indicator in the middle and maybe flags (interference detected?) at the very end?

anthyz, 2022-08-11:
The sensitivity setting is not in the message, and it looks like there's occasionally a bad bit flip at the end  (see the two examples with digest 0x3b72). For distance, byte[7] could be a candidate. I wonder if experimenting with the sensitivity set to high would get more varied results.

AndreyMihaylov85, 2022-08-12:
Good morning, 

messages i've posted in the begging were with sensitivity level - **High**, 

> i follow your advice and use power drill which trigger a lighting sensor here is the results.
>
> time : 2022-08-10 16:52:45 model : Bresser-Lightning count : 1 num_rows : 1 rows : len : 95 data : [...]

AndreyMihaylov85, 2022-08-15:
@zuckschwerdt, what might be the decoder for distance and eventually for distance as per suggestion above for byte [7]. im in area where the forecast for the next days is heavy thunderstorms,  which could be good opportunity for testing. 

thank you in advance,

zuckschwerdt, 2022-08-15:
Sorry, I don't have any experience with lightning sensors. Log all data if you can and maybe note down some prominent events (with the "every 3 seconds between lightning and thunder equals a kilometer" estimate).

matthias-bs, 2022-08-15:
Just a wild guess, but IF the sensor uses an "AS3935 Franklin Lightning Sensor IC" by AMS, the datasheet (e.g. https://www.digikey.de/htmldatasheets/production/1124289/0/0/1/as3935.html) might provide some insights about the kind of data and encoding to be expected.

AndreyMihaylov85, 2022-08-15:
> Just a wild guess, but IF the sensor uses an "AS3935 Franklin Lightning Sensor IC" by AMS, the datasheet (e.g. https://www.digikey.de/htmldatasheets/production/1124289/0/0/1/as3935.html) might provide some insights about the kind of data and encoding to be expected.

here is the picture of the board, there is RFID transponder coil ma5532-ae,  google says _Developed to work with austriamicrosystems AS3935 Franklin Lightning Sensor IC_ (e.g. https://www.coilcraft.com/en-us/products/rf/rfid-transponders/x-y-axis-transponder-coil-2mhz/ma5532/) 
[image]

matthias-bs, 2022-08-15:
I would like to bet on what kind of IC U3 is...
And we have nice little labelled test points (I²C interface) on the PCB! There shines a little light into the black box!

matthias-bs, 2022-08-15:
And there is a serial interface (RX, TX), too! Maybe it has something to tell you, too!

anthyz, 2022-08-15:
Excellent finds! According to Table 17 in section 8.9.3 of the AS3935 datasheet there are 16 possible distance values. The byte[7] values seen so far in the bitbench data (01, 05, and 08) are consistent with this table.

AndreyMihaylov85, 2022-08-15:
more data from the heavy thunderstorms we just had + video :) 

https://user-images.githubusercontent.com/110809838/184691686-d7a2e7e2-96a9-4662-abaf-8e88b6de5ecb.MP4

[data.docx](https://github.com/merbanan/rtl_433/files/9338990/data.docx)

zuckschwerdt, 2022-08-15:
Removing dups and broken codes [this BitBench](https://triq.net/bitbench#c=51b10d9cacca98bffff8%20%5B19%3A49%3A33%5D&c=4a160d9cacda98abaaaa%20%5B19%3A49%3A45%5D&c=d55b0d9cac2a98abaaaa%20%5B19%3A49%3A54%5D&c=cefc0d9cac3a98abaaaa%20%5B19%3A49%3A56%5D&c=cefc0d9cac3a98abaaaa%20%5B20%3A06%3A30%5D&c=cefc0d9cac3a98abaaaa%20%5B20%3A10%3A24%5D&f=DIGEST%3A8h8h%20ID%3A8h8h%20CTR12h%20%3F4h8h%20KM8d%208h8h%20EOF%20&x=aa&cw=4) shows 4 events, the first at 21 km, then 3 overhead ;)
Nothing apparent in the remaining 12+16 bits. Might be good enough for a decoder, then watch the unknown fields long term, maybe some pattern reveals someday.

matthias-bs, 2022-08-15:
> more data from the heavy thunderstorms we just had + video :)
> 1660585221297348.MP4
> 
> [data.docx](https://github.com/merbanan/rtl_433/files/9338990/data.docx)

Nice test setup! :-)

matthias-bs, 2022-08-15:
Probably there isn't anything else in it. From the "4Cast PRO WIFI Weather Center with 7in1 Outdoor Sensor" (PN 7003210/7803210/7903210) manual, section *4.3.13.2 LIGHTNING DETECT MODE (OPTIONAL SENSOR)*:
[image]
*Period of time since last lightning* and *Number of lightnings per hour* are probably added by the base station.

The energy provided by the AS3935 might be contained in the radio message. From the IC datasheet:
```
8.9.2  Energy Calculation
If the received signal is classified as lightning, the energy is calculated. The result of the energy calculation
is then stored in the registers REG0x06[4:0], REG0x05[7:0] and  REG0x04[7:0]. 
This value is just a pure number and has no physical meaning.
```
Well, if it has no physical meaning and is just a relative value (due to noise adjustment etc.), then it wouldn't be too useful anyway.

zuckschwerdt, 2022-08-15:
The last 16 bit (mostly `0000` seen) is not included in the digest. Any message with anything there needs that stripped to pass the checksum. So we are now only looking to find a meaning for the  12 bits of fixed `032`.

matthias-bs, 2022-08-15:
Any chance to identify the battery status bit by connecting an adjustable power supply?

matthias-bs, 2022-08-15:
The next most interesting things could be the sensitivity switch position and the noise level, which might be available in the message.

matthias-bs, 2022-08-16:
Just for completeness: 
https://ams.com/search#/AS3935
links to
https://www.sciosense.com/products/wireless-sensor-nodes/as3935-franklin-lightning-sensor-ic/

rct, 2022-08-16:
Sorry didn't see the mention from a number of days ago.

I've got experience with the two Acurite devices that have the AS3935 in them.  There is also an RPI board that has an AS3935 that I haven't tried.

At least on the two Acurite devices, the strike count is non-volatile.  On the 6045M sensor it is a 7 bit counter (...so wraps back to 0 after 127.)  The module embedded in the Atlas uses a 9 bit counter. 

On the Acurite devices, and IIRC the AS3935 too, the distance to the edge of the storm estimation resets to the maximum value when the power is cycled.  So you might want to try removing/reinserting the batteries after you've recorded some strikes to see what changes and hopefully you'll find the distance estimation field.

The 16 possible distance values for the AS3935 are: 1, 5, 6, 8, 10, 12, 14, 17, 20, 24, 27, 31, 34, 37, 40, 63

63 indicates that no distance estimate has been done/no lightning has been detected since power up.   40KM is the stated maximum detection range, so the highest valid distance estimate you should see is 40.   1 indicates that the storm is overhead.   (Acurite consoles show that as 0 miles/km from what I've been told).

Two other bits that are in the Acurite messages to look for:

- RFI detected - There is some sort of continual detection that is likely RF interference
- Active mode - The AS3935 has an active listening mode that uses more power. When in active mode the Acurite 6045 transmits every 8 seconds.  When in standby mode (I think they call it something like passive listening) the 6045 transmits only every 24 seconds. 

Hope this helps.

franki29, 2022-08-29:
> And there is a serial interface (RX, TX

I tested it by soldering RX TX and connected to a serial to usb (two different ones) converter with different baud rates, no outputs at my side. Even if you press the reset button, nothing.

MacH-21, 2022-09-06:
Hi All,

I do not have this sensor but looking at the pcb
Bresser = badged CCL Electronics 
RF chip U1 looks like Cmostek CMT2119A See Datasheet
this chip is also used by HopeRF RFM119

FYI  ..  CCL Electronics
https://www.fc.gov/oet/ea/fccid
Enter>> Grantee code  2ALZ7 .. press Search
should present CCL Electronics Ltd - Products
Lightning Detector is under 2ALZ7-3129A2103
under column named Display Exhibits click on Detail  .. select ok 
View Attachment of your choice.

Mac

franki29, 2022-09-08:
HI, did anyone wrote a test decoder yet?

franki29, 2022-09-08:
Hi, I made some tests using near empty battery and the messages are changing for  
Full Battery:
7369b508aaa290aaaaaa
Not so full Battery:
f6aab508aaaa90aaaaaa
Using this data in bitbench you can see the different at the magic ?032

[image]

franki29, 2022-09-08:
I also tried to write a decoder, but I am not a coder and so I could only copy data from the bresser 7 in 1 and made a long try and error, not knowing really what I am doing :-) 

Maybe some one can look at the code and check if I am totally wrong? Thanks

[bresser_lightning.zip](https://github.com/merbanan/rtl_433/files/9528085/bresser_lightning.zip)

franki29, 2022-09-14:
Hi looks like after the CTR comes the battery indicator. 8 for full 0 for battery low.
DIGEST:8h8h ID:8h8h CTR12h Batt4h ?8h KM8d 8h8h EOF

matthias-bs, 2023-01-19:
> Some more info: after reset, ? will show 3a, after ~1 hours it was changing to 32.

Could be the 'startup' bit at bit3 (for pairing with the station). But ~1 hour seems to be a little long for this purpose.

gdt, 2023-10-15:
What's the status and path forward?  Is someone working on a PR?

matthias-bs, 2023-10-15:
Hi,
I can provide a working decoder in a few days.
Regards,
Matthias

AndreyMihaylov85, 2024-02-24:
Hey,

i just try  the decoder, but no messages return from the sensor with `rtl_433  -f 868M`
Am i doing something wrong ? 

Exact sample rate is: 1000000.026491 Hz

many thanks in advance,

bilhert, 2024-02-25:
Hi I tried the decoder today unfortunatly I did not get a response either.

I used  the latest docker container which currently comes with rtl_433 version 23.11 (2023-11-28)
When using the following command:

rtl_433 -R249 -f 868.3M -X 'n=Bresser-Lightning,m=FSK_PCM,s=120,l=120,r=2000,preamble=aa2dd4'

I obtain the values:
time      : 2024-02-25 23:29:12
model     : Bresser-Lightning                      count     : 1             num_rows  : 1             rows      : 
len       : 95           data      : 3af79ce2aa82988daaaa0000
codes     : {95}3af79ce2aa82988daaaa000

When using "rtl_433 -f 868.3M -v -R249" I get the following output

Pulse data: 1 pulses
[  0] Pulse:   20, Gap: 10001, Period: 10021
[pulse_slicer_pcm] Exact bit width (in us) is 121.63 vs 124.00, 46 bit preamble
[bresser_lightning_decode] 153 too short
[pulse_slicer_pcm] Bresser lightning
codes     : {218}55555555555516ea1d7bce7155414c46d5550000000000000000000
Pulse data: 121 pulses
[  0] Pulse:    0, Gap:    2, Period:    2

franki29, 2024-02-26:
Hi, I tested with my device and also it does not work.

Changing for testing
 if (len < sizeof(msg) * 8 ) to  if (len < sizeof(msg) * 6) the ABORT_LENGTH will not shown but still no data.

Changing the DECODE_FAIL_MIC to:

int chk    = (msg[0] << 8) | msg[1];
      int digest = lfsr_digest16(&msg[2], 8, 0x8810, 0xabf9);
         //fprintf(stderr, "DIGEST %04x vs %04x (%04x) \n", chk, digest, chk ^ digest);
      if (((chk ^ digest) != 0x899e) && ((chk ^ digest) != 0x8b9e)) {
        decoder_logf(decoder, 2, __func__, "Digest check failed %04x vs %04x (%04x)", chk, digest, chk ^ digest);
        

it shows:
[image]

Maybe different sensors around?

ProfBoc75, 2024-02-27:
Hi @franki29 :  I guess you're on the good way.

Confirmed by another github repo, https://github.com/RFD-FHEM, @elektron-bbs already decoded this lightning device, protocol 131, added in January.

Extract:

    # BRESSER Blitzsensor Art.No.: 7009976, Hersteller CCL Electronics LTD Model C3129A
    # ---------------------------------------------------------------------------------
    # The sensor transmits immediately when a flash is detected, otherwise approximately every 60 seconds.
    # 0         1         
    # 01234567890123456789
    # --------------------
    # 73FB2866AAA298AAAAAA   original message
    # 8BF082CC138832120000   message after all nibbles xor 0xA
    # CCCCIIIIcccB?FDD????
    # C = LFSR-16 digest, generator 0x8810, key 0xABF9, final xor 0x899E
    # I = station ID
    # c = 3 nibbles lightning count, BCD
    # B = flags, 4 bit
    #     Bit:    0123
    #             1000
    #             b???
    #             b:   1 bit batteryState, 1 = ok, 0 = low
    #             ?:   3 bit unknown always 000
    # ? = 1 nibble, unknown, always 0x3 (type?)
    # F = flags, 4 bit
    #     Bit:    0123
    #             1010 xor 0xA = 0000
    #             r???
    #             r:   1 bit device reset, 1 after device reset
    #             ?:   3 bit unknown always 000
    # D = 2 nibbles last distance, 0 after reset, BCD
    # ? = 4 nibbles, unknown, always 0x0

Few gaps, one at the msg length , should be 10 byte / 80 bit (20 nibbles in RFD-FHEM) and not 25 bytes as other Bresser sensors.
If we add the preamble + msg, the total data length is at least 112 bit.

[Line 39](https://github.com/merbanan/rtl_433/blob/ac1247521145613bb13480f6916aa692c4256aab/src/devices/bresser_lightning.c#L39)

    uint8_t msg[10];   // not 25

another at digest level, should be , LFSR-16 gen 8810 key abf9 final xor 899e , as already proposed by @zuckschwerdt [here](https://github.com/merbanan/rtl_433/issues/2140#issuecomment-1210977583)

[Line 81](https://github.com/merbanan/rtl_433/blob/ac1247521145613bb13480f6916aa692c4256aab/src/devices/bresser_lightning.c#L81)

    int digest = lfsr_digest16(&msg[2], 8, 0x8810, 0xabf9);
    if ((chk ^ digest) != 0x899e) {

The rest is the same, type = 0x3 after dewhitened, so normal that we have type = 0x9 here before the XOR 0xA.

For me the rtl_433 is easy to update accordingly.

franki29, 2024-02-27:
Hi @ProfBoc75 ,yes I can confirm that it works:
[image]

Some more information:
codes     : {218}55555555555516ea428dda8454d94c5655550000000000000000000
Pulse data: 127 pulses
[  0] Pulse:    0, Gap:    2, Period:    2
[  1] Pulse:    4, Gap:    2, Period:    6
[  2] Pulse:    3, Gap:    4, Period:    7
[  3] Pulse:    2, Gap:    4, Period:    6
[  4] Pulse:    2, Gap:    2, Period:    4
[  5] Pulse:    3, Gap:    2, Period:    5
[  6] Pulse:    3, Gap:    4, Period:    7
[  7] Pulse:    3, Gap:    1, Period:    4
[  8] Pulse:    2, Gap:    2, Period:    4
[  9] Pulse:    1, Gap:    1, Period:    2
[ 10] Pulse:    2, Gap:    2, Period:    4
[ 11] Pulse:    1, Gap:    4, Period:    5
[ 12] Pulse:    2, Gap:    5, Period:    7
[ 13] Pulse:    2, Gap:    1, Period:    3
[ 14] Pulse:    3, Gap:    2, Period:    5
[ 15] Pulse:    1, Gap:    2, Period:    3
[ 16] Pulse:    2, Gap:    2, Period:    4
[ 17] Pulse:    2, Gap:    3, Period:    5
[ 18] Pulse:    2, Gap:    5, Period:    7
[ 19] Pulse:    5, Gap:    3, Period:    8
[ 20] Pulse:    4, Gap:    3, Period:    7
[ 21] Pulse:    2, Gap:    4, Period:    6
[ 22] Pulse:    4, Gap:    6, Period:   10
[ 23] Pulse:    2, Gap:    2, Period:    4
[ 24] Pulse:    4, Gap:    3, Period:    7
[ 25] Pulse:    1, Gap:    3, Period:    4
[ 26] Pulse:    2, Gap:    3, Period:    5
[ 27] Pulse:    5, Gap:    5, Period:   10
[ 28] Pulse:    3, Gap:    1, Period:    4
[ 29] Pulse:    3, Gap:    4, Period:    7
[ 30] Pulse:    2, Gap:    2, Period:    4
[ 31] Pulse:    3, Gap:    1, Period:    4
[ 32] Pulse:    2, Gap:    5, Period:    7
[ 33] Pulse:    2, Gap:    3, Period:    5
[ 34] Pulse:    3, Gap:    3, Period:    6
[ 35] Pulse:    2, Gap:    2, Period:    4
[ 36] Pulse:    3, Gap:    4, Period:    7
[ 37] Pulse:    2, Gap:    2, Period:    4
[ 38] Pulse:    3, Gap:    3, Period:    6
[ 39] Pulse:    2, Gap:    2, Period:    4
[ 40] Pulse:    4, Gap:    3, Period:    7
[ 41] Pulse:    2, Gap:    3, Period:    5
[ 42] Pulse:    3, Gap:    1, Period:    4
[ 43] Pulse:    3, Gap:    2, Period:    5
[ 44] Pulse:    3, Gap:    3, Period:    6
[ 45] Pulse:    3, Gap:    1, Period:    4
[ 46] Pulse:    3, Gap:    2, Period:    5
[ 47] Pulse:    4, Gap:    3, Period:    7
[ 48] Pulse:    3, Gap:    2, Period:    5
[ 49] Pulse:    3, Gap:    2, Period:    5
[ 50] Pulse:    4, Gap:    1, Period:    5
[ 51] Pulse:    3, Gap:    2, Period:    5
[ 52] Pulse:    4, Gap:    1, Period:    5
[ 53] Pulse:    3, Gap:    2, Period:    5
[ 54] Pulse:    4, Gap:    1, Period:    5
[ 55] Pulse:    4, Gap:    3, Period:    7
[ 56] Pulse:    6, Gap:    4, Period:   10
[ 57] Pulse:    3, Gap:    1, Period:    4
[ 58] Pulse:    4, Gap:    2, Period:    6
[ 59] Pulse:    2, Gap:    3, Period:    5
[ 60] Pulse:    4, Gap:    2, Period:    6
[ 61] Pulse:    4, Gap:    2, Period:    6
[ 62] Pulse:   12, Gap:    3, Period:   15
[ 63] Pulse:    3, Gap:    1, Period:    4
[ 64] Pulse:    3, Gap:    3, Period:    6
[ 65] Pulse:    3, Gap:    2, Period:    5
[ 66] Pulse:   19, Gap:    1, Period:   20
[ 67] Pulse:    3, Gap:  123, Period:  126
[ 68] Pulse:  121, Gap:  121, Period:  242
[ 69] Pulse:  121, Gap:  121, Period:  242
[ 70] Pulse:  121, Gap:  121, Period:  242
[ 71] Pulse:  121, Gap:  122, Period:  243
[ 72] Pulse:  121, Gap:  121, Period:  242
[ 73] Pulse:  121, Gap:  121, Period:  242
[ 74] Pulse:  120, Gap:  121, Period:  241
[ 75] Pulse:  121, Gap:  122, Period:  243
[ 76] Pulse:  121, Gap:  121, Period:  242
[ 77] Pulse:  121, Gap:  121, Period:  242
[ 78] Pulse:  121, Gap:  121, Period:  242
[ 79] Pulse:  121, Gap:  122, Period:  243
[ 80] Pulse:  121, Gap:  121, Period:  242
[ 81] Pulse:  120, Gap:  122, Period:  242
[ 82] Pulse:  120, Gap:  121, Period:  241
[ 83] Pulse:  121, Gap:  122, Period:  243
[ 84] Pulse:  121, Gap:  121, Period:  242
[ 85] Pulse:  121, Gap:  121, Period:  242
[ 86] Pulse:  121, Gap:  121, Period:  242
[ 87] Pulse:  121, Gap:  122, Period:  243
[ 88] Pulse:  121, Gap:  121, Period:  242
[ 89] Pulse:  120, Gap:  122, Period:  242
[ 90] Pulse:  121, Gap:  120, Period:  241
[ 91] Pulse:  121, Gap:  364, Period:  485
[ 92] Pulse:  121, Gap:  121, Period:  242
[ 93] Pulse:  242, Gap:  121, Period:  363
[ 94] Pulse:  363, Gap:  121, Period:  484
[ 95] Pulse:  121, Gap:  121, Period:  242
[ 96] Pulse:  121, Gap:  243, Period:  364
[ 97] Pulse:  121, Gap:  483, Period:  604
[ 98] Pulse:  122, Gap:  120, Period:  242
[ 99] Pulse:  123, Gap:  362, Period:  485
[100] Pulse:  242, Gap:  121, Period:  363
[101] Pulse:  364, Gap:  121, Period:  485
[102] Pulse:  242, Gap:  121, Period:  363
[103] Pulse:  121, Gap:  121, Period:  242
[104] Pulse:  122, Gap:  483, Period:  605
[105] Pulse:  121, Gap:  364, Period:  485
[106] Pulse:  121, Gap:  121, Period:  242
[107] Pulse:  121, Gap:  121, Period:  242
[108] Pulse:  121, Gap:  242, Period:  363
[109] Pulse:  243, Gap:  121, Period:  364
[110] Pulse:  242, Gap:  242, Period:  484
[111] Pulse:  120, Gap:  122, Period:  242
[112] Pulse:  121, Gap:  243, Period:  364
[113] Pulse:  242, Gap:  363, Period:  605
[114] Pulse:  121, Gap:  121, Period:  242
[115] Pulse:  121, Gap:  121, Period:  242
[116] Pulse:  242, Gap:  243, Period:  485
[117] Pulse:  121, Gap:  120, Period:  241
[118] Pulse:  122, Gap:  120, Period:  242
[119] Pulse:  121, Gap:  121, Period:  242
[120] Pulse:  122, Gap:  121, Period:  243
[121] Pulse:  122, Gap:  121, Period:  243
[122] Pulse:  120, Gap:  121, Period:  241
[123] Pulse:  121, Gap:  122, Period:  243
[124] Pulse:  120, Gap: 8999, Period: 9119
[125] Pulse:    2, Gap:    2, Period:    4
[126] Pulse:    6, Gap:    5, Period:   11

bilhert, 2024-02-28:
@ProfBoc75 I just tested the code and it seems to work, cool! However I tried to integrate the sensor in home assistant using mqtt output. The device gets automatically detected with mqtt-auto-discovery but only shows the strike count as a sensor.
Using mqtt explorer I saw that the value is reported as  "distance_km"

Looking into the mqtt-auto-discovery script showed that that this key is not picked up however "storm_dist_km" is, which is also used in fineoffset_wh31l.c, should be picked up in the latest version and is used for lightning distance

Other keys that are used for lightning distance are "storm_dist" and "strike_distance" both are used in acurite.c however these report in miles. Which would imply that the suffix "_km" should be added to keep it clear.  

I would suggest to either use "storm_dist_km" since it is already used and support by mqtt auto-discovery or use strike_distance_km to keep the key more consistent with strike_count.

 mqtt-auto-discovery script is found here https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py)

zuckschwerdt, 2024-02-28:
I have already prepared that change and I'll PR it now.

gdt, 2024-06-02:
Seems like this is fixed, closing as PR is merged.

closed 2024-06-02 by gdt