merbanan/rtl_433 #2152: Acurite 515 TX sensor not recognized by rtl_433

marcvl64, 2022-08-17:
With option -vvvv this is what the log shows:

> rtl433tomqtt | Badly formatted OS v2.1 message: {114} 05 56 ab 6a a8 15 5a ad aa a0 55 6a b6 aa 80
> rtl433tomqtt | Unrecognized Msg in OS v3: {114} 05 56 ab 6a a8 15 5a ad aa a0 55 6a b6 aa 80
> rtl433tomqtt | calibeur_rf104_callback: DECODE_FAIL_SANITY data all 0x00 or 0xFF
> rtl433tomqtt | Exact bit width (in us) is 610.29 vs 500.00 (pulse width 227.43 vs 248.00), 7 bit preamble
> rtl433tomqtt | Exact bit width (in us) is 610.86 vs 500.00 (pulse width 227.43 vs 248.00), 7 bit preamble
> rtl433tomqtt | Exact bit width (in us) is 610.86 vs 500.00 (pulse width 227.43 vs 248.00), 7 bit preamble
> rtl433tomqtt | Possible DSC Contact: bitbuffer:: Number of rows: 1
> rtl433tomqtt | [00] {12} c0 00 : 11000000 0000
> rtl433tomqtt | row 0 bit count 12
> rtl433tomqtt | DSC row 0 invalid bit count 12
> rtl433tomqtt | acurite_txr_decode: bitbuffer:: Number of rows: 3
> rtl433tomqtt | [00] {50} c0 7a 48 11 ac 3f 00 : 11000000 01111010 01001000 00010001 10101100 00111111 00
> rtl433tomqtt | [01] {50} c0 7a 48 11 ac 3f 00 : 11000000 01111010 01001000 00010001 10101100 00111111 00
> rtl433tomqtt | [02] {50} c0 7a 48 11 ac 3f 00 : 11000000 01111010 01001000 00010001 10101100 00111111 00
> rtl433tomqtt | acurite_txr_decode: row 0 bits 50, bytes 7
> rtl433tomqtt | acurite_txr_decode: skipping wrong len
> rtl433tomqtt | acurite_txr_decode: row 1 bits 50, bytes 7
> rtl433tomqtt | acurite_txr_decode: skipping wrong len
> rtl433tomqtt | acurite_txr_decode: row 2 bits 50, bytes 7
> rtl433tomqtt | acurite_txr_decode: skipping wrong len
> rtl433tomqtt | acurite_986_decode: row 0 bits 161, bytes 5
> rtl433tomqtt | acurite_986_decode: skipping wrong len
> rtl433tomqtt | radiohead_ask_extract: preamble not found
> rtl433tomqtt | acurite_00275rm_decode: bitbuffer:: Number of rows: 3
> rtl433tomqtt | [00] {50} c0 7a 48 11 ac 3f 00 : 11000000 01111010 01001000 00010001 10101100 00111111 00
> rtl433tomqtt | [01] {50} c0 7a 48 11 ac 3f 00 : 11000000 01111010 01001000 00010001 10101100 00111111 00
> rtl433tomqtt | [02] {50} c0 7a 48 11 ac 3f 00 : 11000000 01111010 01001000 00010001 10101100 00111111 00
> rtl433tomqtt | philips_aj3650_decode: wrong number of bits (162)
> rtl433tomqtt | radiohead_ask_extract: preamble not found
> rtl433tomqtt | AmbientWeather-TX8300: wrong size (161 bits)
> rtl433tomqtt | philips_aj7010_decode: wrong number of bits (0)
> rtl433tomqtt | gt_tmbbq05_decode: Possible Quigg BBQ: bitbuffer:: Number of rows: 1
> rtl433tomqtt | [00] {161} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> rtl433tomqtt | norgo_decode: wrong size of bit per row 1
> rtl433tomqtt | Exact bit width (in us) is 1226.00 vs 1072.00 (pulse width 631.00 vs 536.00), 4 bit preamble
> rtl433tomqtt | Exact bit width (in us) is 1217.00 vs 1072.00 (pulse width 623.00 vs 536.00), 4 bit preamble
> rtl433tomqtt | Exact bit width (in us) is 1219.00 vs 1072.00 (pulse width 624.00 vs 536.00), 4 bit preamble
> rtl433tomqtt | cotech_36_7959_decode: Couldn't find preamble
> rtl433tomqtt | secplus_v2_callback: DECODE_FAIL_SANITY

zuckschwerdt, 2022-08-17:
Please add a sample file (`-S unknown` option) as zip here.

zuckschwerdt, 2022-08-17:
Also which version (first output line of rtl_433) is this? Is it possibly from 2020? The 515 support was added with 8cdb56d
The data shown above works.:

```
rtl_433 -vvvv -R 40 -y '{50}3f85b7ee53c0c0 {50}3f85b7ee53c0c0 {50}3f85b7ee53c0c0'

Verifying test data with device Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 899 Rain, 3N1, Atlas.
acurite_txr_decode: row 0 bits 50, bytes 7
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-08-17 11:08:53
model     : Acurite-515  id        : 122
channel   : AR           Battery   : 1             Temperature: 74.0 F       Integrity : CHECKSUM
```
…
(note that {50}3f85b7ee53c0c0 is just the inverse of {50}c07a4811ac3f00 from above)

marcvl64, 2022-08-17:
I used a prebuilt docker image and just installed in last week. So I was assuming to be on the latest version. 
The first few output lines are:
`
Starting rtl433tomqtt ... done
Attaching to rtl433tomqtt
rtl433tomqtt    | rtl_433 version unknown inputs file rtl_tcp RTL-SDR
rtl433tomqtt    | Use -h for usage help and see https://triq.org/ for documentation.
rtl433tomqtt    | Trying conf file at "rtl_433.conf"...
rtl433tomqtt    | Trying conf file at "/home/user/.config/rtl_433/rtl_433.conf"...
rtl433tomqtt    | Reading conf from "/home/user/.config/rtl_433/rtl_433.conf".
rtl433tomqtt    | Publishing MQTT data to mosquitto port 1883
rtl433tomqtt    | Publishing device info to MQTT topic "rtl_433/e466a871c32a/devices[/type][/model][/subtype][/channel][/id]".
rtl433tomqtt    | Publishing events info to MQTT topic "rtl_433/e466a871c32a/events".
rtl433tomqtt    | Publishing states info to MQTT topic "rtl_433/e466a871c32a/states".
rtl433tomqtt    | Registering protocol [1] "Silvercrest Remote Control"
`

zuckschwerdt, 2022-08-17:
> rtl_433 version unknown

That does sound like an older version, hard to tell. Can you post exactly which docker image that is?

marcvl64, 2022-08-17:
hertzg/rtl_433:alpine-3.12-latest

marcvl64, 2022-08-17:
Ok, i switched the image to
`hertzg/rtl_433:latest`
and that seems to have fixed the issue. The version now shows as 21.12. Thanks for your help. Seems like I did have an older image.

closed 2022-08-17 by marcvl64