merbanan/rtl_433 #2159: Bresser Weather Station - sporadic reception issues

Thoro, 2022-08-26:
Hi,

I have a Bresser 7-in-1 and two 6-in-1 sensors (pool + indoor) and I'm experiencing outages in the data, even though the weatherstation and the antenna are completely stationary. The outages take from 10 min to 1 hour.

What could be causing this?
How could I improve the reception?
Is it a software issue, or a radio issue?

Attached is the SNR Graph over the past 7 days, the big outages can be ignored, there the power was off. Also the graph with noise values.

[image]
[image]

I'm using a RTL-SDR with this antenna: https://www.amazon.de/gp/product/B07CZMCJ8Z/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

Config options for rtl_433 are: RTL_433_OPTIONS= -f 868.3M -M level -M noise -s 1024k -Y autolevel -Y minmax -Y magest -F mqtt://127.0.0.1:1883

zuckschwerdt, 2022-08-26:
Could it be interference? Can you record (`-w file_868.3M_1024k.cu8 -T 120`) a few minutes of the band while the reception isn't working?

Thoro, 2022-08-26:
The weird part is, that it's not all sensors at the same time. All of them should be sending on 868.3M.

Here also the received messages chart:

[image]

I will try to record it.

Thoro, 2022-08-26:
Oh, so that they themselves create the interference. Is there anything I could do to improve such situations?

Thoro, 2022-08-27:
@zuckschwerdt Uploaded the file here: https://github.com/Thoro/rtl433-dumps/blob/master/file_868.3M_1024k_2.cu8.gz

zuckschwerdt, 2022-08-27:
There is no issue with that capture, the band looks fine and the Bresser-6in1 is found every 12 seconds.
Did you record that randomly or when you saw the reception as broken?

Thoro, 2022-08-27:
I recorded it when the message rate of the weather station Bresser-7in1 (!) fell

Thoro, 2022-08-27:
Here the specific time:
[image]

zuckschwerdt, 2022-08-27:
Oh sorry, okay, so there should be a 7-in-1 among the 6-in-1's there?
We see 22 transmissions in that capture (`rtl_433 -S all file_868.3M_1024k_2.cu8` to split).
Removing the 6-in-1 that leaves 12 unknown transmissions.

- g001_868.3M_1024k.cu8 (43019 samples) faint FSK
- g003_868.3M_1024k.cu8 (41943 samples) FSK (A)
- g004_868.3M_1024k.cu8 (147675 samples) FSK (B)
- g006_868.3M_1024k.cu8 (41943 samples) FSK (A)
- g008_868.3M_1024k.cu8 (41942 samples) FSK (A)
- g009_868.3M_1024k.cu8 (147672 samples) FSK (B)
- g011_868.3M_1024k.cu8 (41943 samples) FSK (A)
- g013_868.3M_1024k.cu8 (147675 samples) FSK (B)
- g015_868.3M_1024k.cu8 (41942 samples) FSK (A)
- g017_868.3M_1024k.cu8 (41943 samples) FSK (A)
- g018_868.3M_1024k.cu8 (147675 samples) FSK (B)
- g021_868.3M_1024k.cu8 (147712 samples) FSK (B)

(A) seems to be a 10 µs (100 kbps) PCM signal
(decode with `rtl_433 -Y minmax -R 0 -X 'n=name,m=FSK_PCM,s=10,l=10,r=100' g003_868.3M_1024k.cu8`)
(B) seems to be a 30 µs (33 kbps) PCM signal
(decode with `rtl_433 -Y minmax -R 0 -X 'n=name,m=FSK_PCM,s=30,l=30,r=200' g004_868.3M_1024k.cu8`)

Both are very fast and contain a long payload, not TPMS and unlikely to be a weather sensor.

The 7-in-1 would be a 125 µs (8 kpbs) PCM signal. And it's not seen anywhere in the capture.

Thoro, 2022-08-27:
interesting that there's soo many additional items, probably they are encoded and therefore not detected and not reported to the MQTT Broker?

Since I do not have any Wireless M-Bus devices they must be from somewhere in the neighborhood.

And based on the capture it would mean the station is actually not sending, or the antenna is blocked from the signal?

Interestingly enough, after the dump ended, it came back just as normal.

zuckschwerdt, 2022-08-27:
If you can, grab a sample of the 7-in-1. It might be more towards the edge of the band and thus unreliable to detect. Perhaps `-f 868.4M` or so might work better. But it could also just be the AGC on the RTL-SDR that doesn't recover fast enough for faint signals between those very frequent strong signals.

Thoro, 2022-08-27:
New sample with 7-in-1: https://github.com/Thoro/rtl433-dumps/blob/master/file_868.3M_1024k_3.cu8.gz

I confirmed with the command from above, and it printed the 7-in-1 samples, haven't seen any 6-in-1 though.

zuckschwerdt, 2022-08-27:
Ah ok, I did find that in the original capture now. It's just slightly to faint because the strong signals drown out the hardware AGC and our autolevel.

Compare the files below in https://triq.org/pdv/

Also run `rtl_433 -Y minlevel=-24 file_868.3M_1024k_2.cu8` to uncover the 7-in-1.
Your band looks perfectly noise-free, you might want to replace autolevel with a fixed lower level like that.

[g005_868.3M_1024k.cu8.zip](https://github.com/merbanan/rtl_433/files/9438493/g005_868.3M_1024k.cu8.zip)
[g117_868.3M_1024k.cu8.zip](https://github.com/merbanan/rtl_433/files/9438494/g117_868.3M_1024k.cu8.zip)

Thoro, 2022-08-27:
Thanks for analyzing this! Absolutly not sure how to read the spectrogram ..  - Does it mean the data is sent on 868.360 and 868.237 Mhz ? 

I will try with -Y minlevel=-24 and see if it helps!

Will report back tomorrow if it happened again.

zuckschwerdt, 2022-08-27:
>  not sure how to read the spectrogram

just note that both shapes are the same and the second one is fainter ;)
Nothing more to see there really, except that it's FSK, which means there are two Frequencies the carrier Shifts between to Key the data :)

Thoro, 2022-08-28:
Sadly it also happened again with -Y minlevel=-24.

Can I record samples, while also writing them to mqtt via -F ? Then I could record everything and pinpoint where the signal "cuts-out".

zuckschwerdt, 2022-08-28:
You can do that, but storage needed is 2 MB per second…

Pretty sure that we found the cause. The 7-in-1 signal is there but others around it are too strong.
Use `-M level` to keep an eye on the levels and go down to maybe `-Y minlevel=-28`

Thoro, 2022-08-28:
What's the idea regarding the -M level vs -Y minlevel ?

I have logs like this:

Aug 28 10:42:50 raspberrypi rtl_433[1868]: Current signal level -28.0 dB, estimated noise -33.0 dB
Aug 28 10:50:20 raspberrypi rtl_433[1868]: Current signal level -28.8 dB, estimated noise -33.0 dB
Aug 28 10:51:50 raspberrypi rtl_433[1868]: Current signal level -24.0 dB, estimated noise -32.6 dB
Aug 28 10:53:00 raspberrypi rtl_433[1868]: Current signal level -19.4 dB, estimated noise -32.5 dB

Should I try to match the current signal level?

zuckschwerdt, 2022-08-28:
It's `-Y autolevel` vs `Y minlevel=-N` -- if autolevel works, as in the output above all is well. If the levels get too high despite the clear band then fix them at a lower level (somewhere like -24 to -28). The `-M level` will just show you the signal level (RSSI) with each transmission:
```
model     : Bresser-7in1 id        : 32070
...
RSSI      : -21.1 dB     SNR       : 11.5 dB       Noise     : -32.6 dB
```

Thoro, 2022-08-28:
Sorry can't follow, the logs from above are with -Y minlevel=-24 and -Y level=-24

The noise level and estimated noise are always around -32.9 db (changes by +1).

```
Aug 28 11:01:50 raspberrypi rtl_433[7875]: Current noise level -32.9 dB, estimated noise -32.9 dB
Aug 28 12:08:00 raspberrypi rtl_433[7875]: Current noise level -33.2 dB, estimated noise -32.9 dB
Aug 28 12:13:20 raspberrypi rtl_433[7875]: Current noise level -33.2 dB, estimated noise -32.5 dB
```

I tried with -Y minlevel=-28 and had more frequent outages almost immediately. 

Can I further limited the received frequencies?
Based on the logs I would need 868.35 - 868.40  and 868.20 - 868.25

zuckschwerdt, 2022-08-28:
`-Y level=N` is not useful at all, don't use. The parameter is named `-Y minlevel=N`. Autolevel will set that and report, e.g. `adjusting minimum detection level to -15.2 dB` but you can just set a fixed level yourself (instead of autolevel). Set it too low and reception will suffer.

The `-f 868.3M` is already optimal.

Thoro, 2022-08-28:
Ok, removed the -Y level=N

Oh, okay:

```
Estimated noise level is -19.3 dB, adjusting minimum detection level to -16.3 dB
Estimated noise level is -21.0 dB, adjusting minimum detection level to -18.0 dB
Estimated noise level is -22.5 dB, adjusting minimum detection level to -19.5 dB
Estimated noise level is -23.8 dB, adjusting minimum detection level to -20.8 dB
Estimated noise level is -24.9 dB, adjusting minimum detection level to -21.9 dB
Estimated noise level is -26.7 dB, adjusting minimum detection level to -23.7 dB
Estimated noise level is -28.2 dB, adjusting minimum detection level to -25.2 dB
Estimated noise level is -29.3 dB, adjusting minimum detection level to -26.3 dB
Estimated noise level is -30.5 dB, adjusting minimum detection level to -27.5 dB
Estimated noise level is -31.5 dB, adjusting minimum detection level to -28.5 dB
Estimated noise level is -32.5 dB, adjusting minimum detection level to -29.5 dB
```

so that's the autolevel!

So setting it to -24 or so is way higher than autolevel would set it to.

Thoro, 2022-08-28:
I have made now an additional capture and listened at the same time to the mqtt topics.

The file is here:  https://github.com/Thoro/rtl433-dumps/blob/master/file_868.3M_1024k_4.cu8.gz

Every 12 seconds I got a message from the 6-in-1 Bresser Pool Thermometer
Every 45 seconds I got a message from the 6-in-1 Bresser Inside Thermometer

And there should be ever 12 seconds a message from the 7-in-1, but it was never logged!

I think the 7-in-1 would be at 58.2s, 70.2s, 83.2s 

and the stronger 6-in-1 (pool) at 62.6s, 74.7s, 86.7s   (still seen with -Y minlevel=-12 down to -34)

the slow 6-in-1 (indoor) at 79.3s, 139.37s (still seen with -Y minlevel=-13 down to -34)

Current settings are as following: RTL_433_OPTIONS= -f 868.3M -M level -M noise -s 1024k -Y minlevel=-24 -Y minmax -Y magest -F mqtt://127.0.0.1:1883

gdt, 2023-10-02:
What's the status on this?   Generally intermittent reception turns out to be not a bug, and collisions or noise.

closed 2024-06-04 by gdt