merbanan/rtl_433 #2160: Windows "RTL-SDR device driver is not available" but sdr# works fine

judahr, 2022-08-26:
I'm using the exe from the release: rtl_433-win-msvc-x64-21.12    It is just exiting without outputting anything meaningful.  Below are the outputs of things I have tried.  However, if I open sdr#, it seems to work.  

rtl_433.exe -d 0  
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\***\Downloads\rtl_433-win-msvc-x64-21.12\rtl_433.conf"...
Reading conf from "C:\Users\***\Downloads\rtl_433-win-msvc-x64-21.12\rtl_433.conf".
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]

rtl_433.exe -d help outputs this:

rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "C:\Users\***\Downloads\rtl_433-win-msvc-x64-21.12\rtl_433.conf"...
Reading conf from "C:\Users\***\Downloads\rtl_433-win-msvc-x64-21.12\rtl_433.conf".
                = Input device selection =
        RTL-SDR device driver is not available.
  [-d <RTL-SDR USB device index>] (default: 0)
  [-d :<RTL-SDR USB device serial (can be set with rtl_eeprom -s)>]
        To set gain for RTL-SDR use -g <gain> to set an overall gain in dB.
        SoapySDR device driver is not available.
  [-d ""] Open default SoapySDR device
  [-d driver=rtlsdr] Open e.g. specific SoapySDR device
        To set gain for SoapySDR use -g ELEM=val,ELEM=val,... e.g. -g LNA=20,TIA=8,PGA=2 (for LimeSDR).
  [-d rtl_tcp[:[//]host[:port]] (default: localhost:1234)
        Specify host/port to connect to with e.g. -d rtl_tcp:127.0.0.1:1234

rtl_433.exe -s2M -c NUL
rtl_433 version 21.12 branch  at 202112141644 inputs file rtl_tcp
Use -h for usage help and see https://triq.org/ for documentation.
Registered 176 out of 207 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-207 ]

Mindavi, 2022-10-19:
It appears the rtl-sdr device is not detected. Might be good to check if drivers are installed correctly (with zadig I think?)

gdt, 2023-10-02:
Please update to the latest versions of everything and feel free to open a discussion asking for help.

closed 2023-10-02 by gdt