merbanan/rtl_433 #2166: Efergy & rtl_433 not reporting current over 53.xx Amps

MksRasp, 2022-09-03:
I have Efergy 3 phase CT and transmitter and when I run (**rtl_433 -f 433.55M -R 36**) it report any reading below (53.xx Amps) but **NOT** the reading above (53.xx Amps) and the rtl_433 seems to be stuck until the current gets below (53.xx A) mark.
the efergy package does have screen and  when both central AC runs the screen shows (**21.xx KW**) and I should get reading around (**180 Amps**) from rtl_433.

how can I fix this?

merbanan, 2022-09-03:
Record a sample signal below and above 53.xx A. Then we can look into the actual message to see what we have missed.

MksRasp, 2022-09-04:
I have used this (**rtl_433 -f 433.55M -S all -T 120**) command to save the files (**g001** & **g016**) where **g001** is captured during the low Amps (below 53.xx) and **g016** when it got above (53.xx).
[g001&g016.zip](https://github.com/merbanan/rtl_433/files/9483776/g001.g016.zip)

MksRasp, 2022-09-04:
running (**rtl_433 -f433.55M -R 36:vv**)

**below 53.xx mark**

Battery   : 1            Current   : 5.25 A        Interval  : 18s
[00] {65} 0c c0 55 60 53 ff 03 d6 80
**consol LCD reading:0.577 kw**

Battery   : 1            Current   : 5.29 A        Interval  : 18s
[00] {65} 0c c0 55 60 54 aa 03 82 80
**consol LCD reading:0.582 kw**

**the code below was reported when power demand increased but I feel the code is not complete**
Battery   : 1            Current   : 14.76 A       Interval  : 18s
[00] {64} 0c c0 55 60 76 13 04 0e
**consol LCD reading:1.624 kw**

**Above 53.xx mark**

pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {25} ff ff ff 80 : 11111111 11111111 11111111 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {64} 0c c0 55 60 73 82 07 7d
**consol LCD reading:12.71 kw**

pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {26} ff ff ff c0 : 11111111 11111111 11111111 11
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {65} 0c c0 55 60 50 e5 08 be 80
**consol LCD reading:17.80 kw**

pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {26} ff ff ff c0 : 11111111 11111111 11111111 11
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {65} 0c c0 55 60 66 7e 08 6d 80
**consol LCD reading:22.55 kw**

pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {25} ff ff ff 80 : 11111111 11111111 11111111 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {65} 0c c0 55 60 65 0f 08 fd 80
**consol LCD reading:22.23 kw**

pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {26} ff ff ff c0 : 11111111 11111111 11111111 11
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] { 1} 80        : 1
pulse_demod_pwm(): Efergy e2 classic
bitbuffer:: Number of rows: 1
[00] {65} 0c c0 55 60 65 d4 08 c2 80
**consol LCD reading:22.40**

MksRasp, 2022-09-14:
any suggestion to overcome this problem?

zuckschwerdt, 2022-09-14:
Sorry, yes, was just a minor validation check.
Test with e.g. `rtl_433 -R 36:vv -y '{65}0cc055605c9408798'`

Thanks for reporting!

commit 2f5fac3d (2022-09-14): Fix Efergy-e2CT current above 53 Amp (closes #2166)
closed 2022-09-14 by zuckschwerdt