merbanan/rtl_433 #2171: Decoder for Vauno EN8822C-1 thermometer/hygrometer

gumbald, 2022-09-12:
I've captured multiple measurements from this sensor and began to establish the bit layout, looking to coordinate with writing a decoder! Will likely need some guidance on common checksums etc.

Relevant PR: https://github.com/merbanan/rtl_433_tests/pull/433

zuckschwerdt, 2022-09-12:
Your decoding notes looks really good! Can you collect [some codes into a BitBench](https://triq.net/bitbench#c=%7B42%7D8510f9740c40&c=%7B42%7Daf0fa27c01c0&f=ID%3A8h%20F%3A2b%20CH%3A2d%20TEMP%3A12d%20HUM%3A7d%20%3F%3A11b&cw=4) like this so we can take a guess at the trailing bits?

merbanan, 2022-09-12:
One of the 2 0-bits before the channel bits are probably low battery indicators. Try with bad batteries and see if anything changes.

Regarding the checksum there are tools to help out with that, we just need a set of different codes. Other then that look into the code some other temperature sensor and base your device of that. We'll help out with the details in the review process.

gumbald, 2022-09-12:
Thanks for the help! Been a while since I've done any C, but will give it a go.

I've added some more codes into bitbench, trying to vary one value at a time and get a range of values for each field.

zuckschwerdt, 2022-09-12:
Share the BitBench link when you are finished (no data is stored there).

gumbald, 2022-09-12:
Whoops, had copied but forgot to paste! [Here we go](https://triq.net/bitbench#c=%7B42%7D%2085%2010%20f9%2074%200c%2040&c=%7B42%7D%20af%200f%20a2%207c%2001%20c0&c=%7B42%7D%2085%2020%20f9%2074%200c%2080&c=%7B42%7D%20af%200f%2067%2072%200f%2080&c=%7B42%7D%2085%2000%20ff%206e%200f%20c0&c=%7B42%7D%20af%2001%204e%205a%200e%20c0&c=%7B42%7D%203d%2001%2024%207c%200a%2080&c=%7B42%7D%203d%2011%2024%207c%200a%20c0&c=%7B42%7D%203d%2021%2024%207c%200b%2000&f=ID%3A8h%20F%3A2b%20CH%3A2d%20TEMP%3A12d%20HUM%3A7d%20%3F%3A11b&cw=4)

zuckschwerdt, 2022-09-12:
Sort and look at
```
{42} 3d 01 24 7c 0 a8
{42} 3d 11 24 7c 0 ac
{42} 3d 21 24 7c 0 b0

```
that's (nibble) addition with carry as checksum. A [BitBench to show](https://triq.net/bitbench#c=%7B42%7D3d01247c0a8&c=%7B42%7D3d11247c0ac&c=%7B42%7D3d21247c0b0&c=%7B42%7D8500ff6e0fc&c=%7B42%7D8510f9740c4&c=%7B42%7D8520f9740c8&c=%7B42%7Daf014e5a0ec&c=%7B42%7Daf0f67720f8&c=%7B42%7Daf0fa27c01c&f=ID%3A8h%20F%3A2b%20CH%3A2d%20TEMP%3A12d%20HUM%3A7d%20%3F5b%20%3F%3A6h&cf=ADD&cl=32&cw=4) this mostly works.

gumbald, 2022-09-12:
I just put some bad batteries in and got this decoded repeatedly:

> time      : 2022-09-12 11:58:38
> model     : Conrad-S3318P ID        : 56
> Channel   : 1            Battery   : 1             Temperature: -14.50 F     Humidity  : 129 %
> Button    : 0            Integrity : CRC

When I look at S3318P - the bit layout seems to match! https://github.com/merbanan/rtl_433/blob/master/src/devices/s3318p.c

Maybe instead, how can this new device be collected using the same decoder?

zuckschwerdt, 2022-09-12:
Ah, nice. That's a common protocol, the esperanza_ews, kedsum share this. One of those will fit (with minor changes maybe).

zuckschwerdt, 2022-09-12:
> What determines which decoder gets used on a detected signal?

Every decoder with matching timings is tried and should decode or reject.

gumbald, 2022-09-12:
Do I have a first attempt and then create a PR into the main project (here)?

gdt, 2023-10-14:
Status?

gdt, 2024-06-04:
Looks like a PR that does this was merged.  Speak up if otherwise with status.

closed 2024-06-04 by gdt