merbanan/rtl_433 #2174: Decoding Smartwares remote model SH4-90152

louisnichols, 2022-09-15:
Hi!

I am trying to decode this remote. rtl_433 does not decode it by default.

However, if I run it with `rtl_433 -a 4`, I get something which seems to make some kind of sense. Below is the output from pressing 4 buttons consecutively:

```
rtl_433 version 21.05-105-g4cf34682 branch master at 202110112341 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 168 out of 198 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
*** signal_start = 298965, signal_end = 474469, signal_len = 175504, pulses_found = 528
Iteration 1. t: 68    min: 67 (127)    max: 69 (401)    delta 10
Iteration 2. t: 67    min: 66 (19)    max: 69 (509)    delta 1
Iteration 3. t: 67    min: 66 (2)    max: 69 (526)    delta 0
Distance coding: Pulse length 67

Short distance: 64, long distance: 318, packet distance: 2622

p_limit: 67
bitbuffer:: Number of rows: 8 
[00] {65} d4 aa d3 2d 2a aa ab 2b 00 
[01] {65} d4 aa d3 2d 2a aa ab 2b 00 
[02] {65} d4 aa d3 2d 2a aa ab 2b 00 
[03] {65} d4 aa d3 2d 2a aa ab 2b 00 
[04] {65} d4 aa d3 2d 2a aa ab 2b 00 
[05] {65} d4 aa d3 2d 2a aa ab 2b 00 
[06] {65} d4 aa d3 2d 2a aa ab 2b 00 
[07] {65} d4 aa d3 2d 2a aa ab 2b 00 
*** signal_start = 638309, signal_end = 813864, signal_len = 175555, pulses_found = 528
Iteration 1. t: 68    min: 67 (131)    max: 69 (397)    delta 17
Iteration 2. t: 67    min: 66 (28)    max: 69 (500)    delta 1
Iteration 3. t: 67    min: 66 (1)    max: 69 (527)    delta 0
Distance coding: Pulse length 67

Short distance: 64, long distance: 319, packet distance: 2622

p_limit: 67
bitbuffer:: Number of rows: 8 
[00] {65} d4 aa d3 2d 2a aa aa ab 00 
[01] {65} d4 aa d3 2d 2a aa aa ab 00 
[02] {65} d4 aa d3 2d 2a aa aa ab 00 
[03] {65} d4 aa d3 2d 2a aa aa ab 00 
[04] {65} d4 aa d3 2d 2a aa aa ab 00 
[05] {65} d4 aa d3 2d 2a aa aa ab 00 
[06] {65} d4 aa d3 2d 2a aa aa ab 00 
[07] {65} d4 aa d3 2d 2a aa aa ab 00 
*** signal_start = 988901, signal_end = 1259365, signal_len = 270464, pulses_found = 660
Iteration 1. t: 149    min: 90 (568)    max: 208 (92)    delta 18560
Iteration 2. t: 148    min: 90 (564)    max: 206 (96)    delta 4
Iteration 3. t: 147    min: 90 (563)    max: 205 (97)    delta 1
Iteration 4. t: 147    min: 90 (562)    max: 204 (98)    delta 1
Iteration 5. t: 147    min: 90 (562)    max: 204 (98)    delta 0
Pulse coding: Short pulse length 90 - Long pulse length 204

Short distance: 192, long distance: 1085, packet distance: 3547

p_limit: 147
bitbuffer:: Number of rows: 4 
[... 88 lines cut ...]
model     : Proove-Security                        House Code: 11999232
Channel   : 3            State     : OFF           Unit      : 1             Group     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-09-16 01:10:06
model     : Nexa-Security House Code: 11999232
Channel   : 3            State     : OFF           Unit      : 1             Group     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-09-16 01:10:06
model     : KlikAanKlikUit-Switch                  id        : 11999232
Unit      : 2            Group Call: No            Command   : Off           Dim       : No            Dim Value : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-09-16 01:10:06
model     : Proove-Security                        House Code: 11999232
Channel   : 3            State     : OFF           Unit      : 1             Group     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-09-16 01:10:06
model     : Nexa-Security House Code: 11999232
Channel   : 3            State     : OFF           Unit      : 1             Group     : 0

```
So... it could be that we are dealing with 2 different protocols from the switches and the remote, and that the wall plug recognizes both. Or, it could be that for some strange reason, rtl_433 does not recognize the remote in spite of using the same protocol as the switches.

Any suggestions?

The Smartwares devices seem to be the bane of all the 433MHz apps for Linux. rtl_433 has some trouble and rc-switch, pigpio, rp-rf all fail to decode them.

zuckschwerdt, 2022-09-16:
Don't use `-a`, but use `-A`.

Looks like your remote is the same 32 Manchester bits (64 half-bits), which is unusal and fits the Nexa/Proove/Kaku scheme exactly.

Upload some samples (`-S`) as zip here please.

louisnichols, 2022-09-17:
I tried `-A` but the output seemed to make less sense. :) Just joking.

The problem I have with all these options is that I have quite a few sensors around the house, so whenever I try some debugging, I always get a lot of unnecessary stuff. With `-A`, all that was difficult to follow.

From that point of view, this option `-S unknown` is quite ingenious. Perhaps it might be an idea to extend `-A` in a similar way. Just a thought.

Anyway, I am attaching the samples, generated with `rtl_433 -S unknown`. I pressed 4 buttons again. No idea why there are 5 files. I am hoping there are only these buttons in the capture. At the moment, I don't have any sensors running that are not decoded by rtl_433. Hope the neighbors don't, either.

Thank you for your support.
[smartwares_SH4-90152_rtl_433-S_unknown_capture_20220917.zip](https://github.com/merbanan/rtl_433/files/9590627/smartwares_SH4-90152_rtl_433-S_unknown_capture_20220917.zip)

zuckschwerdt, 2022-09-17:
Yes, "exclude output of known signals" and "analyze only unknown" would be interested but niche additions.

The usual workflow would be: capture samples, then drop all on https://triq.org/pdv/ to sort through, then analyze only the good ones.
I have planned to add the our pulse detector to pdv someday, to reduce the steps needed.

Here file 4 and 6 are empty.
[The signal looks like PPM](https://triq.org/pdv/#AAB04F060109D0013804B000DC28E06B34829392939293939293929392939293929293929393929293939293929293929393929392939293929392939293929392939293929392939293929392929393929455+AAB050060609D0013804B000DC28E06B3490929392939293939293929392939293929293929393929293939293929293929393929392939293929392939293929392939293929392939293929392929393929455+AAB050060109D0013804B000DC28E06B3490929392939293939293929392939293929293929393929293939293929293929393929392939293929392939293929392939293929392939293929392929393929555) with all pulses 300 µs, short gaps 200 µs, long gaps 1200 µs, sync gap of 2500 µs, and packet repeats at 10 ms.

So this should work: `-X 'n=Smartwares,m=OOK_PPM,s=200,l=1200,y=2500,r=12000,g=1500'`
Giving codes of:
- g001_433.92M_250k.cu8: {64}a955a65a55555656
- g002_433.92M_250k.cu8: {64}a955a65a55555556
- g003_433.92M_250k.cu8: {64}a955a65a55555659
- g005_433.92M_250k.cu8: {64}a955a65a55555559

A [BitBench with these codes](https://triq.net/bitbench#c=a955a65a55555656&c=a955a65a55555556&c=a955a65a55555659&c=a955a65a55555559&f=ID%3A8h8h%208h8h%208h8h%20BTN%3A8h8h%20&cw=4).
This looks too regular, so there is another layer of coding, Manchester (press the button in BitBench).

You get codes of
- `e0d30011`
- `e0d30001`
- `e0d30012`
- `e0d30002`

Which looks like exactly the Proove/Nexa/Kaku: the next to last digit is the "on/off" bit, the last ist the "unit" bits.

With `-R` select only the one that works best and that should be it :)
- `R 15` KlikAanKlikUit Wireless Switch
- `R 51` Proove / Nexa / KlikAanKlikUit Wireless Switch
- `R 96` Nexa

But I still need to review why not all samples are picked up reliably here.

zuckschwerdt, 2022-09-17:
Looks like your sender is just a bit slower than expected and the "sync" is not accepted. If you widen the tolerance from 200 to e.g. 250 (recompile, install) it works. I'll need to test if this is okay to change for other devices though.
https://github.com/merbanan/rtl_433/blob/master/src/devices/proove.c#L112

commit e3b91622 (2022-09-17): Changed Proove/Nexa/Kaku to support Smartwares SH4-90152 (closes #2174)
closed 2022-09-17 by zuckschwerdt