merbanan/rtl_433 #2176: 2 SDR Sticks, both freeze independently, one lacks WS80

fankyy, 2022-09-18:
I have some strange behavoiur with two SDR Sticks on a RPi 3. 

One is R820T2:
In the summer during hot weather, sometimes it didn't get any signal at all for multiple hours, but also at night.. the issue only comes up occasionally now, how can I debug when it happens? Could it be absolute humidity dependent?

The other SDR is a FC0012, both with their factory antennas only 10cm apart. It never got any signal from the WS80, but receives a new installed WHx080 on the same spot as the WS80, roughly 60 to 70m distance to the SDR antennas, quite flawlessly*. Worth mentioning that the WS80 signal should be way stronger than the WHx080.. 

*This stick also shows some hangups now, sometimes only for a minute, sometimes more than 1 hour. While the other one works fine currently. If it would be the signal, weewx would just show n/a, but when these occur, weewx stops completely until the stick works again. It's a remote station so I couldn't check the light indicators when this happens. Maybe rather an usb power issue? But then again, similar installations on my other locations never had this.

Could it be that the second SDR device uses an older driver which doesn't detect the WS80 signal? But when I start rtl433 the 213 device protocol seems to load. 

So that's 2 different issues, one is the SDR freeze, the other is the missing WS80 signal on the second SDR stick.

Any Ideas?

zuckschwerdt, 2022-09-18:
There is a watchdog, rtl_433 will exit if the device does not deliver data anymore. So I guess the device just gets deaf?
As first step I'd use SDR software to connect to the rtl_tcp output of rtl_433 to check the waterfall.
The `rtl_433 -F rtl_tcp` doesn't show "Starting rtl_tcp server" use the latest rtl_433 (uninstall the old, then compile, install).

gdt, 2023-09-30:
There has been an important bug fix for the FC0012.  Please update to current git master and retry.   Also log and graph frequency to see if you are drifting out of range.  Then feel free to open a discussion asking for help.

closed 2023-09-30 by gdt