merbanan/rtl_433 #2177: Has someone experience with Tire Moni/Tyredog TPMS sensors?

a8193, 2022-09-18:
Hi, I'm using Tire Moni TD-1000 TPMS sensors. These sensors are the same as the Tyredog sensors.
These sensors are using FSK modulation (yellow label).
Does anyone perhaps have experience with reading this type of sensor?
BR
Lodewijk

zuckschwerdt, 2022-09-18:
TPMS data mostly looks the same: 50 µs FSK Manchester.
Grab a sample, inspect, upload here as zip, see https://triq.org/rtl_433/ANALYZE.html
We can then pick the data apart.

a8193, 2022-09-18:
[analyse_20220918_2146.zip](https://github.com/merbanan/rtl_433/files/9595175/analyse_20220918_2146.zip)
Hi, in this example the pressure 
- start with 1,65 bar
- after ~4-8 sec drops to 0,86 bar
- after 2 sec drops to 0,62 bar
- after 4 sec drops to 0,00 bar
hope this is enough?
BR
Lodewijk

zuckschwerdt, 2022-09-19:
Run `-A` and open [one of the pdv links](https://triq.org/pdv/#AAB0110701003000900064000C01CC004000008455+AAB02C0701003000900064000C01CC004000008580808080808080808080808292A0808280A0808280A2A082A081B655). You'll see:
- 50 µs data rate
- 18 preamble bits
- a desync of 150 µs
- ~16 data bits

This should work: `-X 'n=name,m=FSK_MC_ZEROBIT,s=50,l=50,y=150,g=150,r=200'` *but doesn't*, the desync gets in the way I guess.

But that is too little data in a single packet. And there are too much packets.
We have seen this once before, each byte of the transmission could be in it's own packet.

Might be better to look at the raw PCM to verify. `-X 'n=name,m=FSK_PCM,s=50,l=50,y=150,g=150,r=200,preamble=5553'`
With the preamble match of 5553 and MC in the and we get a nice [pattern in this BitBench](https://triq.net/bitbench#c=%7B40%7D9a6555a591%5B1.03%5D&c=%7B40%7D996566a968%5B1.04%5D&c=%7B39%7D9a55999950%5B1.06%5D&c=%7B39%7D9a55a66a50%5B1.07%5D&c=%7B40%7D99a6599a69%5B1.08%5D&c=%7B39%7D999665a590%5B1.%5D&c=%7B39%7D99a696a690%5B1.11%5D&c=%7B40%7D9a96a59a68%5B1.12%5D&c=%7B4%7D8%5B1.13%5D&c=%7B40%7D9a996a5568%5B1.15%5D&c=%7B40%7D9a69999968%5B1.16%5D&c=&c=%7B40%7D9a6555a591%5B1.17%5D&c=%7B40%7D996566a968%5B1.1%5D&c=%7B40%7D9a55999951%5B1.20%5D&c=%7B40%7D9a55a66a51%5B1.21%5D&c=%7B40%7D99a6599a69%5B1.22%5D&c=%7B39%7D999665a590%5B1.24%5D&c=%7B39%7D99a696a690%5B1.25%5D&c=%7B40%7D9a96a59a69%5B1.26%5D&c=%7B40%7D9a99566568%5B1.2%5D&c=%7B40%7D9a996a5569%5B1.29%5D&c=%7B40%7D9a69999969%5B1.30%5D&c=&c=%7B40%7D9a6555a591%5B1.31%5D&c=%7B40%7D996566a968%5B1.33%5D&c=%7B39%7D9a55999950%5B1.34%5D&c=%7B39%7D9a55a66a50%5B1.35%5D&c=%7B40%7D99a6599a68%5B1.37%5D&c=%7B40%7D999665a591%5B1.38%5D&c=%7B39%7D99a696a690%5B1.39%5D&c=%7B40%7D9a96a59a68%5B1.40%5D&c=%7B40%7D9a99566568%5B1.42%5D&c=%7B40%7D9a996a5569%5B1.43%5D&c=%7B40%7D9a69999969%5B1.44%5D&c=&c=%7B40%7D9a6555a591%5B1.46%5D&c=%7B40%7D996566a968%5B1.47%5D&c=%7B39%7D9a55999950%5B1.48%5D&c=%7B40%7D9a55a66a51%5B1.%5D&c=%7B40%7D99a6599a68%5B1.51%5D&c=%7B40%7D999665a591%5B1.52%5D&c=%7B39%7D99a696a690%5B1.53%5D&c=%7B40%7D9a96a59a68%5B1.55%5D&c=%7B40%7D9a99566568%5B1.56%5D&c=%7B40%7D9a996a5569%5B1.57%5D&c=%7B40%7D9a69999969%5B1.5%5D&c=&c=%7B40%7D9a6555a591%5B1.60%5D&c=%7B7%7Dd0%5B1.61%5D&c=%7B39%7D9a55999950%5B1.62%5D&c=%7B40%7D9a55a66a51%5B1.64%5D&c=%7B40%7D99a6599a68%5B1.65%5D&c=%7B39%7D999665a590%5B1.66%5D&c=%7B39%7D99a696a690%5B1.6%5D&c=%7B40%7D9a96a59a68%5B1.69%5D&c=%7B40%7D9a99566569%5B1.70%5D&c=%7B40%7D9a996a5569%5B1.71%5D&c=%7B40%7D9a69999968%5B1.73%5D&c=%7B39%7D9a6555a590%5B1.74%5D&c=%7B40%7D996566a968%5B1.75%5D&c=%7B39%7D9a55999950%5B1.77%5D&c=%7B39%7D9a55a66a50%5B1.78%5D&c=%7B40%7D99a6599a69%5B1.79%5D&c=%7B39%7D999665a590%5B1.80%5D&c=%7B39%7D99a696a690%5B1.82%5D&c=%7B40%7D9a96a59a69%5B1.83%5D&c=%7B40%7D9a99566569%5B1.84%5D&c=%7B40%7D9a996a5569%5B1.86%5D&c=%7B40%7D9a69999969%5B1.87%5D&c=%7B39%7D9a6555a590%5B1.88%5D&c=%7B40%7D996566a969%5B1.%5D&c=%7B6%7Da0%5B1.91%5D&c=%7B39%7D9a55a66a50%5B1.92%5D&c=%7B40%7D99a6599a69%5B1.93%5D&c=%7B40%7D999665a591%5B1.95%5D&c=%7B39%7D99a696a690%5B1.96%5D&c=%7B40%7D9a96a59a68%5B1.97%5D&c=%7B40%7D9a99566569%5B1.9%5D&c=%7B40%7D9a996a5568%5B2.00%5D&c=%7B40%7D9a69999969%5B2.01%5D&c=%7B39%7D9a6555a590%5B2.02%5D&c=%7B40%7D996566a969%5B2.04%5D&c=%7B39%7D9a55999950%5B2.05%5D&c=%7B39%7D9a55a66a50%5B2.06%5D&c=%7B40%7D99a6599a69%5B2.0%5D&c=%7B39%7D999665a590%5B2.09%5D&c=%7B39%7D99a696a690%5B2.10%5D&c=%7B40%7D9a96a59a68%5B2.11%5D&c=%7B40%7D9a99566569%5B2.13%5D&c=%7B40%7D9a996a5569%5B2.14%5D&c=%7B40%7D9a69999968%5B2.15%5D&c=%7B39%7D9a6555a590%5B2.17%5D&c=%7B40%7D996566a968%5B2.18%5D&c=%7B40%7D9a55999951%5B2.19%5D&c=%7B39%7D9a55a66a50%5B2.20%5D&c=%7B40%7D99a6599a68%5B2.22%5D&c=%7B39%7D999665a590%5B2.23%5D&c=%7B40%7D99a696a691%5B2.24%5D&c=%7B40%7D9a96a59a68%5B2.26%5D&c=%7B40%7D9a99566568%5B2.27%5D&c=%7B40%7D9a996a5568%5B2.28%5D&c=%7B40%7D9a69999968%5B2.%5D&c=%7B39%7D9a6555a590%5B2.31%5D&c=%7B40%7D996566a969%5B2.32%5D&c=%7B39%7D9a55999950%5B2.33%5D&c=%7B39%7D9a55a66a50%5B2.35%5D&c=%7B6%7Da0%5B2.36%5D&c=%7B39%7D999665a590%5B2.37%5D&c=%7B39%7D99a696a690%5B2.3%5D&c=%7B40%7D9a96a59a68%5B2.40%5D&c=%7B40%7D9a99566569%5B2.41%5D&c=%7B40%7D9a996a5568%5B2.42%5D&c=%7B40%7D9a69999968%5B2.44%5D&c=%7B39%7D9a6555a590%5B2.45%5D&c=%7B40%7D996566a968%5B2.46%5D&c=%7B40%7D9a55999951%5B2.4%5D&c=%7B39%7D9a55a66a50%5B2.49%5D&c=%7B40%7D99a6599a69%5B2.50%5D&c=%7B39%7D999665a590%5B2.51%5D&c=%7B39%7D99a696a690%5B2.53%5D&c=%7B40%7D9a96a59a68%5B2.54%5D&c=%7B40%7D9a99566568%5B2.55%5D&c=%7B40%7D9a996a5569%5B2.57%5D&c=%7B39%7D9a6555a590%5B2.98%5D&c=%7B4%7D8%5B2.99%5D&c=%7B40%7D99a5959aa8%5B3.00%5D&c=%7B40%7D9a55aa6a51%5B3.02%5D&c=%7B40%7D99a6559a68%5B3.03%5D&c=%7B39%7D999669a590%5B3.04%5D&c=%7B40%7D99a696a691%5B3.06%5D&c=%7B40%7D9a96a59a69%5B3.07%5D&c=%7B40%7D9a99566568%5B3.08%5D&c=%7B39%7D9a696a5950%5B3.09%5D&c=%7B39%7D9999999a90%5B3.11%5D&c=&c=%7B39%7D9a6555a590%5B3.12%5D&c=%7B40%7D99656aa969%5B3.13%5D&c=%7B40%7D99a5959aa9%5B3.15%5D&c=%7B39%7D9a55aa6a50%5B3.16%5D&c=%7B40%7D99a6559a69%5B3.17%5D&c=%7B39%7D999669a590%5B3.1%5D&c=%7B39%7D99a696a690%5B3.20%5D&c=%7B7%7Dd0%5B3.21%5D&c=%7B40%7D9a99566569%5B3.22%5D&c=%7B39%7D9a696a5950%5B3.24%5D&c=%7B40%7D9999999a91%5B3.25%5D&c=&c=%7B39%7D9a6555a590%5B3.26%5D&c=%7B40%7D99656aa969%5B3.2%5D&c=%7B40%7D99a5959aa9%5B3.29%5D&c=%7B39%7D9a55aa6a50%5B3.30%5D&c=%7B40%7D99a6559a69%5B3.31%5D&c=%7B39%7D999669a590%5B3.33%5D&c=%7B39%7D99a696a690%5B3.34%5D&c=%7B40%7D9a96a59a69%5B3.35%5D&c=%7B40%7D9a99566569%5B3.3%5D&c=&c=%7B39%7D9a696a5950%5B3.38%5D&c=%7B39%7D9999999 [... cut ...] %7D99656aa968%5B4.41%5D&c=%7B40%7D99a5959aa8%5B4.42%5D&c=%7B39%7D9a55aa6a50%5B4.44%5D&c=%7B40%7D99a6559a69%5B4.45%5D&c=%7B7%7Dd0%5B4.46%5D&c=%7B40%7D99a696a691%5B4.4%5D&c=%7B40%7D9a96a59a69%5B4.49%5D&c=%7B40%7D9a99566568%5B4.50%5D&c=%7B39%7D9a696a5950%5B4.51%5D&f=2b%204h%204d%208h%20&d=MC&cw=4).
Looks like 2 fixed bits (still part of the preamble/sync) and then two bytes.
I've marked a few blocks and the second nibble seems to count, but the start and sequence or data part is not really obvious.

a8193, 2022-09-21:
@franki29 only from 1 sensor

franki29, 2022-09-22:
Hi, is this really FSK?
If I open your samples in Audacity it is looking strange.

The first one in the picture is a Bresser Sensor with FSK.
The second one is your second file and it look completely different.
@zuckschwerdt am I right that it does not look like FSK?

[image]

zuckschwerdt, 2022-09-22:
The packets are tiny and many. In the broad overview packets might look like OOK. But zoom into one packet and you see two frequencies, one close to DC:

[image]

You can also use the spectrogram of https://triq.org/pdv/ and spot two freqs, offset by 50kHz (the red marker is DC):

[image]

a8193, 2022-09-22:
I've also made a dump of the second sensor. I only have two sensors because I'm using these sensors for my caravan.
Regarding also the yellow label on this second sensor, I expected the same type of modulation.
However it seems different from sensor 1, but I cannot determine which modulation it is.
Can you help again?
[analyse_sensor2_20220922.zip](https://github.com/merbanan/rtl_433/files/9629136/analyse_sensor2_20220922.zip)

franki29, 2022-09-23:
Hi ,you capture more than only your sensor, like file g002_433.92M_250k.cu8 is a inFactory-TH.
g001 contains your sensor, g004 nothing, g005 your sensor, g006 nothing, g007 your sensor, g008 nothing g009 nothing.

According to the manual of your sensor, your sensor can also detect the temperature. So we should find in the data of your sensor the temperature.
According to the inFactory-TH it was 52,5 F (~11 Celsius). When you record the data, was your caravan also outside?
So we can search for the temperature of ~ 52,5F in the data ?

I have to say that I am not a expert and I try to learn how to decode the data, sorry for the silly questions.
 
[image]

zuckschwerdt, 2022-09-23:
@franki29 to summarize: decode raw PCM and use MC and format in a BitBench as above. Then find the 10 packets of each message (presumed), the first nibble and last byte should be data and checksum (not sure which is which). The result looks like this:

```
d-32 9-79 c-a8 c-5c b-ad a-32 b-76 e-2d e-51 e-c1 d-a9
d-32 9-79 c-a8 c-5c b-ad a-32 b-76 e-2d e-51 e-c1 d-a9
d-32 9-79 c-a8 c-5c b-ad a-32 b-76 e-2d e-51 e-c1 d-a9
d-32 9-79 c-a8 c-5c b-ad a-32 b-76 e-2d e-51 e-c1 d-a9
d-32 9-79 c-a8 c-5c b-ad a-32 b-76 e-2d e-51 e-c1 d-a9
d-32 9-79 c-a8 c-5c b-ad a-32 b-76 e-2d e-51 e-c1 d-a9
d-32 9-f9 b-2f c-dc b-2d a-b2 b-76 e-2d e-51 d-c8 a-ae
d-32 9-f9 b-2f c-dc b-2d a-b2 b-76 e-2d e-51 d-c8 a-ae
d-32 9-f9 b-2f c-dc b-2d a-b2 b-76 e-2d e-51 d-c8 a-ae
d-32 9-f9 b-2f c-dc b-2d a-b2 b-76 e-2d e-51 d-c8 a-ae
```

We will need a ramp of values (like pressure slowly changing) over 5 to 10 minutes. That way we get a steady progression of values changing and can spot the data fields (mostly changing by 1 each time)  and checksum (changing randomly).

a8193, 2022-09-24:
@franki29 
no silly questions :). But you're right, these sensors are registring both temperature and pressure.
The TPMS sensors are during the tests only inside my house. Temperature of a sensor during pressue measurement is ~21-23 degr C. 
The inFactory-TH is somewere outside (at one of my neighbors).

@zuckschwerdt 
Concerning the pressure:
I understand what you mean with a ramp of values. However testing with slowly changing the pressure downwards is allways going in steps of ~ 0,2 bar lower (as far as I can see on the Tire Moni display). The timing of the rtl_433 output is always in time sync with the change on the Tire Moni display.
pre-conclusion: pressure changes are only measured in steps of ~0,2 bar. Changes are send directly.

Concerning the temperature:
Changing the temperature of the sensor will allways take ~30-60 seconds before the temperature on the display is updated. So I assume the temperature sensor is only sending it's changes periodic.

Using the options -X 'n=name,m=FSK_PCM,s=50,l=50,y=150,g=150,r=200,preamble=5553' is only registering the changes of the pressure. Changes of temperature are not registered with these settings (even without preamble).

Therefore I've run the -A unknown while changing the sensors temperature by using a hairdryer. See uploaded files (special at g005_433.92M_250k.cu8). For me it looks like the temperature data isn't modulate by using FSK but by using ASK modulation. Can you confirm this?
And which -X settings are needed to read these temperature data?

All in all these sensors seems very complicated to read, compared to other vendors. However I hope you can help me a bit further again.

ps Tire Moni seems a brand of Tyredog. Same devices.

[20220924-sensor3-temp1.zip](https://github.com/merbanan/rtl_433/files/9639318/20220924-sensor3-temp1.zip)

franki29, 2022-09-25:
Hi a8193, g005 is again the inFactory-TH  Sensor.
Try rtl_433 -r g005_433.92M_250k.cu8

franki29, 2022-09-25:
Hi a8193, can you tell me the pressure of the "heated" sensor?
It looks like you still get data not only from one sensor.

franki29, 2022-09-26:
Hi a8193, I am guessing that g001.. and g004 are two different sensors.
Is it possible to get some data from only one sensor by removing the antenna of your receiver and record some data close to your sensor?

Maybe I am totally wrong, but the last data you send are very close to the ford sensors.
I modified the Ford sensor decoder in RTL_433 and get some data, that looks not bad, but I am not sure if the data that I can see are correct. At least I can see two ID's with different pressure and one with Temperature and one without.

[tpms_tire_moni_test.zip](https://github.com/merbanan/rtl_433/files/9644783/tpms_tire_moni_test.zip)

As I already told, I am not a expert and try to learn, I can be at the "wooden way" as we say in Germany (so totally wrong)
Br.

zuckschwerdt, 2022-09-26:
The two working samples from `20220924-sensor3-temp1.zip` have the same timing but much longer packets. They read:
`aaaaaaaaa9caaacaccb32acaad2b4cb4cb53534acd3332b53328`
`aaaaaa72aab3552d2ab34d4cad54d2d4d4d2b34cccad4acd0`
which MC decoded is:
`808a50831a69dd8b54750`
`80be60b747d9dd8b54713`
but from just two codes it really is not visible where the fields are. If we assume the ID in the first bits then these must be different sensors (or maybe the data is encrypted).

a8193, 2022-09-26:
> Hi a8193, g005 is again the inFactory-TH Sensor. Try rtl_433 -r g005_433.92M_250k.cu8

I know, this sensor is somewhere in the neighbourhood. How can I ignore this sensor within my tests?

a8193, 2022-09-26:
@franki29 and @zuckschwerdt Every measurement I've been done was always only with one sensor. As said I've only two sensors and the not-used sensor is always wrapped in aluminium foil and therefore 'invisible' so it's always one sensor which can be seen.
I'll do some new tests for each sensor with lowering pressure and a test with changing temperature. I'll write down the value on the Tire Moni display with each change. I'll upload these files when done. 
BR Lodewijk

zuckschwerdt, 2022-09-26:
You can use `-R 0 -X 'n=tyredog,m=FSK_PCM,s=50,l=50,y=150,g=150,r=200,preamble=5553'` and you will only see the TPMS codes. Just post those codes then.

a8193, 2022-09-29:
Hi, hereby an additional test for each sensor while decreasing the pressure slowly.

test sensor P_bar T_dC  P_psi   T_dF     pck00 pck01 pck02 pck03 pck04 pck05 pck06 pck07 pck08 pck09 pck10
11-1   #3   1,96  26  28,42745  78,8  |  d-32  d-39  a-88  8-1c  f-8b  b-0e  b-76  e-2d  e-51  9-9f  a-d3
11-3   #3   1,41  25  20,45036  77,0  |  d-32  9-79  f-d8  d-ec  c-d3  f-ce  b-76  e-2d  e-51  8-87  b-eb
11-4   #3   1,17  25  16,96945  77,0  |  d-32  d-b9  a-1f  a-bc  8-13  b-8e  b-76  e-2d  e-51  8-9e  d-fc
11-6   #3   0,00  24    0,00000  75,2  |  d-32  d-39  8-c2  d-6c  c-9f  b-0e  b-76  e-2d  e-51  f-9d  e-fd
1        #3   0,00  23    0,00000  73,4  |  d-32  e-c9  b-3e  b-4c  8-31  f-b7  b-76  e-2d  e-51  f-60  8-06
21      #3   0,00  23    0,00000  73,4  |  d-32  9-79  c-82  b-cc  b-a5  a-32  b-76  e-2d  e-51  a-63  e-1b

12-1   #4   2,17  25  31,47325  77,0  |  e-33  c-12  a-ea  9-0e  9-41  c-43  a-75  d-77  f-61  b-37  c-99
12-2   #4   1,86  25  26,97707  77,0  |  e-33  8-d2  f-a3  e-5e  d-01  8-83  a-75  d-77  f-61  9-08  a-80
12-3   #4   1,69  25  24,51142  77,0  |  e-33  8-52  f-3a  b-ae  d-81  8-03  a-75  d-77  f-61  8-07  9-e9
12-4   #4   1,34  25  19,43509  77,0  |  e-33  8-52  f-32  9-8e  d-81  8-03  a-75  d-77  f-61  e-6f  f-c1
12-5   #4   1,10  25  15,95418  77,0  |  e-33  8-d2  f-c9  a-1e  d-01  8-83  a-75  d-77  f-61  d-66  e-aa
12-6   #4   0,86  24  12,47327  75,2  |  e-33  8-52  e-40  f-6e  d-89  8-03  a-75  d-77  f-61  c-65  a-5b
12-7   #4   0,00  24    0,00000  75,2  |  e-33  9-62  a-6a  f-ee  d-89  8-03  a-75  d-77  f-61  e-ef  f-c1
23      #4   0,00  23    0,00000  73,4  |  e-33  f-c2  8-bb  c-3e  d-63  8-83  a-75  d-77  f-61  d-22  c-8a
24      #4   0,00  23    0,00000  73,4  |  e-33  c-92  a-17  b-2e  f-cf  c-c3  a-75  d-77  f-61  c-1a  8-4a

in the above table I've presented testset 11 (with sensor 3) and 12 (with sensor 4). 
The columns are:
test 	= my testnumber
sensor	= sensor number (I only have 2 sensors, marked as sensor 3 and 4)
P_bar	= pressure in bar according the Tire Moni display
T_dC	= temperature in degrees Celsius according the Tire Moni display
P_psi	= calculated pressure in psi (= bar * 14,5038)
T_dF	= temperature in degrees Fahrenheit (= (T_dC * 1,8) + 32
pck00 - pck10 = the packets are allways in the format: "10 n p# bb" where
	n = nibble
	p# = packetnumber, 00 - 10
	bb = byte
This results for instance for test 12-6 is in the sequence:
10 e 00 33
10 8 01 52
10 e 02 40
10 f 03 6e
10 d 04 89
10 8 05 03
10 a 06 75
10 d 07 77
10 f 08 61
10 c 09 65
10 a 10 5b

All these tests are generated by only changing the pressure for the respective sensor.
Changing the temperature results in a different packetform as above mentioned (longer).

In the above table only the nibble and byte are represented, seperated by a hyphen.
My conclusions so far:
- This data is only generated when the pressure changes. When changing the temperature there is a different dataformat send.
- For each sensor the data of packet00/06/07/08 is always the same for a sensor.
- Furthermore the second nibble of the byte in packet 01/03/05 is always the same for a sensor.
- with pressure 0 and equal temperature the data is different
- there seems no logical increment/relation between the different measurement 
- therefore it seems logical this data is encrypted or encoded somehow

Some other suggestions to break this 'chaos'?

BR 
Lodewijk 

[test11-sensor3-pres - part1.zip](https://github.com/merbanan/rtl_433/files/9678963/test11-sensor3-pres.-.part1.zip)
[test11-sensor3-pres - part2.zip](https://github.com/merbanan/rtl_433/files/9678964/test11-sensor3-pres.-.part2.zip)
[test12-sensor4-pres - part1.zip](https://github.com/merbanan/rtl_433/files/9678965/test12-sensor4-pres.-.part1.zip)
[test12-sensor4-pres - part2.zip](https://github.com/merbanan/rtl_433/files/9678966/test12-sensor4-pres.-.part2.zip)

franki29, 2022-10-01:
Hi Lodewijk, great work. While you had a look at the data for pressure change, I had a look at the other data.
It looks like the have the same amount of data.
Here the [bitbench for all](https://triq.net/bitbench#c=aaaaaaaa72aab3552d4b2b534b4d352d52d4d2b34cccacb3552&c=5555555554e55566aa5a9a56aa96996a5aa5a9a56699995965aa0&c=5555555554e55565aa599aa956959a695aa5a9a56699995965a40&c=aaaaaa9caaacad4b34b35352ab2d3354b534acd3332b2d328&c=aaaaaaaaa9caaad4b2cd4ccab54b2ab2b2d3533353534ab333548&c=eaaaaaaaa9caaad4b2cd54ab4d4b2ab2b2d3533353534ab333348&c=55555555539555a66599aa65aa95596665a6a666a6a695699990&c=2aaaaaaaa9caaad2b2cb55352d4d2ab4b2d3533353534ab4d2c8&c=55555555539555aa6595a5996a99556a65a6a666a6a695695aa9&c=2aaaaaaaa9caaad332cd2cb54d4aacb332d3533353534ab4ad488&c=55555555539555a965995559aa966aa565a6a666a6a6956965a8&c=aaaaaaaaa9caaad54acb4ccd554cb55532d3533353534ab2ab340&f=8h%208h%208h%208h%208h%208h%2016h8h%208h%20&a=Preamble&m=5553&d=MC&cw=4) data that I have found 

The problem I have: the packages pck06 pck07 pck08 seems to be the ID for sensor 1 in your result text seems to be 
b-76  e-2d  e-51 = 762d51
In the data that I have send to bitbench Sensor 1 has:
9d8b54

So I do not know what we need to do to match the data, but it looks like that the data packages have same lengths, no matter how the sensor sends the information.

Shure that @zuckschwerdt can help out

Br. Frank

franki29, 2022-10-01:
Hi Lodewijk, one more question:
For sensor 4, did you warm up the sensor with a hairdresser or something else?
Br Frank

a8193, 2022-10-01:
As it seems the Tire Moni sensors have a different dataformat for pressure and for temperature, I've also done some additional testing by only changing the temperature.
The initial pressure for these tests is about 1 bar but is changing a little bit due to increasing the temperature. This has something to do with the law of Boyle-Gay Lussac (PxV)/T=C  and some additional thermo dynamic rules.
These tests are done with the respective sensor in a pressure sprayer (for the 1 bar pressure) and using a hairdryer for heating up the pressure sprayer and with it also the sensor.
Always only one sensor a time is 'visible the second sensor is wrapped in aluminium foil.

The format of the temperature data is, as @franki29 also has mentioned, in the format:
AA BB CC DD EE FF GGGGGG HH I

In the cu8-files I've seen the pressure changes were generating codes in the same format as described in my update from 2 days ago.

...
test	sensor	P_bar	T_dC	P_psi	T_dF	AA	BB	CC	DD	EE	FF	GGGGGG	HH	I
32-1	#3	1,00	25	14,50380	77,0	80	aa	79	1b	25	f7	9d8b54	4d	7
32-2	#3	1,07	27	15,51907	80,6	80	ba	68	93	31	e7	9d8b54	59	6
32-3	#3	1,07	28	15,51907	82,4	80	ae	78	87	2f	f7	9d8b54	77	3
32-4	#3	1,07	31	15,51907	87,8	80	ae	78	8b	29	f7	9d8b54	4d	3
32-5	#3	1,07	35	15,51907	95,0	80	9e	48	9f	61	c7	9d8b54	59	0
32-6	#3	1,07	37	15,51907	98,6	80	9e	48	83	6d	c7	9d8b54	51	0
32-7	#3	1,13	40	16,38929	104,0	80	8e	58	37	67	d7	9d8b54	6c	2
														
34-1	#3	1,03	26	14,93891	78,8	80	8e	58	eb	03	d7	9d8b54	77	2
34-2	#3	1,03	29	14,93891	84,2	80	8e	58	6f	0d	d7	9d8b54	4d	0
34-3	#3	1,10	31	15,95418	87,8	80	8e	58	73	09	d7	9d8b54	45	0
34-4	#3	1,10	34	15,95418	93,2	80	be	68	67	43	e7	9d8b54	50	7
														
33-1	#4	0,96	26	13,92365	78,8	80	d4	b3	57	86	54	dd5dd8	52	2
33-2	#4	0,96	27	13,92365	80,6	80	c4	83	7b	b4	64	dd5dd8	54	2
33-3	#4	0,96	29	13,92365	84,2	80	c4	83	7f	b8	e4	dd5dd8	6f	0
33-4	#4	0,96	33	13,92365	91,4	80	f4	93	63	d0	f4	dd5dd8	78	7
33-5	#4	0,96	35	13,92365	95,0	80	d4	b3	07	f4	d4	dd5dd8	70	2
33-6	#4	1,00	37	14,50380	98,6	80	d4	b3	3b	f8	d4	dd5dd8	48	2
33-7	#4	1,00	39	14,50380	102,2	80	d4	b3	3f	fc	d4	dd5dd8	40	2
...

in the above table I've presented testset 32 and 34 (with sensor 3) and 33 (with sensor 4).
The columns are:
test = my testnumber
sensor = sensor number (I only have 2 sensors, marked as sensor 3 and 4)
P_bar = pressure in bar according the Tire Moni display
T_dC = temperature in degrees Celsius according the Tire Moni display
P_psi = calculated pressure in psi (= bar * 14,5038)
T_dF = temperature in degrees Fahrenheit (= (T_dC * 1,8) + 32
AA = byte 1 -> it seems this is always 0x80 regardless of the sensor number
BB = byte 2
CC = byte 3
DD = byte 4
EE = byte 5
FF = byte 6
GG = byte 7-9 -> it seems this byterange is unique for each sensor
HH = byte 10
I = nibble 21

As far as I can see, nowhere is a logical increasing of data visible between the different tests. Also testresults with the same result are not comparable.
I wouldn't be surprised if this data is encrypted. How can we try to decode this?

a8193, 2022-10-01:
...
| First Header  | Second Header |
| ------------- | ------------- |
| Content Cell  | Content Cell  |
| Content Cell  | Content Cell  |
...

a8193, 2022-10-01:
...
test|sensor|P_bar|T_dC|P_psi|T_dF|AA|BB|CC|DD|EE|FF|GGGGGG|HH|I
----|---|---|---|---|---|----|----|----|----|----|----|--------|----|--
32-1|#3|1,00|25|14,50380|77,0|80|aa|79|1b|25|f7|9d8b54|4d|7
...

franki29, 2022-10-01:
Hi Lodewijk, so for me it looks like the second data format that I was looking at might work.
At least for the ID , Pressure and Temperature. It could be that the formula for pressure and temperature are not OK.
Please try the second test decoder and compare it with the display of your receiver.

For test11 :

model     : Tire Moni    type      : TPMS          id        : 9d8b54
Pressure  : 0.55 Bar     Temperature: 22.0 C       

model     : Tire Moni    type      : TPMS          id        : 9d8b54
Pressure  : 0.00 Bar     Temperature: 18.0 C       

For test12

model     : Tire Moni    type      : TPMS          id        : dd5dd8
Pressure  : 0.55 Bar     Temperature: 41.0 C    

model     : Tire Moni    type      : TPMS          id        : dd5dd8
Pressure  : 0.55 Bar     Temperature: 37.0 C       
  
model     : Tire Moni    type      : TPMS          id        : dd5dd8
Pressure  : 0.41 Bar     Temperature: 18.0 C       

[tpms_tiremoni.zip](https://github.com/merbanan/rtl_433/files/9690762/tpms_tiremoni.zip)

Br. Frank

franki29, 2022-10-01:
I update the[ Bitbench](https://triq.net/bitbench#c=aaaaaaaa72aab3552d4b2b534b4d352d52d4d2b34cccacb3552&c=5555555554e55566aa5a9a56aa96996a5aa5a9a56699995965aa0&c=5555555554e55565aa599aa956959a695aa5a9a56699995965a40&c=aaaaaa9caaacad4b34b35352ab2d3354b534acd3332b2d328&c=aaaaaaaaa9caaad4b2cd4ccab54b2ab2b2d3533353534ab333548&c=eaaaaaaaa9caaad4b2cd54ab4d4b2ab2b2d3533353534ab333348&c=55555555539555a66599aa65aa95596665a6a666a6a695699990&c=2aaaaaaaa9caaad2b2cb55352d4d2ab4b2d3533353534ab4d2c8&c=55555555539555aa6595a5996a99556a65a6a666a6a695695aa9&c=2aaaaaaaa9caaad332cd2cb54d4aacb332d3533353534ab4ad488&c=55555555539555a965995559aa966aa565a6a666a6a6956965a8&c=aaaaaaaaa9caaad54acb4ccd554cb55532d3533353534ab2ab340&f=8h%208h%208h%208h%208h%208h%20ID%3A16h8h%20TEMPERATURE%20F%3A8d%20PRESSURE%3A4d%20&a=Preamble&m=5553&d=MC&cw=4) and hope that this are the right values. CRC is missing and other possible values like moving yes/no

a8193, 2022-10-01:
@franki29 I'm afraid your format isn't the right one. In my table you can see there is no direct relation between de values in column T_dC (which are incrementing within every test) and column HH, which seems to change randomly.

btw In the table from 4 hours ago I've added a column with the original values from the cu8 files.

a8193, 2022-10-01:
> Hi Lodewijk, so for me it looks like the second data format that I was looking at might work. At least for the ID , Pressure and Temperature. It could be that the formula for pressure and [...]

How do I add this c-file to my rtl_433 config? btw I'm running rtl_433 in a docker container

a8193, 2022-10-01:
> Hi Lodewijk, one more question: For sensor 4, did you warm up the sensor with a hairdresser or something else? Br Frank

In test11 and test12 I've only increased the pressure. However this often influences the temperature a little bit.
BR
Lodewijk

franki29, 2022-10-02:
> > Hi Lodewijk, so for me it looks like the second data format that I was looking at might work. At least for the ID , Pressure and Temperature. It could be that the formula for pressure and temperature are not OK. Please try the second test decoder and compare it with the display of your receiver.
> > For test11 :
> > model : Tire Moni type : TPMS id : 9d8b54 Pressure : 0.55 Bar Temperature: 22.0 C
> > model : Tire Moni type : TPMS id : 9d8b54 Pressure : 0.00 Bar Temperature: 18.0 C
> > For test12
> > model : Tire Moni type : TPMS id : dd5dd8 Pressure : 0.55 Bar Temperature: 41.0 C
> > model : Tire Moni type : TPMS id : dd5dd8 Pressure : 0.55 Bar Temperature: 37.0 C
> > model : Tire Moni type : TPMS id : dd5dd8 Pressure : 0.41 Bar Temperature: 18.0 C
> > [tpms_tiremoni.zip](https://github.com/merbanan/rtl_433/files/9690762/tpms_tiremoni.zip)
> > Br. Frank
> 
> How do I add this c-file to my rtl_433 config? btw I'm running rtl_433 in a docker container BR Lodewijk

Hi Lodewijk, I am using a Laptop with Linux and cloned the git repository. I am not using a pre build version. For your docker container I do not know how to do. I would recommend to read the https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md where you can find the way to build the newest version. 
At https://github.com/merbanan/rtl_433/wiki/Decipher-a-new-device-and-write-a-decoder#declare-the-new-decoder-to-rtl_433-in-includertl_433_devicesh you find how to add a new decoder, in my case I had to add  'DECL(tpms_tiremoni) \ ' in the file include/rtl_433_devices.h and  "**devices/tpms_tiremoni.c**" in the file src/CMakeLists.txt. I have no file src/Makefile.am so I did nothing there. After recompiling with 'make' in the folder ../rtl_433/build and 'sudo make install' the decoder will be added.
This might not work if you already installed a rtl_433 version, but you find a compiled version after you did the 'make' under ../rtl_433/build/src. that can be executed with ./rtl_433

franki29, 2022-10-08:
Hi Lodewijk, did you managed to test the decoder?
Br. Frank

a8193, 2022-10-08:
Hi Frank, no I didn't. Was on a holiday last week 😀.
However looking to the results you've seen and uploaded, they didn't match with my measurements in both tests 11 and 12. So i don't expected serious results.
Nevertheless I'll give it a try next week
BR Lodewijk

72ka, 2023-03-08:
Hi, I have [TireMoni TM-240](https://shop.tiremoni.com/tiremoni/messbereich/bis-11-bar/reifendruck-kontrollsystem-tiremoni-tm-240-rundum-sorglos-paket/a-155) TPMS for truck. I found this thread and at first, it looks that my sensors are similar or the same. At least, the modulation and preamble is the same, so I can see some output. It looks that the sensor ID is correct (is identical in each packet). So, what next? I need only to read pressure and temperature.

For your information, there are my recordings of this sensor (all recordings are one sensor).
[TireMoni-TM240.zip](https://github.com/merbanan/rtl_433/files/10921794/TireMoni-TM240.zip)

Thanks

zuckschwerdt, 2023-03-08:
Open a new issue. Collect data with
`-Yminmax -X 'n=name,m=FSK_PCM,s=50,l=50,y=150,g=150,r=200,bits>=200,match=aaaa'`
Post a long lis of those codes.
The [codes so far](https://triq.net/bitbench#c=aaaaaaaa9caaaaad4b34cb4caab2acad2d354d2b52ab34cb4a8&c=555555554e555556a59a65aa955aa656969aa695a9559a6a954&c=555555554e555556659a65a5555aa656969aa695a9559a66954&c=555555554e555556659a65a6555a9956969aa695a9559a65a64&c=555555554e5555556596659a9566996a969aa695a9559a66664&c=aaaaaaaa9caaaaaacb2ccb2b2acd32d52d354d2b52ab34cacc8&c=555555554e555556a59a65a955599a56969aa695a9559a6a994&c=555555554e555556a59a65aa55595656969aa695a9559a69a54&c=555555554e555556659a65a5555aa656969aa695a9559a66954&c=555555554e555556659a65a6555aa656969aa695a9559a65954&f=8h8h%20%20&a=Preamble&m=554e&d=MC&cf=XOR&cl=80&cw=8) don't show the preamble (could be 5553 or 554e) or alignment. There is high entropy with XOR relation at the end but only 6 bits, so not a CRC?

gdt, 2023-09-29:
@a8193 What's the status on this?  It looks like you figured out a lot and getting this to a PR so it can be shared with others would be great.   Are you able to move forward?  Or is this already in git master?

a8193, 2023-10-29:
In the previous datasets for the short codes (Oct 1, 2022 & Oct 2, 2022) the sensor id's for the short and for the long code where different from each other. By changing the preamble for the short code to aac6 this is corrected. With this corrected preamble the sensor id's are equal for both code strings (sensor 3 = 762d51 and sensor 4 = 757761).

With this new preamble some of the tests resulted in

...
sensor|P_psi|T_dC|AA|BB|CC|DD|EE|FF|GGGGGG|HH|II
----|---|---|---|---|---|----|----|----|--------|----|----
#3|36|28|32|f9|79|3c|7d|4e|762d51|06|b4
#3|31,5|28|32|79|c2|bc|fd|ce|762d51|0f|8f
#3|27|28|32|b9|0b|fc|3d|8e|762d51|30|86
#3|23,5|28|32|79|d2|4c|fd|ce|762d51|0f|6f
#3|20|27|32|b9|19|0c|05|8e|762d51|3e|5c
#3|15,5|27|32|79|a2|4c|c5|ce|762d51|27|27
#3|11|26|32|f9|2b|dc|4d|4e|762d51|20|36
#3|7,5|27|32|79|b2|5c|c5|ce|762d51|27|27
|||||||||||
#4|16|26|33|c2|c9|1e|19|83|757761|6e|a2
#4|19|27|33|92|73|ae|d1|c3|757761|d8|b0
#4|23|26|33|d2|bb|7e|19|83|757761|28|a0
#4|26,5|26|33|52|24|fe|99|03|757761|21|bf
#4|33|27|33|d2|97|4e|11|83|757761|04|b4
#4|36,5|27|33|12|d0|0e|51|43|757761|1d|b3
#4|40|28|33|92|59|9e|e9|c3|757761|7e|92
#4|43,5|28|33|12|c2|1e|69|43|757761|77|89
#4|47,5|27|33|52|0a|ae|91|03|757761|47|c9
#4|51,5|27|33|12|32|6e|51|43|757761|5f|31
#4|59,5|28|33|12|22|7e|69|43|757761|b7|09
#4|69,5|29|33|12|1e|5e|61|bf|757761|ff|e1
#4|73,5|28|33|52|c6|6e|a9|03|757761|c3|fd

...

a8193, 2023-10-29:
Summary of my testresults
====================
The sequence of bits and the different codeformats are made so complex that it takes a lot of time to determine exactly which bit is used for what. I therefore stopped trying to figure this out further.
Attached, however, is the summary of the results I found along with some help from this forum.

The results found are:
preamble: aac6
modulation: FSK_PULSE_PCM
bitnumbers used: LSB 0 (b7 b6 b5 ... b0)

For these tests I've used:
- 2 x Tire Moni TD-1000 TPMS sensors configured by default as sensor 3 and sensor 4
- high pressure plant sprayer (max 5,5 bar)
- hairdryer for warming up a sensor
- freezer for cooling down a sensor
- DVB-T+FM+DAB 820t2 & sdr (433 MHz) USB stick connected to my Ubuntu laptop
- hertzg/rtl_433 Docker image
- aluminium foil for isolating a sensor for sending information

My Tire Moni sensors are sending 2 types of codes:

1 Long code (11 bytes) based on 11 lines
2 Short code (10 bytes) based on 1 line

=======
1 Long code (11 bytes) based on 11 lines

Temperature+pressure is send directly when the pressure or temperature is decreasing or increasing.
This code consists of a serie of 11 lines which belongs together.
Each line has in fact only one relevant databyte. 
The other bytes are only info about sequence and, as I suppose, some kind of CRC. It's possible that some form of encryption is taking place.
These codesequences are in the structure as described in the message from Sep 29,2022 see
https://github.com/merbanan/rtl_433/issues/2177#issuecomment-1262792032

Because these codes are sent immediately when there is a pressure or temperature change, they seems to me the most important codes.
Unfortunately I have not been able to understand these codes further.

=======
2 Short code (10 bytes) based on 1 line

Temperature+pressure code is send periodically. This code is send every 30 sec when the sensor -wheel- is moving.
There are 80 bits (10 bytes) named wb[0]-wb[79]

Calculation of Temperature and Pressure seems something like:
 
  ================ Sensor ID
    wb[48] .. wb[71]

  ================ Temperature Calculation in degrees Celsius
    t8 = wb[29] ^ 1;
    t7 = wb[30] ^ wb[7];
    t6 = wb[31];
    t5 = wb[32] ^ wb[79] ^ wb[6] ^ wb[7] ^ (wb[7] && wb[9]) ^ (wb[7] && (wb[9] ^ wb[75] ^ 1));
    t4 = wb[33] ^ wb[75] ^ wb[7];
    t3 = wb[34];
    t2 = wb[35] ^ wb[30];
    t1 = wb[36] ^ 1;

 CalcTemperature =  -(t8 * 128) + (t7 * 64) + (t6 * 32) + (t5 * 16) + (t4 * 8) + (t3 * 4) + (t2 * 2) + t1;

 Known/detected mismatches: in case of negative temperatures mostly it seems the calculated value is 2 degrees to low 

  ================ Pressure Calculation in psi
    p8 = wb[16] ^ wb[40] ^ wb[42] ^ wb[7] ^ (wb[2] && wb[9]) ^ (wb[3] && wb[10]);
    p7 = wb[17] ^ wb[41] ^ wb[43] ^ 1;
    p6 = wb[18] ^ wb[7];
    p15a = (wb[19]*16) + (wb[20]*8) + (wb[21]*4) + (wb[22]*2) + wb[21];
    p15b = p15a ^ (wb[7]*8);
      
 CalcPressure = ( (p8*128) + (p7*64) + (p6*32) + (1-p15b) ) / 2;
      
 Known/detected mismatches: 
    - measured (real) pressure = 22,5 -> calculated = -32
    - measured (real) pressure = 37,5 -> calculated = -32
    - measured (real) pressure = 45,0 -> calculated = -32

=======
As said the long code is the most important but also the most complex one. For the short code I think I've found most of the algorithm however there are still some errors in it.
Takes me too much time to look into this further. That is why I have stopped researching.

gdt, 2024-06-04:
@a8193 Status?  Are you working towards a PR?

closed 2024-06-04 by a8193