merbanan/rtl_433 #2186: How to publish unixtime in utc?

HolgerAusB, 2022-09-24:
Hi,
I tried for `output mqtt`:

```
report_meta time:unix:utc:usec
report_meta time:unix:utc
report_meta time:utc:unix
report_meta utc:unix
```
only one row at the time. But I never get the unixtime for UTC, always in localtime. Is that a bug or did I miss something else?

Tried on two Bullseye-Raspies
- one with rtl-433 from bullseye-repo: version 20.11-1
- one installed from scratch: version 21.05-79-ga43c2a05

zuckschwerdt, 2022-09-24:
Those are correct. Strange, if I don't use `unix` it works fine. Same for you?
The `utc` option just sets the env var "TZ" to "UTC" and `unix` or `iso` just set a format string.

HolgerAusB, 2022-09-24:
Yes, @zuckschwerdt!
`report_meta time:utc` and even undocumented
`report_meta utc`

brings the UTC-time in human readble format

gdt, 2022-09-24:
It seems very strange for "unix time" (as returned by time(2)) to be in anything but UTC, no matter what someone tries to ask for.    What does it mean "unix time in local time"?  Is that seconds since 1970-01-01 at 0000 local?  If so, I would say it's a bug that it can be produced.

zuckschwerdt, 2022-09-24:
@gdt good point. Somewhat unexpected how `localtime_r()` and `strftime()` interact still.
https://github.com/merbanan/rtl_433/blob/master/src/r_api.c#L359-L363
https://github.com/merbanan/rtl_433/blob/master/src/r_util.c#L66-L75

I don't think I'll investigate that further, PR to clear up the docs or fix something welcome though.

HolgerAusB, 2022-09-24:
OK, seems to be my fault. As I now read at wikipedia, unixtime **is** always UTC. I fell for a Python example that obviously gave a wrong value here. And now knowing that unix=utc it is oviously wrong:

```
from datetime import timezone, datetime
currentTime = int(datetime.strftime(datetime.utcnow(), "%s"))
```
In comparison, I naturally thought that the error was with rtl_433, since example `time:unix:utc:usec` is explicitly listed in example.conf.

closed 2022-09-24 by HolgerAusB

gdt, 2022-09-24:
> @gdt good point. Somewhat unexpected how `localtime_r()` and `strftime()` interact still. https://github.com/merbanan/rtl_433/blob/master/src/r_api.c#L359-L363 [...]

Interesting!  To me the odd thing is taking the seconds-since-epoch and converting it to tm and then using %s and it getting converted back.  IMHO that's just asking for trouble, vs just printing the seconds.  I'll save the notification and perhaps someday will look at this.

zuckschwerdt, 2022-09-25:
Huh, [strftime(3)](https://man7.org/linux/man-pages/man3/strftime.3.html) tells us that `"%s"` will always print UTC given a `struct tm` and that works with a struct from gmtime_r() and also localtime_r() which [ctime(3)](https://man7.org/linux/man-pages/man3/ctime.3.html) tells us has 9 fields that don't include *any* TZ information.
But [glibc `struct tm`](https://sourceware.org/git/?p=glibc.git;a=blob_plain;f=time/bits/types/struct_tm.h;hb=HEAD) has two more fields to make that work:
`long int tm_gmtoff; /* Seconds east of UTC.  */`
`const char *tm_zone; /* Timezone abbreviation.  */`

So, yeah, it works, is guaranteed to work, but somewhat inefficient and obscure.