merbanan/rtl_433 #2188: Fine Offset WH24 not detected

handly-aero, 2022-09-25:
Hi,
I am using rtl_433 to read data from WH24 and other sensors (WH51 and WS2032).
A couple of weeks ago rtl_433 stopped reading transmissions only from WH24. Other sensors are OK.
I am pretty sure that WH24 is proprely transmitting because I also have a GW1000 WiFi Weather Station Gateway, which is receiving data with good signal strengh. My DVB dongle is close to it.
Can someone help me?
Thanks!

zuckschwerdt, 2022-09-25:
Try https://triq.org/rtl_433/ANALYZE.html#verify-a-transmission to make sure you are getting the data packets.

gdt, 2023-10-14:
This was a question, answered long ago.

Feel free to open a discsussion for more help or post on the list.

closed 2023-10-14 by gdt