merbanan/rtl_433 #2189: Unable to decode packets for [192] Govee Water Leak Detector H5054

otanim, 2022-09-27:
Using the old version `20.02` (the latest apt package for Ubuntu 20.02) which doesn't support the device `Govee Water Leak Detector H5054` yet, detects it as `Smoke-GS558`:
```
[REDACTED]@[REDACTED]:~/rtl_433/build$ ./src/rtl_433 
rtl_433 version 20.02 branch  at 202002171252 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/otanim/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-09-27 22:27:06
model     : Smoke-GS558  id        : [REDACTED]
unit      : 31           learn     : 0             Raw Code  : [REDACTED]
```

However the latest versions (`master` branch) doesn't seem like is able to detect anything, no output, but here is the verbose message for it (used `-vv` argument, only trigger-receive output):
```
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
[... 62 lines cut ...]
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v2_1_decode: Badly formatted OS v2.1 message: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
oregon_scientific_v3_decode: Unrecognized Msg in OS v3: {50} 7f ff 99 3d 93 6d 80 : 01111111 11111111 10011001 00111101 10010011 01101101 10
[... cut ...]
norgo_decode: wrong size of bit per row 60
norgo_decode: wrong size of bit per row 82
norgo_decode: wrong size of bit per row 106
norgo_decode: wrong size of bit per row 132
markisol_decode: bits_per_row[0] = 28
markisol_decode: bits_per_row[0] = 49
markisol_decode: bits_per_row[0] = 49
markisol_decode: bits_per_row[0] = 49
markisol_decode: bits_per_row[0] = 49
markisol_decode: bits_per_row[0] = 48
```

Any thoughts on what could've caused this, or do I need to provide any additional data for the debugging?

franki29, 2022-10-04:
Hi , the sensor is already supported see https://github.com/merbanan/rtl_433/issues/1518#issuecomment-770428323
In this thread you will see a cu8 file that works fine with rtl_433, just try it in your installation it should give you:

time      : @0.090488s
model     : Govee-Water  id        : 26902
event     : Water Leak   Raw Code  : 96e904040740  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @1.670216s
model     : Govee-Water  id        : 26902
event     : Water Leak   Raw Code  : 96e904040740  Integrity : PARITY

If not please generate a cu8 file and post it, than we can have a look

otanim, 2022-10-05:
Hi @franki29,  did test the `cu8` file from [here](https://github.com/merbanan/rtl_433/issues/1518#issuecomment-721436958), and it worked for me quite fine, I got the above outputs.

Here is the cu8 file from me, for the same above mentioned device:
[DATE-2022-10-05_FCC-2AQA5-H5054.zip](https://github.com/merbanan/rtl_433/files/9717594/DATE-2022-10-05_FCC-2AQA5-H5054.zip)

zuckschwerdt, 2022-10-05:
The timing seems a little shorter, but works. Problem is the ID is invalid `FFFF` (raw dat all zeros) and the checksum/parity doesn't work. It's mostly the same timing and the packet structure looks like we expect though. Strange, someone needs to take a closer look to find out what's going on.

The Bitbuffer here is
```
{48} 00 00 cf ab 6a cb : 00000000 00000000 11001111 10101011 01101010 11001011
```
inverted this is
```
{48} ff ff 30 54 95 34 : 11111111 11111111 00110000 01010100 10010101 00110100
```

franki29, 2022-10-06:
Hi @otanim , can you send us a file that include a alarm ?

franki29, 2022-10-07:
Hi @otanim , I am to new to understand what is happening with your sensor.
The only common with your data is ff ff 3 but with only two different data files it is hard to say.
Hope that someone with more knowledge can help, sorry

otanim, 2022-10-07:
@franki29, would this info help to decode the data?
it is working on 20.02 version, it's just detecting the device as `Smoke-GS558`.

franki29, 2022-10-07:
Hi @otanim unfortunately not, there is also a thread regarding wrong detection. 
Do you have also the gateway for the detector? If so , does it show a ID in your mobile app?
But It may help if you send us some more examples of press the button and alarm cu8 files to compare the data

zuckschwerdt, 2022-10-07:
For the alarm the Bitbuffer here is
```
{48} 00 00 cd f9 76 1e : 00000000 00000000 11001101 11111001 01110110 00011110
```
inverted this is
```
{48} ff ff 32 06 89 e1 : 11111111 11111111 00110010 00000110 10001001 11100001
```

As @franki29 correctly analyzed, all bits change and the ID of just FFFF is not plausible.
We need more codes, and you should try to reset the device (remove batteries) and check if you get a proper ID somewhere.

We don't need more .cu8 files, get just the codes (only new unique codes) with
```
rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=440,l=940,g=900,r=9000,bits>=48,invert'
```

otanim, 2022-10-07:
@franki29 / @zuckschwerdt, here are some samples:
```
{27}401d24e, {1}0, {48}ffff3200e927, {1}0, {48}ffff3200e927, {1}0, {48}ffff3200e927, {1}0, {48}ffff3200e927, {1}0, {48}ffff3200e927
{27}403f20c, {1}0, {48}ffff3201f906, {1}0, {48}ffff3201f906, {1}0, {48}ffff3201f906, {1}0, {48}ffff3201f906, {1}0, {48}ffff3201f906
{27}40592ca, {1}0, {48}ffff3202c965, {1}0, {48}ffff3202c965, {1}0, {48}ffff3202c965, {1}0, {48}ffff3202c965, {1}0, {48}ffff3202c965
{27}407b288, {1}0, {48}ffff3203d944, {1}0, {48}ffff3203d944, {1}0, {48}ffff3203d944, {1}0, {48}ffff3203d944, {1}0, {48}ffff3203d944
{27}4095346, {1}0, {48}ffff3204a9a3, {1}0, {48}ffff3204a9a3, {1}0, {48}ffff3204a9a3, {1}0, {48}ffff3204a9a3, {1}0, {48}ffff3204a9a3
{27}40b7304, {1}0, {48}ffff3205b982, {1}0, {48}ffff3205b982, {1}0, {48}ffff3205b982, {1}0, {48}ffff3205b982, {1}0, {48}ffff3205b982
{27}40d13c2, {1}0, {48}ffff320689e1, {1}0, {48}ffff320689e1, {1}0, {48}ffff320689e1, {1}0, {48}ffff320689e1, {1}0, {48}ffff320689e1
{27}40f3380, {1}0, {48}ffff320799c0, {1}0, {48}ffff320799c0, {1}0, {48}ffff320799c0, {1}0, {48}ffff320799c0, {1}0, {48}ffff320799c0
{27}410d05e, {1}0, {48}ffff3208682f, {1}0, {48}ffff3208682f, {1}0, {48}ffff3208682f, {1}0, {48}ffff3208682f, {1}0, {48}ffff3208682f
{27}412f01c, {1}0, {48}ffff3209780e, {1}0, {48}ffff3209780e, {1}0, {48}ffff3209780e, {1}0, {48}ffff3209780e, {1}0, {48}ffff3209780e
{27}41490da, {1}0, {48}ffff320a486d, {1}0, {48}ffff320a486d, {1}0, {48}ffff320a486d, {1}0, {48}ffff320a486d, {1}0, {48}ffff320a486d
{27}416b098, {1}0, {48}ffff320b584c, {1}0, {48}ffff320b584c, {1}0, {48}ffff320b584c, {1}0, {48}ffff320b584c, {1}0, {48}ffff320b584c
{27}4185156, {1}0, {48}ffff320c28ab, {1}0, {48}ffff320c28ab, {1}0, {48}ffff320c28ab, {1}0, {48}ffff320c28ab, {1}0, {48}ffff320c28ab
{27}41a7114, {1}0, {48}ffff320d388a, {1}0, {48}ffff320d388a, {1}0, {48}ffff320d388a, {1}0, {48}ffff320d388a, {1}0, {48}ffff320d388a
{27}41c11d2, {1}0, {48}ffff320e08e9, {1}0, {48}ffff320e08e9, {1}0, {48}ffff320e08e9, {1}0, {48}ffff320e08e9, {1}0, {48}ffff320e08e9
{27}41e3190, {1}0, {48}ffff320f18c8, {1}0, {48}ffff320f18c8, {1}0, {48}ffff320f18c8, {1}0, {48}ffff320f18c8, {1}0, {48}ffff320f18c8
{27}421f62c, {1}0, {48}ffff3210fb16, {1}0, {48}ffff3210fb16, {1}0, {48}ffff3210fb16, {1}0, {48}ffff3210fb16, {1}0, {48}ffff3210fb16
```

- it seems like if I remove the battery and put it back and try to trigger the alarm it will generate the exact same sequences of the code in the same order.
- if I trigger the alarm within 1-5 seconds, it keeps generating the same lines of codes, but the order won't be broken.

otanim, 2022-10-07:
@franki29 / @zuckschwerdt,
More elaboration on this, I've initially purchased 2 devices (batch #1), then purchased another 5 devices (batch #2), so presumably, the initial logs I've provided were from batch #1's device (device #1), here are the 2 device records presumably from batch #1:
- device #2:
```
{27}7f7f14e, {1}0, {48}1621fbfbf8a7, {1}0, {48}1621fbfbf8a7, {1}0, {48}1621fbfbf8a7, {1}0, {48}1621fbfbf8a7, {1}0, {48}1621fbfbf8a7
```
- device #3:
```
{32}fbfbf8a7, {1}0, {48}338cfbfbf8a7, {1}0, {48}338cfbfbf8a7, {1}0, {48}338cfbfbf8a7, {1}0, {48}338cfbfbf8a7, {1}0, {48}338cfbfbf8a7
```

Interestingly enough, those last 2 devices (devices #2 and #3) are behaving differently than the first device (#1). They never change codes even if I remove batteries and put them back multiple times.

@franki29, to your question:
> Do you have also the gateway for the detector? If so , does it show a ID in your mobile app?

No, I don't have the gateway.

franki29, 2022-10-07:
Does N2 an N3 shows correct output ? If you just start rtl_433  -R 192 ?

otanim, 2022-10-07:
@franki29, yes, it does O_O, It didn't work before tho, I've rebuilt the project multiple times (different versions), and this one is the latest build.
```
time      : 2022-10-07 21:10:10
model     : Govee-Water  id        : 13196
event     : Water Leak   Raw Code  : cc7304040758  Integrity : PARITY
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-10-07 21:10:33
model     : Govee-Water  id        : 5665
event     : Water Leak   Raw Code  : e9de04040758  Integrity : PARITY
```

franki29, 2022-10-07:
So I think that the N1 is just maybe broken or not correctly programmed in the factory.

otanim, 2022-10-07:
@franki29, I'm second-guessing that theory, but I can test the rest of the 4 devices as well, to see if there are any patterns. I'll keep update here.

otanim, 2022-10-07:
@franki29, you were right, can confirm, the rest 4 devices worked for me without any issues as well...
I've opened the 2nd device, it had the same PCB and date of creation/manufacturing as the device n1.

It is quite interesting that I've been using one defective device out of 7 devices for my project.

@zuckschwerdt / @franki29, thanks again for the support and for your time, I appreciate it. I'm closing the issue.

closed 2022-10-07 by otanim

zuckschwerdt, 2022-10-07:
That was unexpected, but really seems the best explanation. Glad to hear we worked out what was wrong.