merbanan/rtl_433 #2197: Feature request:  define/override MQTT client identifier

TheKorn2, 2022-10-09:
Scenario:  two rtl_433 instances using identical hardware on two different machines talking to _**the same**_ MQTT server. (It's setup and works, why would I need two?)

Problem:  automatically generated mqtt identifiers are identical across both machines, so they take turns killing the other's MQTT session. MQTT sees identical session identifiers and assumes they're duplicate sessions of each other (mosquitto log, machine #1 is 127.0.0.1 and machine #2 is 172.18.0.84):

```
2022-10-09T04:21:24: New client connected from 172.18.0.84 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T04:21:24: New connection from 127.0.0.1 on port 1883.
2022-10-09T04:21:25: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T04:21:25: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T04:21:25: New connection from 172.18.0.84 on port 1883.
2022-10-09T04:21:25: Client rtl_433-0a97ffff already connected, closing old connection.
```

So need the ability to set/override the MQTT client identifier.  I think the current default behavior is good, just need a way to override the default for situations like these.

merbanan, 2022-10-09:
Try -d

TheKorn2, 2022-10-09:
```
 [-d <RTL-SDR USB device index> | :<RTL-SDR USB device serial> | <SoapySDR device query> | rtl_tcp | help]
````

I don't see how -d could help, please explain.

TheKorn2, 2022-10-09:
I tried whipping a "-d 0" onto one of the clients, same result:

```
2022-10-09T05:04:27: New connection from 127.0.0.1 on port 1883.
2022-10-09T05:04:28: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:28: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:29: New connection from 172.18.0.84 on port 1883.
2022-10-09T05:04:29: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:29: New client connected from 172.18.0.84 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:29: New connection from 127.0.0.1 on port 1883.
2022-10-09T05:04:30: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:30: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:30: New connection from 172.18.0.84 on port 1883.
2022-10-09T05:04:30: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:30: New client connected from 172.18.0.84 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:31: New connection from 127.0.0.1 on port 1883.
2022-10-09T05:04:31: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:31: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:32: New connection from 172.18.0.84 on port 1883.
2022-10-09T05:04:32: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:32: New client connected from 172.18.0.84 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:32: New connection from 127.0.0.1 on port 1883.
2022-10-09T05:04:33: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:33: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:33: New connection from 172.18.0.84 on port 1883.
2022-10-09T05:04:33: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:33: New client connected from 172.18.0.84 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:34: New connection from 127.0.0.1 on port 1883.
2022-10-09T05:04:35: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:35: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:35: New connection from 172.18.0.84 on port 1883.
2022-10-09T05:04:35: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:35: New client connected from 172.18.0.84 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:36: New connection from 127.0.0.1 on port 1883.
2022-10-09T05:04:36: Client rtl_433-0a97ffff already connected, closing old connection.
2022-10-09T05:04:36: New client connected from 127.0.0.1 as rtl_433-0a97ffff (p2, c0, k60, u'mqtt').
2022-10-09T05:04:36: New connection from 172.18.0.84 on port 1883.
2022-10-09T05:04:37: Client rtl_433-0a97ffff already connected, closing old connection.
```

zuckschwerdt, 2022-10-09:
The client ID is derived from hostname and device spec.
https://github.com/merbanan/rtl_433/blob/e066b669f7574c37f943f251ec8023b00cff1df6/src/output_mqtt.c#L484-L488
Either use different hostnames (I guess both are named the default `raspberrypi`?) or use a unique `-d`. Using a serial number is recommended, i.e. `rtl_eeprom` and `-d :myserial`.

closed 2022-10-09 by zuckschwerdt

TheKorn2, 2022-10-09:
I'm afraid I don't understand how to use the -d properly when it's a USB device.  I can use -d 0 and get device 0, but -d :myserial just tanks rtl_433:

```
Could not find device with serial 'myserial' (err -3)
```

Any example I could look at?

TheKorn2, 2022-10-09:
Nevermind, I got it.

TheKorn2, 2022-10-09:
For anyone else...  read your rtl_433 serial with rtl_eeprom, then add -d:(that serial number) to one instance to get a different mqtt client identifier.

In my case, I added "-d :00000001" to one of the clients to get them to stop bashing each other.