merbanan/rtl_433 #2198: rtl_433 does not report frequency, RSSI, SN, and noise for 915MHz (but does for 433 MHz)

TheKorn2, 2022-10-09:
When tuned to 433 MHz, frequency, RSSI, SN, and noise are reported as expected. But when tuned to 915MHz, all are zeroes. 

`rtl_433 -M level -M stats -f 433M -v -F kv`

produces:

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-10-09 05:32:59
model     : Acurite-5n1  subtype   : 56            id        : 3090
channel   : A            sequence_num: 2           battery   : 1             wind_speed: 2.7 km/h      temperature: 42.8 F
humidity  : 69           Integrity : CHECKSUM
Modulation: ASK          Freq      : 433.0 MHz
RSSI      : -0.7 dB      SNR       : 7.0 dB        Noise     : -7.7 dB
```

`rtl_433 -M level -M stats -f 915M -v -F kv`

produces (same stick same machine same everything, just different frequency):

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2022-10-09 05:25:04
model     : Fineoffset-WH24                        ID        : 90
Battery   : 1            Temperature: 6.6 C        Humidity  : 61 %          Wind direction: 208       Wind speed: 0.0 m/s
Gust speed: 0.0 m/s      Rainfall  : 6.0 mm        UV        : 0             UVI       : 0             Light     : 0.0 lux
Integrity : CRC
Modulation: ASK          Freq      : 0.0 MHz
RSSI      : 0.0 dB       SNR       : 0.0 dB        Noise     : 0.0 dB
```

I could maybe buy that RSSI was 0dB (...holy cows!), but frequency is zero? Wha???

zuckschwerdt, 2022-10-09:
Which version is that? IIRC there was a bug in an older version?

TheKorn2, 2022-10-09:
You are correct, sir. I was running the debian focal distribution version (literally reports "version unknown" for -V). 

`rtl_433 version 21.12-156-ge066b669 branch master at 202209242003 inputs file rtl_tcp RTL-SDR`

generates

```
time      : 2022-10-09 10:15:28
model     : Fineoffset-WH24                        ID        : 90
Battery   : 1            Temperature: 15.0 C       Humidity  : 42 %          Wind direction: 229       Wind speed: 0.7 m/s
Gust speed: 1.1 m/s      Rainfall  : 6.0 mm        UV        : 1116          UVI       : 2             Light     : 42890.0 lux
Integrity : CRC
Modulation: FSK          Freq1     : 914.9 MHz     Freq2     : 915.0 MHz
RSSI      : -1.2 dB      SNR       : 18.0 dB       Noise     : -19.2 dB
```

Thanks!

closed 2022-10-09 by TheKorn2

zuckschwerdt, 2022-10-09:
Happy to help! And Ubuntu Focal Fossa is from 20.04 -- we fixed quite some things in the last 2.5 years ;)

TheKorn2, 2022-10-09:
Yeah sorry about that, double checking the version is usually the first thing I do.