merbanan/rtl_433 #2199: Multiple 'rtlsdr_cancel_async()'

gvanem, 2022-10-13:
I note that on exit, `rtl_433.exe` calls `rtlsdr_cancel_async()` several times from `sdr_stop()`.
Adding some traces to `librtlsdr.c`, I see this:
```
librtlsdr.c(2742): rtlsdr_cancel_async(): r: 0, state: RTLSDR_RUNNING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
librtlsdr.c(2742): rtlsdr_cancel_async(): r: -2, state: RTLSDR_CANCELING -> RTLSDR_CANCELING
```

All except the first call is superfluous and does nothing in rtlsdr. 
Wouldn't once be enough by patching `sdr.c` like:

```diff
--- a/src/sdr.c 2022-04-15 08:51:50
+++ b/src/sdr.c 2022-10-13 11:46:29
@@ -1637,7 +1637,7 @@
 #endif

 #ifdef RTLSDR
-    if (dev->rtlsdr_dev) {
+    if (dev->rtlsdr_dev && dev->running) {
         dev->running = 0;
         return rtlsdr_cancel_async(dev->rtlsdr_dev);
     }
```

AFAICS, the comment in `librtlsdr.c` / `rtlsdr_close()`:
 `/* block until all async operations have been completed (if any) */` explains it.

Then let `rtlsdr_close()` from `r_api.c` do the rest.

zuckschwerdt, 2022-10-13:
The `dev->running` flag is more of an input to the loop, think `keep_running`. The flag to check would more likely be `dev->polling`.
But all this should soon be replaced with https://github.com/merbanan/rtl_433/tree/feat-sdrthread anyway -- the SDR data acquisition needs to be just a source of (data buffer) events and not the main loop for rtl_433 ;)

gvanem, 2022-10-13:
> But all this should soon be replaced with https://github.com/merbanan/rtl_433/tree/feat-sdrthread anyway 

OK. When will it get merged to master?

zuckschwerdt, 2022-10-13:
After the next release -- which is overdue ;)

gdt, 2023-09-26:
Is the code merged, and this issue should now be closed?  Or is it a valid bug against current git master?  (The most recent release is not relevant :-( )

zuckschwerdt, 2023-10-25:
The thread code is merged and this is outdated.

closed 2023-10-25 by zuckschwerdt