merbanan/rtl_433 #2200: Request for new Protocol or config for Mercator Ceiling Fan Remote

sabbrowne, 2022-10-14:
Dear Folks,
I have 3 off Mercator ceiling fans with RF Remote.
These remotes have 5 usable buttons. See photos attached
[image]
I have label each button from 1 to 5
There are attached gfiles each file corresponds to the button number ie button 1 (MED) is file g001_433.92M_250k.cu8

**_Thanks for any advice 
Shane_**

I have used rtl_433 command with the following options:
shane@emonpi:~$ rtl_433 -S unknown
then analysed each gfile using:
shane@emonpi:~$ rtl_433 -A g001_433.92M_250k.cu8
The following is the command output:

**shane@emonpi:~$ rtl_433 -A g001_433.92M_250k.cu8**
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/shane/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Test mode active. Reading samples from file: g001_433.92M_250k.cu8
Detected OOK package	@0.265648s
Analyzing pulses...
Total count:   13,  width: 12.51 ms		( 3128 S)
Pulse width distribution:
 [ 0] count:    7,  width:  304 us [300;312]	(  76 S)
 [ 1] count:    6,  width:  644 us [640;648]	( 161 S)
Gap width distribution:
 [ 0] count:    6,  width:  372 us [368;380]	(  93 S)
 [ 1] count:    6,  width:  712 us [708;716]	( 178 S)
Pulse period distribution:
 [ 0] count:    2,  width:  680 us [680;684]	( 170 S)
 [ 1] count:    8,  width: 1016 us [1008;1028]	( 254 S)
 [ 2] count:    2,  width: 1356 us [1352;1360]	( 339 S)
Level estimates [high, low]:  15973,    641
RSSI: -0.1 dB SNR: 14.0 dB Noise: -14.1 dB
Frequency offsets [F1, F2]:  -11028,      0	(-42.1 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 304, long_width: 0, reset_limit: 720, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=304,l=0,r=720'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {15} 3e 20     : 00111110 0010000

Detected OOK package	@0.290020s
Analyzing pulses...
Total count:   13,  width: 12.52 ms		( 3129 S)
Pulse width distribution:
 [ 0] count:    7,  width:  304 us [296;316]	(  76 S)
 [ 1] count:    6,  width:  640 us [636;648]	( 160 S)
Gap width distribution:
 [ 0] count:    6,  width:  372 us [372;376]	(  93 S)
 [ 1] count:    6,  width:  712 us [708;720]	( 178 S)
Pulse period distribution:
 [ 0] count:    2,  width:  688 us [684;692]	( 172 S)
 [ 1] count:    8,  width: 1016 us [1008;1024]	( 254 S)
 [ 2] count:    2,  width: 1352 us [1352;1356]	( 338 S)
Level estimates [high, low]:  15941,    552
RSSI: -0.1 dB SNR: 14.6 dB Noise: -14.7 dB
Frequency offsets [F1, F2]:  -11428,      0	(-43.6 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 304, long_width: 0, reset_limit: 724, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=304,l=0,r=724'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {15} 3e 20     : 00111110 0010000
[... 697 lines cut ...]
Pulse period distribution:
 [ 0] count:    2,  width:  684 us [684;688]	( 171 S)
 [ 1] count:    9,  width: 1012 us [1004;1028]	( 253 S)
 [ 2] count:    1,  width: 1348 us [1348;1348]	( 337 S)
Level estimates [high, low]:  15958,    604
RSSI: -0.1 dB SNR: 14.2 dB Noise: -14.3 dB
Frequency offsets [F1, F2]:  -11113,      0	(-42.4 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 312, long_width: 0, reset_limit: 712, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=312,l=0,r=712'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1 
[00] {15} 3e 02     : 00111110 0000001

**CU8 Files for each button**

[g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/9782895/g001_433.92M_250k.cu8.zip)

[g002_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/9782896/g002_433.92M_250k.cu8.zip)

[g003_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/9782898/g003_433.92M_250k.cu8.zip)

[g004_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/9782899/g004_433.92M_250k.cu8.zip)

[g005_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/9782900/g005_433.92M_250k.cu8.zip)

sabbrowne, 2022-10-14:
**Sorry I forgot the other photos!**
[image]

[image]

[image]

zuckschwerdt, 2022-10-14:
Good documentation!

Drop a file on https://triq.org/pdv/ -- you'll get the OOK decode.
It's  [Fixed bit-width PWM with leading gaps](https://triq.org/rtl_433/PULSE_FORMATS.html#fixed-bit-width-pwm-with-leading-gaps), bit-width is ~1000µs, short pulse ~300µs, long pulse ~700 µs, this will work:
rtl_433 -R 0 -X 'n=Mercator,m=OOK_PWM,s=300,l=700,g=1000,r=12000,bits=13'

See our [conf folder](https://github.com/merbanan/rtl_433/tree/master/conf) on how to complete a simple remote with a flex decoder conf.

sabbrowne, 2022-10-14:
> Good documentation!
>
> Drop a file on https://triq.org/pdv/ -- you'll get the OOK decode. It's [Fixed bit-width PWM with leading [...]

Wow Thats amazing... 
I have used that cmd and options. I now get formatted data... Just need to work out how to get that into Homeassistant using a transeiver and automations

Thanks for your quick reply

sabbrowne, 2022-10-14:
> Good documentation!
>
> Drop a file on https://triq.org/pdv/ -- you'll get the OOK decode. It's [Fixed bit-width PWM with leading [...]

Okay I can see how I can create the new config file, but where exactly do I put it in the HomeAssitant config folder
perhaps: /config/rtl_433

gdt, 2023-10-14:
This is basically now a request for help with homeassistant which is out of scope, but it seems that perhaps we should be capturing a flex decoder config and stashing it before closing.

sabbrowne, 2023-10-14:
Hi.
I have Mercator fans. I was able to use Flex and an SDR to decode the rf
remotes, with some help from the rf community. Unfortunately I lost those
decodes when my laptop died. Never got back to it to redo. Besides I
couldn’t work how retransmit using HA and openMQTTgatewat on esp32 with rf
433 module

Any help would be great

Cheers

Shane

klohner, 2024-01-02:
Here's a decoder that should work:

`-X "n=Mercator,m=OOK_PCM,s=333,l=333,r=2000,match={20}b6db2,preamble={1}8,bits=43,symbol_one={3}6,symbol_zero={3}2,get=Button:@6:{6}:[16:1 8:2 2:3 32:4 1:5 20:1 12:2 6:3 36:4 5:5],get=Battery:@9:{1}:[1:LOW 0:OK]"`

And if you want something pretty for your rtl_433.conf  file:
```
decoder {
    name=Mercator,
    modulation=OOK_PCM,
    short=333,
    long=333,
    reset=2000,
    match={20}b6db2,
    preamble={1}8,
    bits=43,
    symbol_one={3}6,
    symbol_zero={3}2,
    get=Button:@6:{6}:[16:1 8:2 2:3 32:4 1:5 20:1 12:2 6:3 36:4 5:5],
    get=Battery:@9:{1}:[1:LOW 0:OK]
}

```

I'm not sure about the Low Battery detection - I just made a guess that it might report this condition.  If anyone has this remote and a low battery, please test.

[BitBench](https://triq.net/bitbench#c=%7B12%7Df90%20%5Bg001_433.92M_250k.cu8%5D&c=%7B12%7Df88%20%5Bg002_433.92M_250k.cu8%5D&c=%7B12%7Df82%20%5Bg003_433.92M_250k.cu8%5D&c=%7B12%7Dfa0%20%5Bg004_433.92M_250k.cu8%5D&c=%7B12%7Df81%20%5Bg005_433.92M_250k.cu8%5D&f=bbbbb%20b%206b&cw=4)

[image]

klohner, 2024-01-02:
Looking at the photo of the remote, maybe that bit is for the center button and not a low battery?  Are there actually 6 buttons?

Maybe this is better:

`-X "n=Mercator,m=OOK_PCM,s=333,l=333,r=2000,match={20}b6db2,preamble={1}8,bits=43,symbol_one={3}6,symbol_zero={3}2,get=Button:@6:{6}:[1:LIGHT 2:OFF 4:UNKNOWN 8:LOW 16:MED 32:HI]"`
```
decoder {
    name=Mercator,
    modulation=OOK_PCM,
    short=333,
    long=333,
    reset=2000,
    match={20}b6db2,
    preamble={1}8,
    bits=43,
    symbol_one={3}6,
    symbol_zero={3}2,
    get=Button:@6:{6}:[1:LIGHT 2:OFF 4:UNKNOWN 8:LOW 16:MED 32:HI],
}
```

[image]

sabbrowne, 2024-01-02:
> Here's a decoder that should work:
>
> -X
> 8:2 2:3 32:4 1:5 20:1 12:2 6:3 36:4 ***@***.***:{1}:[1:LOW 0:OK]"
>
> And if you want something pretty for your rtl_433.conf file:
>
> decoder {
>     name=Mercator,
>     modulation=OOK_PCM,
>     short=333,
>     long=333,
>     reset=2000,
>     match={20}b6db2,
>     preamble={1}8,
>     bits=43,
>     symbol_one={3}6,
>     symbol_zero={3}2,
> }
>
>
> I'm not sure about the Low Battery detection - I just made a guess that it
> might report this condition. If anyone has this remote and a low battery,
> please test.
>
> BitBench
> <https://triq.net/bitbench#c=%7B12%7Df90%20%5Bg001_433.92M_250k.cu8%5D&c=%7B12%7Df88%20%5Bg002_433.92M_250k.cu8%5D&c=%7B12%7Df82%20%5Bg003_433.92M_250k.cu8%5D&c=%7B12%7Dfa0%20%5Bg004_433.92M_250k.cu8%5D&c=%7B12%7Df81%20%5Bg005_433.92M_250k.cu8%5D&f=bbbbb%20b%206b&cw=4>
>
> image.png (view on web)

Hi Karl
Thanks for your decode. I have tried and works well. I was just hoping the
Mercator fan remotes could be added to the distributed configuration file.
I have since moved and no longer require the decode. Thanks for your help

Shane

gdt, 2024-06-04:
Is anyone up to a PR to add the flex decoder?  Otherwise this seems abandoned.