merbanan/rtl_433 #2203: Option to include/exclude protocols from the default set

drws, 2022-10-16:
The default protocol list can be overriden, but there is no option to simply include or exclude some protocols from it. It would make a handy option sometimes.

zuckschwerdt, 2022-10-16:
Please describe in more detail what you want to achieve and what you tried. The `-R` option should exactly handle that use case:
```
  [-R <device> | help] Enable only the specified device decoding protocol (can be used multiple times)
       Specify a negative number to disable a device decoding protocol (can be used multiple times)
```

drws, 2022-10-16:
Oh, sorry about missing exclusion possibility. However, I believe that a protocol cannot be included with the default list. Specifying positive number overrides the protocol list.

For example, I would like to use the default protocol list while also including protocol 216 (ANT/ANT+). Am I missing something?

zuckschwerdt, 2022-10-16:
In that case use an exclusion first to setup the default list, e.g. `-R -216 -R 216`

drws, 2022-10-23:
That is great to hear and will come in handy. It would still be great to have a shorter and more intuitive way, for example `-R +216`.

gdt, 2023-09-27:
IMHO this is both a doc bug, and a functionality bug.

We have a default set of enabled protocols, basically the ones that are not problematic (due to false decodes).   It makes perfect sense for someone to want to add one and keep the defaults.

gdt, 2023-09-27:
I lean to making -R add/remove, and have another switch, perhaps "-R all" to remove all defaults.   That's breaking.   Another approach is to add a new "-Z n" (but not Z) to add n w/o clearing the defaults.   But basically I find that "enable a decoder" removes other decoders to be very surprising/non-intuitive semantics.