merbanan/rtl_433 #2207: rtl_433_mqtt_hass.py skips the Consumption key for SCMplus devices

bondiblueos9, 2022-10-19:
rtl_433_mqtt_hass.py skips the Consumption key for SCMplus devices.  A quick fix would be to copy the "consumption_data" mapping to a new "Consumption" mapping.

zuckschwerdt, 2022-10-20:
The "consumption_data" value is not ideal, there is no unit known and it might not be consistent across devices.
Some meter types transmit consumption in 1 kWh units, while others use more granular 10 Wh units – and that's likely only true for electric types, what about gas and water meters?

We really should not auto configure fields for which no common semantic is defined. We should refine the use of that field and come up with a [documented](https://triq.org/rtl_433/DATA_FORMAT.html#sensor-data) standard semantic, likely with new field names.

merbanan, 2022-10-20:
I think we should add a configuration option for the decoder that adds a scale parameter per type. If it is configured it can then output a properly scaled parameter.

That is the most sane thing we can do.

bondiblueos9, 2022-10-20:
Those are good points. I agree it is not ideal to have any value pulled in that doesn't have a defined unit of measurement since it is not a fully automatic solution; however, having the value pulled into hass with no unit of measurement (since it is not guaranteed) and then needing to manipulate the value in hass to do what is needed is better in my opinion than not having it pulled in at all and then needing to modify the script locally, run it, check periodically for any new code updates to pull in improvements, do it again, etc., without having that modification in the main codebase for others to use who might not be familiar with how to deal with the code.

In my particular case, I had been using my own solution with rtlamr for an SCM electric meter, was very excited when rtl433 gained support for SCM (since I had been using it for other things), was also excited when rtl433 gained mqtt support, then discovered this script and the rtl433 home assistant add on and auto discovery add on which uses this script (https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433_mqtt_autodiscovery), was pleased it pulled in consumption_data from my SCM meter, and added some configuration in home assistant to use that value (everything else, mostly thermometers, just worked). Then I had my meter replaced with an SCM+ meter, was pleased that rtl433 also decoded SCM+, but saw that the script ignored Consumption, so I copied the consumption_data block as a new Consumption block and ran the script to get the value in. In my case, both the old meter consumption_data and the new meter Consumption are in 10 Wh as you said. If I were simply running the script on my own, then I would build everything in place to keep the script updated and restarting as needed, but since I'm using that addon to do it, I can't make persistent changes as easily.

I also agree that it would be even more ideal to have a separate configuration file for the script to override the default behavior and either add fields that are missing or modify existing ones with scaling (like you said) or unit or other options or to expose diagnostic keys that are intentionally skipped, all without needed to make changes to the base script. Then for anyone using the script through that addon, they just update the addon (or have it set to autoupdate), and the addon is using the latest version of the script, no script modifications are ever needed, and whatever extra configuration there is continues to be used. (Of course, the addon developer would have to add support to read in the separate configuration file, similar to how the rtl433 addon reads a configuration file, but that's a downstream issue/request if it becomes relevant.)

zuckschwerdt, 2022-10-20:
We are in a bit of a corner here, things we add as keys in decoders are seen as API and there will be unhappy user if we then break things. I guess it's even more frustrating should we someday revert what's supported in the autodiscovery.

Perhaps we could add a set of additional keys (like consumption_data) behind an explicit `--experimental` flag? At least there is some implied warning that things will break someday.

bondiblueos9, 2022-10-20:
I like that idea for keys that are something useful and potentially wanted but not well enough defined yet to be feature complete and guaranteed not to change behavior in the future.  Likely much simpler than handling an additional optional configuration file.

gdt, 2023-09-27:
Should we close this and open an enhancement about regularizing fields and units?

bitfliq, 2024-01-12:
@zuckschwerdt , @merbanan , would you be open in the shorter term to matching the behavior between [SCM](https://github.com/merbanan/rtl_433/blob/master/src/devices/ert_scm.c#L83) which does not have units and [SCMPlus](https://github.com/merbanan/rtl_433/blob/ecb0b361ad487cfd9e66d4326797dbbac6b32d54/src/devices/scmplus.c#L141) (rename Consumption to consumption_data or do a dual-publish) to enable importing this consumption data in the interim until both support data with units?

zuckschwerdt, 2024-01-13:
There is also [IDM and NETIDM](https://github.com/merbanan/rtl_433/blob/master/src/devices/ert_idm.c) to consider, they have similar keys.

SCMplus has keys that should be proper values but currently are strings. Should be compatible to rtlamr somehow (also Neptune R900)?

A config (scale) per decoder might not be good enough, really we'd need it per sender ID. I don't think that we should process the data further.

Ylianst, 2024-12-01:
I am getting the same problem with Consumption being skipped.

[image]

I notice the 'rtl_433_mqtt_hass.py' code has this:

```
    "consumption": {
        "device_type": "sensor",
        "object_suffix": "consumption",
        "config": {
            "name": "SCMplus Consumption Value",
            "value_template": "{{ value|int }}",
            "state_class": "total_increasing",
        }
    },
```

However, maybe the capitalization of "Consumption" is the problem?