merbanan/rtl_433 #2213: Bresser 5in1 MQTT problem

Lobuzeczek, 2022-10-23:
I have 5 in 1 bresser stations. The problem is sending MQTT data. When temperature and humidity are sent out, rain is 0. When rain is sent after 12 seconds, temperature and humidity are 0.

[image]

MQTT DATA AFTER 12 SECONDS
[image]
AFTER 24 SECONDS
[image]

zuckschwerdt, 2022-10-23:
Yes the Bresser-6in1 sends:
- 24 seconds interval for temperature, hum, uv and rain (alternating messages)
- 12 seconds interval for wind data (every message)

But we never add "0" values. The MQTT topics either get a fresh value or are unset (unless you use `retain`).
I guess this is a processing error in your display software? What software is that?

Lobuzeczek, 2022-10-24:
> Yes the Bresser-6in1 sends:
>
> * 24 seconds interval for temperature, hum, uv and rain (alternating messages)
> * 12 seconds interval for wind data (every message)
>
> But we never add "0" values. The MQTT [...]

My software is Home Assistant

gdt, 2023-09-26:
Please feel free to open a fresh issue with reproduction information with current git master.    Please include -F json direct from rtl_433.

closed 2023-09-26 by gdt